espressif/vscode-esp-idf-extension #161: Upload firmware via jtag.[Feature Request]:

simark1979, 2020-09-09:
I want to use your extension, but impossible upload firmware in my case.
My pcb has only jtag port.
No serial.....

Thank you!

pwmb, 2020-09-09:
Thanks for the feature request, we already have a plan in our roadmap to integrate jtag flashing directly from vscode, but this feature could take sometime to release.

Till then you can flash individual bins using `openocd` from terminal/cmd

`$ openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp <app_name>.bin 0x10000 verify exit" `

More info can be [obtained here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#upload-application-for-debugging)

simark1979, 2020-09-09:
So, when approx will be released?

pwmb, 2020-09-10:
This is planed to be released in v0.6.0, while we cannot comment on exact date because it's dependent on other projects like `idf`, `openocd`, etc.

We will keep you posted on this issue for any updates

TobiasBoeschMiele, 2020-12-09:
> Till then you can flash individual bins using `openocd` from terminal/cmd
>
> `$ openocd -f board/esp32-wrover-kit-3.3v.cfg -c "program_esp <app_name>.bin 0x10000 verify exit" `
>
> More info can be [...]

@pwmb: I tried this on the [Adafruit HUZZAH32](https://www.adafruit.com/product/3405) without success.

I can flash this board using the serial cable but I cannot using JTAG.
I can debug this board with my JTAG connection - without problem. (The same command as below - without the `-c` option)

Here is the output: 
```
C:\somepath>openocd -f ./esp-huzzah-32.cfg  -c "program_esp build/myapp.elf 0x10000 verify reset exit"
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info: esp-huzzah-32.cfg is not an official configuration file.
Info : Configured 2 cores
Info : J-Link V10 compiled Jul 23 2019 13:46:10
Info : Hardware version: 10.10
Info : VTarget = 3.321 V
Info : clock speed 14000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller 0 was reset.
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : cpu0: Target halted, PC=0x40149F66, debug_reason=00000000
Info : cpu0: Debug controller 0 was reset.
Info : cpu0: Core 0 was reset.
Info : cpu0: Target halted, PC=0x500000CF, debug_reason=00000000
Info : esp32: Core 0 was reset.
Info : esp32: Debug controller 1 was reset.
Info : esp32: Core 1 was reset.
Info : Target halted. CPU0: PC=0x40000400 (active)
Info : Target halted. CPU1: PC=0x40000400
** Programming Started **
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.flash' size 4096 KB
Info : Using flash bank 'esp32.flash' size 4096 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : Using flash bank 'esp32.irom' size 496 KB
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 120 KB
Info : Flash mapping 1: 0x30020 -> 0x400d0020, 492 KB
Info : Using flash bank 'esp32.drom' size 124 KB
Warn : Adding extra erase range, 0x3f410000 .. 0x3f41001f
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : no flash bank found for address 0x3f41f000
Warn : no flash bank found for address 0x3ffcdb60
Warn : no flash bank found for address 0x40090000
Warn : Adding extra erase range, 0x400e0000 .. 0x400e001f
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
Warn : no flash bank found for address 0x4014c000
** Programming Finished **
** Verify Started **
Info : File content exceeds flash bank size. Only comparing the first 4128768 bytes of the file
Info : cpu0: Target halted, PC=0x40091856, debug_reason=00000001
** Verify Failed **
shutdown command invoked

C:\somepath>
``` 

How to work around this problem?
Let me know if this should become a separate issue.

PR #183 (2020-12-28): JTAG Flash Support (VSC-487)
This feature will enable to flash device via JTAG

##### TODOs:

- [x] Add `i18n`
- [x] Restrict simultaneous flash via `UART` & `JTAG`

Closes #161 

---

##### How to test?

- Set the settings `idf.flashType` as `JTAG` and click flash button and it should use the functionality for flashing via JTAG

---

##### Developers Note

> OpenOCD Version Requirements: [v0.10.0-esp32-20201125](https://github.com/espressif/openocd-esp32/releases/tag/v0.10.0-esp32-20201125) and above

> Awaits release and backport to IDF
closed 2020-12-28 by pwmb