espressif/vscode-esp-idf-extension #162: Lots of people getting stuck on install at "Add your ESP-IDF virtual environment python executable absolute path."

mrSilkie, 2020-09-10:
The one video tutorial for installing this extension on youtube doesn't include the "Add your ESP-IDF virtual environment python executable absolute path." requirement. 

When I make a new python virtual env, the result doesn't include a python.exe 

When I checked my equivalent 'C:\.espressif\python_env\idf4.0_py3.8_env\bin\python.exe' , the folder existed but there was no python.exe to be found.

I've even merged my python folder and my tools folder so that one single folder includes the python.exe , check_python_dependencies.py and requirements.txt exists. As a result i get.
> Command failed: "C:\dev\esp\esp-idf\tools\python.exe" "c:\dev\esp\esp-idf\tools\check_python_dependencies.py" -r "c:\dev\esp\esp-idf\requirements.txt"

I've managed to get a compile happening via idf.py build so that portion of my install is definitely working.

I'm pretty lost. At this point, I simply don't know what exactly the installer is looking for. Ideally, I'd wish for it to skip the dependency check.

pwmb, 2020-09-10:
Could you please also add the link to the video so that we can analyse this further

pwmb, 2020-09-10:
@mrSilkie When you're configuring the extension, please select the python path which is working for you, while building using `idf.py build`, the main purpose of virtual python env, is to have an isolated env for python so that all the pip dependencies don't get installed on the `--user` level but remain isolated to that virtual env.

You can also set your absolute path of your python using `idf.pythonBinPath` settings, [to edit settings check this out.](https://github.com/espressif/vscode-esp-idf-extension#esp-idf-settings)

Please refer to [Configuring your extension for more info](https://github.com/espressif/vscode-esp-idf-extension#esp-idf-configure-extension)

closed 2020-10-07 by github-actions[bot]