espressif/vscode-esp-idf-extension #164: Background color broken [Bug Report]:

simark1979, 2020-09-13:
After last update of VSCode from 1.48.2. to 1.49.0 I found that extension`s backgroung has ben broken.
Now all text almost invisible (including project configurator)
[image]

brianignacio5, 2020-09-14:
This is probably due to some change in vscode webview css variables which we reused in the Webviews.
Will take a look thanks for posting !

pwmb, 2020-09-14:
@brianignacio5 this is due to previously we were using `background: transparent` which then use to show dark background in a dark theme but now we explicitly have to tell vscode to paint the background accordingly

Will create a PR to fix this.

PR #166 (2020-09-15): Bugfix for webview background color issue
Previously webview was using `background: transparent` but now from vscode `v1.49.x` need to explicitly specify the background color for the body based on the theme selected by the users

Closes #164
closed 2020-09-15 by pwmb