espressif/vscode-esp-idf-extension #165: command:espIdf.getXtensaGdb error

simark1979, 2020-09-14:
It seems to that getXtensaGdb method broken by default:
Error: xtensa-esp32-elf-gdb is not found in idf.customExtraPaths

brianignacio5, 2020-09-14:
What is the value of your configuration setting `idf.customExtraPaths` ? Did you configure the extension following Onboarding steps ?

simark1979, 2020-09-14:
I’m not at work place now. So cannot show it right now.
But be sure, it was configured via onboard wizard. No any customisation was
done

By the way, could you tell me. Is it possible to start/stop openocd server
it tasks via your extension API?

Отправлено с мобильного устройства

simark1979, 2020-09-14:
It`s default was configured by your wizard.

C:\Users\User\.espressif\python_env\idf4.1_py3.8_env\Scripts;C:\Users\User\AppData\Local\Programs\Python\Python38-32;C:\Users\User\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin;C:\Users\User\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin;C:\Users\User\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\User\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\User\.espressif\tools\cmake\3.13.4\bin;C:\Users\User\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin;C:\Users\User\.espressif\tools\ninja\1.9.0;C:\Users\User\.espressif\tools\idf-exe\1.0.1;C:\Users\User\.espressif\tools\ccache\3.7

closed 2020-10-07 by github-actions[bot]

brianignacio5, 2020-10-16:
Should be fixed with #195