espressif/vscode-esp-idf-extension #168: [Bug Report]:

jankrejci, 2020-09-15:
Hello,
I have a problem to launch the debugger. I can connect OpenOCD and also monitor is running. But when I press the launch button, only the progress bar is running in a never-ending loop, and debugging is not started. Unfortunately, I can't see any hint, what can be wrong. I did a clean install of the vscode with extension and clean onboarding with PATH variable updated. I have tried releases 4.1 and 4.3 with same result. Build, flash, and monitor of the blink example works like a charm, but it is not possible to launch debugging.

OpenOCD output
```
Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: type 'esp32' is missing virt2phys
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
```

ESP-IDF debug adapter output
```
DEBUG_ADAPTER_STARTED
2020-09-15 14:54:31,414 - Debug Adapter (main) - CRITICAL - Printed cmd: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2020-09-15 14:54:31,419 - Debug Adapter (main) - CRITICAL - Printed cmd: DEBUG_ADAPTER_READY2CONNECT
```

Windows 10
VSCode 1.49.0
ESP-IDF release 4.1
Python 3.8
Extension 0.4.0

jankrejci, 2020-09-15:
I tried to use cppdbg with the following config and it seems it's working. But the espidf adapter still without success.

launch.json using cppdbg
```
{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "GDB",
        "type": "cppdbg",
        "request": "launch",
        "MIMode": "gdb",
        "miDebuggerPath": "C:\\Users\\Uzivatel\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r2-8.2.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gdb.exe",
        "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
        "windows": {
          "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
        },
        "cwd": "${workspaceFolder}",
        "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
        "setupCommands": [
          { "text": "target remote :3333" },
          { "text": "mon reset halt" },
          { "text": "thb app_main" },
          { "text": "flushregs" }
        ],
        "externalConsole": false,
        "logging": {
          "engineLogging": true
        }
      }
    ]
  }
```

brianignacio5, 2020-09-16:
Hi @jankrejci  thanks for posting this.

Make your sure the value is :

```
"idf.openOcdConfigs": [
    "interface/ftdi/esp32_devkitj_v1.cfg",
    "board/esp32-wrover.cfg"
  ]
```
by using `ESP-IDF: Set Espressif device target command` and selecting ESP32 in your project's folder.

You should have a `debug.log` in your project's directory and also a extension log file located in

```
**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

```

From your openOCD output it seems there is an issue configuring your target. What is the value of your `idf.openOcdConfigs` configuration settings ? You can find it in your vscode Settings (UI) command for `List of configuration files inside OpenOCD Scripts directory` or using vscode Settings (JSON) and the value for `idf.openOcdConfigs`. Is it the value as the one previously given ?

jankrejci, 2020-09-16:
Sorry for missing logs. I have added "interface/ftdi/esp32_devkitj_v1.cfg" to the settings.json. I have checked the "ESP32" target is selected in the settings and I have also executed `ESP-IDF: Set Espressif device target command` for sure. Unfortunatelly the result is still the same.

debug.log
```
2020-09-16 13:03:48,334 - Debug Adapter (main) - CRITICAL - Printed cmd: DEBUG_ADAPTER_STARTED
2020-09-16 13:03:48,340 - Debug Adapter (main) - CRITICAL - Printed cmd: DEBUG_ADAPTER_READY2CONNECT
2020-09-16 13:04:11,905 - Debug Adapter (main) - CRITICAL - Printed cmd: DEBUG_ADAPTER_STOPPED
```

esp_idf_vsc_ext.log
```
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:44:42.478Z"}
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:44:49.682Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Uzivatel\\.vscode\\extensions\\espressif.esp-idf-extension-0.4.0\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:57:02.796Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Uzivatel\\.vscode\\extensions\\espressif.esp-idf-extension-0.4.0\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:57:14.739Z"}
{"message":"WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 19.2.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\Uzivatel\\.vscode\\extensions\\espressif.esp-idf-extension-0.4.0\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:57:41.159Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\Uzivatel\\.vscode\\extensions\\espressif.esp-idf-extension-0.4.0\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-16T10:58:34.027Z"}
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2020-09-16T10:59:24.283Z"}
[... cut ...] f\\tools\\ninja\\1.9.0;C:\\Users\\Uzivatel\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\Uzivatel\\.espressif\\tools\\ccache\\3.7",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\Uzivatel\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.showOnboardingOnInit": false,
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "board/esp32-wrover.cfg"
    ]
}
```

jankrejci, 2020-09-24:
With the extension update to version 0.5.0 and esp-idf version 4.2 the debugger is working again

closed 2020-09-30 by pwmb