espressif/vscode-esp-idf-extension #169: [Feature Request]: OpenOCD auto start setting

simark1979, 2020-09-16:
I think it`s good idea to have global setting to on/off autostart OpenOCD server, if possible of course....

pwmb, 2020-09-16:
Possible Duplicate of #167

simark1979, 2020-09-16:
#167 not the same.

pwmb, 2020-09-16:
@simark1979 Could you please elaborate what you are thinking the end feature should look like and whats the use case for this scenario

simark1979, 2020-09-16:
As per jtag flashing and debugging based on top of OpenOCD it will be convenient you extension will start automaticly after starting VSCode (this oportunity can be on/off in extension`s setting)
I this case not needed to watch for this, and no need to manage start/stop before/after flashing or debugging

If I start/stop server manually via flashing script status bar doesn`t show real state of server.
The most simple way is to run server one and never close it, No control server state between flashing degugging or any combinations....

For jtag flashing much more better not to restart OpenOCD server (in this case status bar loose actual state.
Now I wrote this python script for flashing, which doesn`t stop/start server, it only connects to server via telnet.

```
import os
import json
import telnetlib
import time

try:
    tn = telnetlib.Telnet("localhost", 4444, 10)

    print(tn.read_until(b'Open On-Chip Debugger').decode("utf-8"))

    buildDir = os.environ["BUILD_DIR"].replace("\\", "/")
    with open(os.path.join(buildDir, "flasher_args.json")) as f:
        flasher_args = json.load(f)

    for addr in flasher_args["flash_files"]:
        cmd = "program_esp32 " + buildDir + "/" + \
        flasher_args["flash_files"][addr] + " " + addr + " verify exit\n"
        tn.write(cmd.encode("utf-8"))
        print(tn.read_until(b'** Programming Finished **').decode("utf-8"))

    print("\nReset target")
    tn.write(b'reset\n')
    print("\nPrograming finished!")
    tn.close()

except:
    print("Error: OpenOCD not started!")

```

pwmb, 2020-09-16:
> As per jtag flashing and debugging based on top of OpenOCD it will be convenient

When we will release the jtag flashing we will add this start of openocd before flashing so that the experience will be seamless

> if your extension could start it automaticly on VSCode startup (this oportunity can be on or off (depends on in extension`s setting)

Currently vscode extension starts automatically whenever you open `esp-idf` project, we detect if open folder is `esp-idf` project or not by determining if the root of the folder contains these files

```json
[
  "workspaceContains:**/project_description.json",
  "workspaceContains:**/sdkconfig",
  "workspaceContains:**/CMakeLists.txt"
]
```

> If I stop/start server inside of script called from task.json, status bar loose the real server status.

We don't have control if user starts the openocd outside of vscode, but we will continue to see how to make the experience better.

closed 2020-09-29 by pwmb