espressif/vscode-esp-idf-extension #175: Cannot finish installation due to python requirements

rtborg, 2020-09-23:
I have been trying to install the VS Code extension for a day, tried different versions of the ISP-IDF and still hit the same block. Here's the terminal output:

```
Checking ESP-IDF Python requirements using C:\Users\721850\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe
Python requirements from c:\Users\721850\Documents\esp-idf-v4.1\requirements.txt are satisfied.

The following Python requirements are not satisfied:
backports.functools-lru-cache>=1.5
gcovr
graphviz
psutil>=5.5.1
pygdbmi>=0.9.0.0
pylint>=1.9.4
requests>=2.21.0
typing>=3.6.6
pywin32; platform_system == "Windows"
xmlrunner>=1.7.7
To install the missing packages, please run "c:\Users\721850\Documents\esp-idf-v4.1\install.bat"
Command failed: "C:\Users\721850\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe" "c:\Users\721850\Documents\esp-idf-v4.1\tools\check_python_dependencies.py" -r "c:\Users\721850\.vscode-insiders\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt"
```

I then run `c:\Users\721850\Documents\esp-idf-v4.1\install.bat` from the VS Code terminal and it succeeds, along with `c:\Users\721850\Documents\esp-idf-v4.1\export.bat`. After that I run `C:\Users\721850\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe c:\Users\721850\Documents\esp-idf-v4.1\tools\check_python_dependencies.py -r c:\Users\721850\.vscode-insiders\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt  ` and get the following output:

```
PS C:\Users\721850\Documents\PlatformIO\Projects\ESP32_WiFiConnect> C:\Users\721850\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe c:\Users\721850\Documents\esp-idf-v4.1\tools\check_python_dependencies.py -r c:\Users\721850\.vscode-insiders\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\requirements.txt     
Traceback (most recent call last):
  File "c:\Users\721850\Documents\esp-idf-v4.1\tools\check_python_dependencies.py", line 44, in <module>
    default=os.path.join(idf_path, 'requirements.txt'))
  File "C:\Users\721850\AppData\Local\Programs\Python\Python38\lib\ntpath.py", line 78, in join
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```
Please let me know if there's a fix for that.

ctz16, 2020-09-23:
I had the same problem today. Then I tried to directly install the missing packages, and that worked. 
In your case, it's like `python -m pip install -r c:\Users\721850\.vscode-insiders\extensions\espressif.esp-idf-extension-0.4.0\esp_debug_adapter\esp_debug_adapter\requirements.txt`  Try this in your directory `C:\Users\721850\.espressif\python_env\idf4.1_py3.8_env\Scripts`

rtborg, 2020-09-24:
> I had the same problem today. Then I tried to directly install the missing packages, and that worked.
> In your case, it's like `python -m pip install -r [...]

That solved my problem - thank you!

closed 2020-09-27 by pwmb

ameerhazo, 2020-11-11:
@rtborg Hi, I have the same problem you encountered and I tried the same thing @ctz16 suggested. The missing packages are installed but I'm still getting the "missing packages error" in my output. How did you manage to solve the problem? Can you please help me.

brianignacio5, 2020-11-12:
What is the value of `idf.pythonBinPath` ? Check ESP-IDF Configuration settings (UI or JSON) for `Python bin path`. Is it the same as the one in your virtual env `C:\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe`?

ameerhazo, 2020-11-13:
> 
> 
> What is the value of `idf.pythonBinPath` ? Check ESP-IDF Configuration settings (UI or JSON) for `Python bin path`. Is it the same as the one in your virtual env `C:\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe`?

@brianignacio5 Hi Brian, I managed to fix the problem but its weird why the missing dependencies were installed somewhere else which is in my C:\Python38 folder when I invoked the python.exe in the Scripts folder to install it? Maybe it should have done that anyways?

I'm now stuck with this error.

`Python requirements from c:\Users\USER\Desktop\esp-idf\requirements.txt are satisfied.
CMake Error at C:/Users/USER/Desktop/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Error at
  C:/Users/USER/Desktop/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:96
  (include):

    include called with wrong number of arguments.  include() only takes one
    file.

  Call Stack (most recent call first):

    C:/Users/USER/Desktop/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:114 (__component_get_requirements)

Call Stack (most recent call first):
  C:/Users/USER/Desktop/esp-idf/tools/cmake/build.cmake:406 (__component_get_requirements)
  C:/Users/USER/Desktop/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!`

vadaliya, 2020-12-23:
> I had the same problem today. Then I tried to directly install the missing packages, and that worked.
> In your case, it's like `python -m pip install -r [...]

Thanks buddy... I also facing same problem... using your method it's solved...

EfraimManurung, 2022-09-29:
> I had the same problem today. Then I tried to directly install the missing packages, and that worked. In your case, it's like `python -m pip install -r [...]

Thanks a lot mate, it's fix my problem on VSCode