espressif/vscode-esp-idf-extension #178: [Bug Report]: ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/Users/xxxx/.vscode/extensions/espressif.esp-idf-extension-0.5.0/esp_debug_adapter/requirements.txt'

KevinDaddy, 2020-09-24:
**Describe the bug**
Atfer upgrade the extension to 0.5.0, I got follow errors when reconfigure esp tools. I checked the directory where requirements.txt should be in, but there is no files, even the esp_debug_adapter directory does not exist.

ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/Users/xxxx/.vscode/extensions/espressif.esp-idf-extension-0.5.0/esp_debug_adapter/requirements.txt'

- OS Name & Version: [e.g. MacOS X 10.15.6]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v4.1]
- Python Version: [e.g. 3.8]

KevinDaddy, 2020-09-24:
I solved the problem temporarily by clone the esp-debug-adapter repo into .vscode/extensions/espressif.esp-idf-extension-0.5.0, change it's to esp_debug_adapter

pwmb, 2020-09-24:
@KevinDaddy Thanks for reporting this due to CI problem for release this wasn't packaged, will be fixed in `v0.5.1`

closed 2020-09-27 by pwmb