espressif/vscode-esp-idf-extension #179: This is the hardest VSCode plug-in I've ever used.

WangYifa, 2020-09-27:
I tried two computers base on vscode and all failed to run the compiler environment for ESP-IDF.
It's always all sorts of weird mistakes.
I hope you can optimize this plugin.
I don't think we want to waste too much time building a compilation environment.
Thank you very much.

pwmb, 2020-09-27:
Thanks for your feedback, could you please __list the areas where you're facing trouble__, also please provide the __details for the 2 computers (OS, Python Version, which IDF version you tried from onboarding?)__ where you tried to install this extension.

sukesh-ak, 2020-10-01:
For starters here is an error if you just Create a new project with Arduino as a component.
How do we point to the compiler where the ESP32-Arduino folder is ?
I see only options for setting path for ESP-IDF

I am on Windows 10.

```
../main/main.cpp
../main/main.cpp:1:10: fatal error: Arduino.h: No such file or directory
 #include "Arduino.h"
          ^~~~~~~~~~~
compilation terminated.
[843/908] Linking CXX static library esp-idf\esp_http_client\libesp_http_client.a
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\cmd.exe /d /c cmake --build ." terminated with exit code: 1.
```

There are several other errors but somehow looks like I managed to fix them earlier.
I am guessing if you take a fresh windows machine and try to use only the steps in document for setup, you should be hitting all the errors yourself.

sukesh-ak, 2020-10-01:
If I use this
[image]

then this shows up. I am on ESP-IDF 4.1
[image]

jvanlangen, 2020-10-03:
You mean. It doesn't work out of the box. Why all these dependencies between software. What about downloading an IDE and just select "file->new project" Build and run... It's all getting so complicated and fragile.

I've made it until this:
> Executing task: cmake -G Ninja .. <

CMake Error at C:/Dev/esp-idf/tools/cmake/kconfig.cmake:38 (message):
  Windows requires a prebuilt mconf-idf for your platform on the PATH, or an
  MSYS2 version of gcc on the PATH to build mconf-idf.  Consult the setup   
  docs for ESP-IDF on Windows.
Call Stack (most recent call first):
  C:/Dev/esp-idf/tools/cmake/idf_functions.cmake:25 (include)
  C:/Dev/esp-idf/tools/cmake/project.cmake:5 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja .." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Not a clue how to solve this.

pwmb, 2020-10-04:
> I am guessing if you take a fresh windows machine and try to use only the steps in document for setup, you should be hitting all the errors yourself.

Sure, we have already tested the extension in a fresh windows installations and never saw the error (Please make sure the [prerequisite](https://github.com/espressif/vscode-esp-idf-extension/#prerequisites) are fulfilled for your OS)

__If you can guide us where exactly in fresh install of extension you're getting stuck and will be happy to help you get unblocked__

Also since there are many combinations of user end software installation (os, python, idf_version, etc., etc.,), since we can't test everything its very helpful when we receive feedback and issue like this, because then we can also try those configuration as your system and see where things are going wrong.

> For starters here is an error if you just Create a new project with Arduino as a component.

Yes, we have noted this bug internally, but if you'd like can create a separate issue on GH to track it specifically (because of some public holidays this week we will be slow in resolutions)

sukesh-ak, 2020-10-04:
It's a Sunday let me see if I have the patience to Uninstall everything and install it fresh again 😁

I had ESP-IDF working properly before using the extension and also python/tools. So it was ideally a matter of running through the configuration wizard. End of the wizard showed everything checked and ready to go. So the issues I had was not due to ESP-IDF dependencies.

Also can you add a way to create/design cake files? Or an elaborate tutorial? The one in the docs are still not good enough.

I have several projects stuck unable to compile because of change to cmake. Mostly are just used as components and some due to change in APIs.

pwmb, 2020-10-04:
@jvanlangen 

> __Why all these dependencies between software__

Because both the extension and IDF depends on other great open-source software to do the heavy lifting.
But we provide wizard like [Onboarding](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) to help you install all the dependencies through the UI.

> __What about downloading an IDE and just select "file->new project" Build and run... It's all getting so complicated and fragile__

I understand where you're coming from, since the IDE you're using (vscode) itself don't offer such ecosystem of `File` -> `New Project`, we can't do such thing ourselves.

We offer opening of examples project (or template project) to kick-start you next IDF project. Please read this [How to use guide](https://github.com/espressif/vscode-esp-idf-extension#how-to-use)

Also alternatively let me suggest you our [Eclipse Plugin Offering](https://github.com/espressif/idf-eclipse-plugin), which could be more familiar for you.

> __Not a clue how to solve this.__

This problem is occuring not because of the vscode extension but because of some cmake/config issue, we will investigate this further and let you know.

```
Windows requires a prebuilt mconf-idf for your platform on the PATH, or an
MSYS2 version of gcc on the PATH to build mconf-idf. Consult the setup
docs for ESP-IDF on Windows.
```

Maybe would be more helpful/quick if you ask a question in the [forum](http://esp32.com) or open an issue in [esp-idf](https://github.com/espressif/esp-idf) repo.

pwmb, 2020-10-04:
> I have several projects stuck unable to compile because of change to cmake. Mostly are just used as components and some due to change in APIs.

So are you facing this CMake related errors in vscode only or from `cmd`/`powershell` while doing `idf.py build` too?

> Also can you add a way to create/design cmake files? Or an elaborate tutorial? The one in the docs are still not good enough.

We have a plan to add a CMake editor in the vscode IDE but can't commit anything on this yet!

sukesh-ak, 2020-10-04:
> > I have several projects stuck unable to compile because of change to cmake. Mostly are just used as components and some due to change in APIs.
> 
> So are you facing this CMake related errors in vscode only or from `cmd`/`powershell` while doing `idf.py build` too?

This was mentioned as a support statement for CMake editor needs. Not part of the bug report for the extension.

> > Also can you add a way to create/design cmake files? Or an elaborate tutorial? The one in the docs are still not good enough.
> 
> We have a plan to add a CMake editor in the vscode IDE but can't commit anything on this yet!

Thanks that will help a lot.

pwmb, 2020-10-04:
@sukesh-ak 

Thanks for the feedback, we have created two tickets 

#188 CMake Editor Feature Request
#189 Arduino Component Issue 

We will receive respective updates there.

sukesh-ak, 2020-10-04:
> @sukesh-ak
> 
> Thanks for the feedback, we have created two tickets
> 
> #188 CMake Editor Feature Request
> #189 Arduino Component Issue
> 
> We will receive respective updates there.

Subscribed to both for update notification.
Will update if I get a chance to re-install the extension to update on the different issues.

Btw, How do I remove the extension config data before re-install?

pwmb, 2020-10-04:
> Btw, How do I remove the extension config data before re-install?

- You can uninstall your extension from the vscode marketplace
- Delete the esp-idf and idf-tools directory manually (optional)

pwmb, 2020-10-05:
@jvanlangen 

> Not a clue how to solve this.

```
Windows requires a prebuilt mconf-idf for your platform on the PATH, or an
MSYS2 version of gcc on the PATH to build mconf-idf. Consult the setup
docs for ESP-IDF on Windows.
```

Windows we have multiple shells namely `powershell` and `cmd` seems like the build process is using the `MSYS` shell maybe from (`git bash` etc) 

Lets first check if environment is set correctly from vscode side, run [`Open ESP-IDF Terminal`](https://github.com/espressif/vscode-esp-idf-extension#available-commands) (<kbd>Ctrl</kbd><kbd>E</kbd>+<kbd>T</kbd>) command, this will launch an IDF Terminal; and you could run `cmake` there and check the results.

Also if the you see the error there as well, maybe try changing the default terminal to `cmd` or `powershell`! _(screen shot for how to change the terminal attached)_

[image]

brianignacio5, 2020-10-12:
@sukesh-ak The thing with the Arduino ESP32 project is the following:

1. In our extension, you are using [Arduino ESP32](https://github.com/espressif/arduino-esp32) as [ESP-IDF Component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)
2. Arduino ESP32 is a separated project from ESP-IDF, which is not actively maintained as ESP-IDF.  **Arduino ESP32 uses a very specific ESP-IDF version and will not work if using other versions (we have working projects tested with v3.3 and v4.0)** You can modify Arduino ESP32 to work with your selected ESP-IDF version yourself as explained in https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-512428852

sukesh-ak, 2020-10-12:
> @sukesh-ak The thing with the Arduino ESP32 project is the following:
> 
> 1. In our extension, you are using [Arduino ESP32](https://github.com/espressif/arduino-esp32) as [ESP-IDF Component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)
> 2. Arduino ESP32 is a separated project from ESP-IDF, which is not actively maintained as ESP-IDF.  **Arduino ESP32 uses a very specific ESP-IDF version and will not work if using other versions (we have working projects tested with v3.3 and v4.0)** You can modify Arduino ESP32 to work with your selected ESP-IDF version yourself as explained in [espressif/arduino-esp32#1142 (comment)](https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-512428852)

Thank you for the response. What I am looking for is an option to set the path for Arduino for ESP32.
Yes I found the Arduino version which is in synch with IDF in one of the issues listed.

I use Arduino as a component since a lot of good libraries are available for Arduino side.

sukesh-ak, 2020-10-12:
Also if this extension doesn't work with the option "Arduino as a component" it can be removed from the menu.
If its part of the extension (as a available command) its expected to work seamlessly.

mitchjs, 2020-10-12:
can you step by step how to get arduino as component even going..
sure i can add it as component... it creates the main.cpp
i can even do my own git clone of the arduino code into components/arduino
using IDF 3.3.4...

but thats all... menuconfig doesnt work
and no way compile

all cmake issues

CMake Error at C:/esp32/ESP32-learning/arduino-as-component/main/CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  C:/Users/mitchjs/esp-idf-v3.3.4/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  C:/Users/mitchjs/esp-idf-v3.3.4/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

i assume thats what all talking about?

sukesh-ak, 2020-10-13:
> can you step by step how to get arduino as component even going..
> sure i can add it as component... it creates the main.cpp
> i can even do my own git clone of the arduino code into components/arduino [...]

Try this branch and see. Guess the 4.x ESP-IDF support for Arduino is still not complete yet.
https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0

Also check this
https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-367867856

brianignacio5, 2020-10-13:
> What I am looking for is an option to set the path for Arduino for ESP32.

Well usually ESP-IDF projects have user components in `${project_dir}/components/component-name` or set in CMakeLists.txt (if using a directory outside your current project). What the command does, is to download Arduino ESP32 as ESP-IDF component in your current project components directory. 

> If its part of the extension (as a available command) its expected to work seamlessly.
How to make it seamlessly? It work for v3.3 and v4.0 (If not please let us know). Should we remove the command until all supported esp-idf versions are supported ?

Message of course could be improved to suggest this works with only specific ESP-IDF versions.

brianignacio5, 2020-10-13:
I'm closing this issue since the original user did not provide required information about the issue and the conversation side-tracked to different topic.  

Please continue the the conversation about Arduino as a component in https://github.com/espressif/vscode-esp-idf-extension/issues/189 and CMake editor in https://github.com/espressif/vscode-esp-idf-extension/issues/188

closed 2020-10-13 by brianignacio5