espressif/vscode-esp-idf-extension #184: [Bug Report]: Debug not working

egrue, 2020-09-30:
With VSCode 1.49.2, IDF Extension 0.5.1 successful onboarding, IDF 4.1,
all defaults, on a fresh Windows 10 2004
debugging (F5) is not possible. 
I can build, flash and monitor blink (example), on ESP Wrover KIt 4.1.
When starting the openocd server I get an error:

Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: type 'esp32' is missing virt2phys
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections

How to make debugging work?

Regards

egrue

egrue, 2020-09-30:
Through onboarding I installed IDF 4.2. Then the error on starten openocd is gone:

Open On-Chip Debugger  v0.10.0-esp32-20200420 (2020-04-20-16:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Warn : Interface already configured, ignoring
Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections

But debugging does not start.

Extension log:
...
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2020-09-30T01:42:26.286Z"}
{"message":"non zero exit code 2\r\n\r\nFEHLER: Ung�ltiges Schema in \"path:pattern\" angegeben.\r\n","stack":"Error: non zero exit code 2\r\n\r\nFEHLER: Ung�ltiges Schema in \"path:pattern\" angegeben.\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\eg\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:4853)\n\tat ChildProcess.emit (events.js:223:5)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)","level":"error","timestamp":"2020-09-30T01:42:47.548Z"}
{"message":"Cannot access filePath: C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe","level":"error","timestamp":"2020-09-30T01:42:47.549Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2020-09-30T02:01:38.629Z"}
{"message":"non zero exit code 2\r\n\r\nFEHLER: Ung�ltiges Schema in \"path:pattern\" angegeben.\r\n","stack":"Error: non zero exit code 2\r\n\r\nFEHLER: Ung�ltiges Schema in \"path:pattern\" angegeben.\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\eg\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:4853)\n\tat ChildProcess.emit (events.js:223:5)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)","level":"error","timestamp":"2020-09-30T02:02:03.589Z"}
{"message":"Cannot access filePath: C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe","level":"error","timestamp":"2020-09-30T02:02:03.590Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\eg\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-30T02:09:57.265Z"}
{"message":"Command failed: dfu-util --version\nDer Befehl \"dfu-util\" ist entweder falsch geschrieben oder\r\nkonnte nicht gefunden werden.\r\n","stack":"Error: Command failed: dfu-util --version\nDer Befehl \"dfu-util\" ist entweder falsch geschrieben oder\r\nkonnte nicht gefunden werden.\r\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-30T02:09:57.281Z"}
{"message":"Command failed: \"C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"C:\\Users\\eg\\esp-idf\\tools\\check_python_dependencies.py\" -r \"C:\\Users\\eg\\esp-idf\\requirements.txt\"\n","stack":"Error: Command failed: \"C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"C:\\Users\\eg\\esp-idf\\tools\\check_python_dependencies.py\" -r \"C:\\Users\\eg\\esp-idf\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-30T02:09:58.616Z"}
{"message":"Command failed: \"C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"C:\\Users\\eg\\esp-idf\\tools\\check_python_dependencies.py\" -r \"C:\\Users\\eg\\esp-idf\\requirements.txt\"\n","stack":"Error: Command failed: \"C:\\Users\\eg\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"C:\\Users\\eg\\esp-idf\\tools\\check_python_dependencies.py\" -r \"C:\\Users\\eg\\esp-idf\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-09-30T02:09:58.616Z"}
{"message":"You are using pip version 19.0.3, however version 20.2.3 is available.\r\nYou should consider upgrading via the 'python -m pip install --upgrade pip' command.\r\n","stack":"Error: You are using pip versi [... cut ...] Users\\eg\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:46:475)\n\tat _.$onDidEndTaskProcess (c:\\Users\\eg\\AppData\\Local\\Programs\\Microsoft VS Code\\resources\\app\\out\\vs\\workbench\\services\\extensions\\node\\extensionHostProcess.js:897:637)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:94:5)","level":"error","timestamp":"2020-09-30T02:21:36.856Z"}

an-dr, 2020-09-30:
@egrue Hello!

**A. Possible fast solution (I've faced with this trying to replicate the problem)**

In terminal:

1. [Export the IDF environment](urlhttps://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-4-set-up-the-environment-variables)

2. Execute: `python -m pip install pywin32`

It seems we missed this dependency.

**B. If it doesn't help. Let's try to dig a bit.**

Please, add to the launch.json `"logLevel": 4`:

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "espidf",
            "name": "Launch",
            "request": "launch",
            "logLevel": 4
        }
    ]
}
```

and show us the output of ESP-IDF Debug Adapter.

Thank you!

egrue, 2020-09-30:
Hello Andrei,
I found, when I re-start VSC and lauch the debugger with F5 I get this error message once:
-----------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter_main.py", line 26, in <module>
    from debug_adapter import cli
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\__init__.py", line 25, in <module>
    from .debug_adapter import DebugAdapter, A2VSC_READY2CONNECT_STRING, A2VSC_STOPPED_STRING, A2VSC_STARTED_STRING
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\debug_adapter.py", line 40, in <module>
    from .command_processor import CommandProcessor
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\command_processor.py", line 14, in <module>
    from .tools import get_good_path, Measurement
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\tools.py", line 36, in <module>
    import win32api
ModuleNotFoundError: No module named 'win32api'
[Stopped] : ESP-IDF Debug Adapter
-----------------------------------------------------------------

Then I ran export.bat (with admin rights) and the successfully ran  python -m pip install pywin32, as proposed.
But when I restart VSC, start openocd server, then F5, I again get the above error message (win32api missing)

egrue, 2020-09-30:
Hello Andrei,
I found, when I re-start VSC and lauch the debugger with F5 I get this error message once:
-----------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter_main.py", line 26, in <module>
    from debug_adapter import cli
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\__init__.py", line 25, in <module>
    from .debug_adapter import DebugAdapter, A2VSC_READY2CONNECT_STRING, A2VSC_STOPPED_STRING, A2VSC_STARTED_STRING
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\debug_adapter.py", line 40, in <module>
    from .command_processor import CommandProcessor
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\command_processor.py", line 14, in <module>
    from .tools import get_good_path, Measurement
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\tools.py", line 36, in <module>
    import win32api
ModuleNotFoundError: No module named 'win32api'
[Stopped] : ESP-IDF Debug Adapter
-----------------------------------------------------------------

Then I ran export.bat (with admin rights) and the successfully ran  python -m pip install pywin32, as proposed.
But when I restart VSC, start openocd server, then F5, I again get the above error message (win32api missing).

egrue, 2020-09-30:
with loglevel 4 in launch.json the output is the same:

Traceback (most recent call last):
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter_main.py", line 26, in <module>
    from debug_adapter import cli
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\__init__.py", line 25, in <module>
    from .debug_adapter import DebugAdapter, A2VSC_READY2CONNECT_STRING, A2VSC_STOPPED_STRING, A2VSC_STARTED_STRING
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\debug_adapter.py", line 40, in <module>
    from .command_processor import CommandProcessor
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\command_processor.py", line 14, in <module>
    from .tools import get_good_path, Measurement
  File "c:\Users\eg\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\tools.py", line 36, in <module>
    import win32api
ModuleNotFoundError: No module named 'win32api'
[Stopped] : ESP-IDF Debug Adapter

egrue, 2020-09-30:
The strange thing is: I get this output only once after VSC re-start. After this the launch/debugger seems to be blocked, even if I restart openocd server, I cannot get the error output again. I alsways need to restart VSC.

egrue, 2020-09-30:
The python version installed by onboarding is:
C:\Users\eg>python
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32

egrue, 2020-09-30:
python cannot import win32api
I tried
...\Python\Python37\Scripts> python pywin32_postinstall.py -install
it runs successfully.
But the error on launching the debugger (F5) stays the same.

egrue, 2020-10-01:
Hm, any solution?
The above error is

File "c:\Users\eg.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\tools.py", line 36, in
import win32api
ModuleNotFoundError: No module named 'win32api'

in line 36 of tools.py we have an import statement: import win32api

The Win32 api is installed:
[image]

Why it is'nt imported?

I found a strange thing, maybe it helps:
In all above logs there is a backslash (\) missing between  "[USER]" and ".vscode" !
for example:
"c:\Users**_\eg.vscode_**\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\debug_adapter\tools.py"

egrue, 2020-10-02:
Any solution?
Note: I installed the actual VSC and the actual released IDF extension completely with defaults on a fresh actual Windows 10 2004.
As debugging is vital in complex projects with 2 cores etc. it should work out of the box!
So it should be tested before releasing (at least with all defaults) !

Is my above finding ( backslash missing between "[USER]" and ".vscode" in logging lines) valuable?

an-dr, 2020-10-05:
Ok, it seems a wrong python is used even after the export operation. Please check your `idf.pythonBinPathWin` setting for the extension (User and Workspace. Workspace's will overwrite User's setting if it wasn't set to the default):

[image]

Then let's check if the module installed in the python environment. For my setting it will be like (please replace the python path to your own):

```powershell
 C:\esp_wip\esp_dap_adapter_test_app
> C:\\Users\\dongr\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip list -l
Package            Version
------------------ ---------
bitstring          3.1.7
Brotli             1.0.9
certifi            2020.6.20
cffi               1.14.3
chardet            3.0.4
click              7.1.2
cryptography       3.1.1
dnspython          2.0.0
esp-windows-curses 0.1
eventlet           0.25.2
Flask              0.12.5
Flask-Compress     1.5.0
Flask-SocketIO     2.9.6
future             0.18.2
gdbgui             0.13.2.0
gevent             1.5.0
gevent-websocket   0.10.1
greenlet           0.4.17
idna               2.10
itsdangerous       1.1.0
Jinja2             2.11.2
MarkupSafe         1.1.1
monotonic          1.5
pip                20.2.3
pycparser          2.20
pyelftools         0.26
pygdbmi            0.9.0.2
Pygments           2.7.1
pyparsing          2.3.1
pyserial           3.4
python-engineio    3.13.2
python-socketio    4.6.0
pywin32            228
reedsolo           1.5.3
requests           2.24.0
setuptools         50.3.0
six                1.15.0
typing             3.7.4.3
urllib3            1.25.10
Werkzeug           0.16.1
wheel              0.35.1
windows-curses     2.1.0
xmlrunner          1.7.7
```

If `pywin32` isn't on place, let's install it (non-admin mode, since the files are in the user folder). For my `idf.pythonBinPathWin` it should look like, please replace the python path to your own:

```powershell
C:\\Users\\dongr\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip install pywin32
```

If it doesn't help, you have found another bug, in that case please write again and I'll try to reproduce it on a clean virtual machine using the same versions of everything you used (but it'll take a while, hope we will fix it using ideas above :-) )

egrue, 2020-10-05:
OK,
I selected USER as the place for settings during installation.
[image]
[image]
The user path is correct, even with the backslash between [USER] and .vscode.
(except the python version which is 3.7 on my system.
But the workspace path seems to be wrong: simply "python".

egrue, 2020-10-05:
In fact pywin is missing in my environment:
`C:\Users\eg\.espressif\python_env\idf4.2_py3.7_env\Scripts>python.exe -m pip list -l
Package            Version
------------------ ---------
bitstring          3.1.7
Brotli             1.0.9
certifi            2020.6.20
cffi               1.14.3
chardet            3.0.4
click              7.1.2
cryptography       3.1.1
esp-windows-curses 0.1
Flask              0.12.5
Flask-Compress     1.5.0
Flask-SocketIO     2.9.6
future             0.18.2
gdbgui             0.13.2.0
gevent             1.5.0
greenlet           0.4.17
idna               2.10
itsdangerous       1.1.0
Jinja2             2.11.2
MarkupSafe         1.1.1
pip                20.2.3
psutil             5.7.2
pycparser          2.20
pyelftools         0.26
pygdbmi            0.9.0.2
Pygments           2.7.1
pyparsing          2.3.1
pyserial           3.4
python-engineio    3.13.2
python-socketio    4.6.0
reedsolo           1.5.3
requests           2.24.0
setuptools         40.8.0
six                1.15.0
typing             3.7.4.3
urllib3            1.25.10
Werkzeug           0.16.1
wheel              0.35.1
windows-curses     2.1.0
xmlrunner          1.7.7

C:\Users\eg\.espressif\python_env\idf4.2_py3.7_env\Scripts>`

egrue, 2020-10-05:
Bingo! Debug seems to work now!
Andrei, thanks a lot!
[image]

egrue, 2020-10-05:
Btw: The missing backslash is an artefact of this forum's editor/viewer.

egrue, 2020-10-07:
The problem with the wrong python path occurs, wenn you select F1->Create a project, then select a project folder.
Then workspace settings are created automatically, which contain a wrong python path.

brianignacio5, 2020-10-12:
Closing this issue since it seems the issue is fixed by the user.

@egrue if you think there is something we can improve, please let us know.

closed 2020-10-12 by brianignacio5