espressif/vscode-esp-idf-extension #185: [Bug Report]: python not accessible during onboarding

jirizaj, 2020-10-02:
**Describe the bug**
During onboarding, in the step "Verify ESP-IDF Tools", after clicking on "Click here to check tools exists." I'm notified that python is not accesible. `C:\Users\j\AppData\Local\Programs\Python\Python38-32\python is not accesible.` Even leaving predefined "python" from a step before ("Verify ESP-IDF Tools") results in `python is not accesible.` The python is installed and working, I tested that by running it in cmd. 

**To Reproduce**
Make clean installation of the plugin in VSCode.

**Expected behavior**
Python should be accessible.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Win10
- VSCode Version: 1.49.2
- ESP-IDF Version: master
- Python Version: 38-32

pwmb, 2020-10-02:
> Python Version: 38-32

Could you please share the python version received from `C:\Users\j\AppData\Local\Programs\Python\Python38-32\python --version`

jirizaj, 2020-10-02:
Python Version: 3.8.6

`v4.1` and `v4.0.1` gives the same "not accessible" message

pwmb, 2020-10-02:
I see then there _could_ be a problem with the `idf.pythonBinPathWin` settings, please make sure it is pointing to the correct path (`C:\Users\j\AppData\Local\Programs\Python\Python38-32\python` in your case)

To update this settings/configuration please check [this](https://github.com/espressif/vscode-esp-idf-extension#esp-idf-settings) out.

jirizaj, 2020-10-02:
Unfortunately it didn't help, the error remains the same :( https://prnt.sc/us31x7 https://prnt.sc/us33dt

pwmb, 2020-10-02:
Very strange, somehow the path is not read accessible from nodejs code, to dig further could you share the log file

Log file can be found here:
- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

_(optional)_

Also if you have nodejs installed locally, can run this chunk of code and see whats the console output you're getting?

```js
const fs = require("fs")
fs.accessSync("C:\Users\j\AppData\Local\Programs\Python\Python38-32\python", fs.constants.R_OK);
```

jirizaj, 2020-10-02:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5320032/esp_idf_vsc_ext.log)

```js
const fs = require("fs")
console.log(fs.existsSync("C:\\Users\\j\\AppData\\Local\\Programs\\Python\\Python38-32\\python", fs.constants.R_OK));
```

`false`

```js
const fs = require("fs")
console.log(fs.existsSync("C:\\Users\\j\\AppData\\Local\\Programs\\Python\\Python38-32\\python.exe", fs.constants.R_OK));
```

`true`

It seems that it helped to add .exe to the end, but there is another issue, which I believe is for idf:

https://prnt.sc/us4nx3

```
ESP-IDF master has been cloned from Github.

The following Python requirements are not satisfied:
gdbgui==0.13.2.0
To install the missing packages, please run "c:\Users\j\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\j\AppData\Local\Programs\Python
    Python interpreter used: C:\Users\j\AppData\Local\Programs\Python\Python38-32\python.exe
Command failed: "C:\Users\j\AppData\Local\Programs\Python\Python38-32\python.exe" "c:\Users\j\esp-idf\tools\check_python_dependencies.py" -r "c:\Users\j\esp-idf\requirements.txt"
```

pwmb, 2020-10-03:
> It seems that it helped to add .exe to the end, but there is another issue, which I believe is for idf:

I think maybe try once with release version and not `master`.

Also could you share the output for `pip list` and also check if `gdbgui==0.13.2.0` is there in the outputs?

jirizaj, 2020-10-03:
It was missing :-) however now the problem changed, the required packages are installed (I installed them manually) and it still complains that they are missing: 

also i'm on `v4.1`

```
Checking ESP-IDF Python requirements using C:\Users\j\AppData\Local\Programs\Python\Python38-32\python.exe
Python requirements from C:\Users\j\esp-idf\requirements.txt are satisfied.

The following Python requirements are not satisfied:
psutil>=5.5.1
requests>=2.21.0
typing>=3.6.6
xmlrunner>=1.7.7
To install the missing packages, please run "C:\Users\j\esp-idf\install.bat"
Command failed: "C:\Users\j\AppData\Local\Programs\Python\Python38-32\python.exe" "C:\Users\j\esp-idf\tools\check_python_dependencies.py" -r "c:\Users\j\.vscode\extensions\espressif.esp-idf-extension-0.5.1\esp_debug_adapter\requirements.txt"
```

```
PS C:\Users\j> pip list
Package            Version
------------------ ---------
bottle             0.12.18  
Brotli             1.0.9    
certifi            2020.6.20
cffi               1.14.3
chardet            3.0.4
click              7.1.2
colorama           0.4.3
cryptography       3.1.1
esp-windows-curses 0.1
Flask              0.12.5
Flask-Compress     1.5.0
Flask-SocketIO     2.9.6
future             0.18.2
gdbgui             0.13.2.0
gevent             1.5.0
greenlet           0.4.17
idna               2.10
itsdangerous       1.1.0
Jinja2             2.11.2
MarkupSafe         1.1.1
marshmallow        3.8.0
pip                20.2.3
platformio         5.0.1
psutil             5.7.2
pycparser          2.20
pyelftools         0.26
pygdbmi            0.9.0.2
Pygments           2.7.1
pyparsing          2.3.1
pyserial           3.4
python-engineio    3.13.2
python-socketio    4.6.0
requests           2.24.0
semantic-version   2.8.5
setuptools         41.2.0
six                1.15.0
tabulate           0.8.7
typing             3.7.4.3
urllib3            1.25.10
Werkzeug           0.16.1
wheel              0.35.1
windows-curses     2.1.0
xmlrunner          1.7.7
```

pwmb, 2020-10-04:
I am unable to replicate this in my windows installation, maybe can you run the onboarding once again?

jirizaj, 2020-10-04:
I did, it didn't help. However I tried to go through whole process again on my laptop and it seems that there it works well. I also tried on my laptop to clone `v4.2` and use it and I got similar package issue... I'm not experienced in python, but I noticed that in .espressif folder, in my user profile there is something called `python_env` and there are some python files with release version in name... is there any good source, that I could read explaining how this works? So I could dig in and give you better information :)

pwmb, 2020-10-04:
Ya so as you spotted that `.espressif` folder in your user directory will contain `python_env` and this is a virtual python environment we create and use it for build, flash etc, this is done to avoid polluting user's actual `python`

So make sure that onboarding is finished successfully, and after that verify that `idf.pythonBinPathWin` is set to the `python_env`

You can read more about onboarding process [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md).

jirizaj, 2020-10-04:
OK, so I cleaned my computer, reinstalled python to `C:\Python38`, ESP-IDF `C:\tools\esp-idf\v4.1\esp-idf` and tools to `C:\tools\esp-idf\v4.1\tools`... then onboarding successfully finished and `idf.pythonBinPathWin` is `C:\tools\esp-idf\v4.1\tools\python_env\idf4.1_py3.8_env\Scripts\python.exe` :-) I created a project from examples (get-started/blink) and hit `Ctrl+E B` to build and the build failed. Should I make another issue for it?

```
cmake : The term 'cmake' is not recognized as the name of a cmdlet, function, 
script file, or operable program. Check the spelling of the name, or if a path 
was included, verify that the path is correct and try again.
At line:1 char:1
+ cmake -G Ninja ..
+ ~~~~~
    + CategoryInfo          : ObjectNotFound: (cmake:String) [], CommandNotFou 
   ndException
    + FullyQualifiedErrorId : CommandNotFoundException
```

pwmb, 2020-10-05:
Windows we have multiple shells namely `powershell` and `cmd` seems like the build process is using the `powershell` 

Now to check if environment is set correctly from vscode side is to run [`Open ESP-IDF Terminal`](https://github.com/espressif/vscode-esp-idf-extension#available-commands) (<kbd>Ctrl</kbd><kbd>E</kbd>+<kbd>T</kbd>) command, this will launch an IDF Terminal; and you could run `cmake` there and check the results.

Also if the you see the error there as well, maybe try changing the default terminal to cmd!

[image]

jirizaj, 2020-10-05:
ESP-IDF Terminal works well, I can run cmake from there.

pwmb, 2020-10-05:
One more question, are you using PlatformIO extension too ?

jirizaj, 2020-10-05:
Yep.

pwmb, 2020-10-05:
PTAL here #190

brianignacio5, 2020-10-16:
Closing since original issue is fixed by the user. Conversation about PlatformIO extension conflict should be discussed in #190

closed 2020-10-16 by brianignacio5

brianignacio5, 2020-10-22:
@jirizaj Can you please share which version of Python did you use ? I can't reproduce this on Windows 10 with Python 3.8

jirizaj, 2020-10-27:
It was 3.8.6, I use PlatformIO

brianignacio5, 2020-10-28:
Have you tried using onboarding with PlatformIO being disabled ? We have many users reported some conflicts and the fix is still under review.

jirizaj, 2020-11-11:
Sorry for the late response, I tested it right now and it works wit PlatformIO disabled.