espressif/vscode-esp-idf-extension #187: [Bug Report]:

jvanlangen, 2020-10-03:
>Prerequisites
>There are a few dependencies which needs to be downloaded and installed before you can continue to use the extension. All the other dependencies like ESP-IDF or toolchain will be taken care by the onboarding process.
>
>Python 3.5+
>Git
>CMake and Ninja for Linux or MacOS users. For Windows users, it is part of the onboarding configuration tools intall.

Ninja is a zipfile which contains an exe. Where should it be stored?

pwmb, 2020-10-04:
> CMake and Ninja for Linux or MacOS users. For Windows users, it is part of the onboarding configuration tools intall.

As mentioned, you don't need to Install CMake/Ninja for windows since they are already taken care by the onboarding.

brianignacio5, 2020-10-12:
If you are using MacOS or Linux, just make sure that `cmake` and `ninja` container directory are available in your environment variable PATH. Install steps can be found in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-1-install-prerequisites

brianignacio5, 2020-10-15:
I find it misleading calling this bug report when it is not, it is a question about extension configuration. 

I'm closing this issue since questions has been answered.

closed 2020-10-15 by brianignacio5