espressif/vscode-esp-idf-extension #189: [Bug Report]: Arduino Component in Extension not working

pwmb, 2020-10-04:
Please refer this https://github.com/espressif/vscode-esp-idf-extension/issues/179#issuecomment-702230362

From #179

sukesh-ak, 2020-10-05:
Why is this marked as duplicate? Can you mention duplicate of which one?

pwmb, 2020-10-05:
Since you originally reported this in #179.

Please note: `Tags` are for internal maintenance purpose.

brianignacio5, 2020-10-15:
From the conversation in #179 we still haven't received an actual bug of the feature.  Renaming this thread since it is misleading. 

IF you find a bug or have some feedback about this feature please let us know here.

sukesh-ak, 2020-11-05:
> From the conversation in #179 we still haven't received an actual bug of the feature. Renaming this thread since it is misleading.
> 
> IF you find a bug or have some feedback about this feature please let us know here.

Hi @brianignacio5 
If you use "Arduino ESP32 as ESP-IDF Component" in Extension actions command it errors out (guess due to mismatch in supported version).
Either provide a working version or you can remove the action in the list till its ready.

brianignacio5, 2020-11-06:
> Hi @brianignacio5
> If you use "Arduino ESP32 as ESP-IDF Component" in Extension actions command it errors out (guess due to mismatch in supported version).
> Either provide a working version or you can [...]

Does it work for ESP-IDF 4.0 and ESP-IDF 3.3 ?

closed 2020-11-27 by github-actions[bot]

edouardreg, 2021-02-04:
Hello, I have tried with all esp-idf 4.0 versions proposed (4.0.2 and 4.0 release branch) by the extension express setup : none is working with "install arduino as component". The error message is : ESP-IDF 4.0 or 3.3 is required...

brianignacio5, 2021-02-05:
Thanks for the feedback @edouardreg . Will take a look soon

Murloc992, 2021-02-07:
Same issue, would love to see a solution. :) For now manually adding the component works.