espressif/vscode-esp-idf-extension #191: [Bug Report]:

hustlelabs, 2020-10-05:
**Describe the bug**
When creating a new project from the template-app, the file .vscode\c_cpp_properties.json doesn't contain a line for the include files that are included with the compiler of choice.  See the following screenshot.

[image]

**To Reproduce**

1.) Press F1 and type ESP-IDF: Create ESP-IDF project
2.) Give the IDE any path in which to create the project
3.) Select template-app as the template to create the project
4.) Compile the project in order to update compile_commands.json
5.) Observe the include line for FreeRTOS.h, it should look like it does in the attached screenshot.

**Expected behavior**
The file .vscode\c_cpp_properties.json should have a line for the compiler's include directory such as:
```
"C:/Users/[user]/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/",
```
It would be improved if it could automatically choose the correct compiler include directory (ESP32 vs ESP32-S2 vs ...), and if it could use environment variables to make it more portable.  I have not identified any appropriate environment variables that are already in use by the extension.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

hustlelabs, 2020-10-05:
Note that this wouldn't be required if the Compiler Path setting were configured.  See [https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference#_configuration-properties](url)

pwmb, 2020-10-05:
Please make sure your config file is okay. Here is the [guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md#default-configuration) for the same.

Also please make sure that __C/C++ Extension Intellisense engine to Tag Parser__

> It would be improved if it could automatically choose the correct compiler include directory (ESP32 vs ESP32-S2 vs ...), and if it could use environment variables to make it more portable. I have not identified any appropriate environment variables that are already in use by the extension.

Thanks for the feedback, If you would want, could create a feature request for this separately so that can be tracked better.

> Note that this wouldn't be required if the Compiler Path setting were configured. See [https://code.visualstudio.com/docs/cpp/c-cpp-properties-schema-reference#_configuration-properties](url)

Thanks for this info will, check this one as well.

hustlelabs, 2020-10-05:
Sorry, I wasn't clear.  I had meant that the fix of adding that line could be greatly improved, vscode-esp-idf is awesome alredy.

closed 2020-10-13 by brianignacio5

dieortin, 2021-04-23:
May I ask why it's required to use the "Tag Parser" mode? This makes suggestions real bad (for example, it doesn't even suggest a variable I just declared in the line before)
I've tried changing to the default mode and it seems to be working fine (and with much better suggestions), but I might be missing something. Thanks in advance!

PR #195 (2020-10-13): Add new project compiler path
Add `compilerPath` to settings.json `C_Cpp.default.compilerPath` for `c_cpp_properties.json` default compiler path on Create project from example.

Fix #191