espressif/vscode-esp-idf-extension #193: [Bug Report]: ESP IDF stopes working properly after reopening code

Asteliks, 2020-10-11:
**Describe the bug**
I’m not sure whether this is the correct place to report this bug. Generally speaking, on a fresh installed window installing code and then the esp idf via the extension works fine but after closing and reopening code everything goes to hell. I have the same problem with eclipse everything seems to be working fine on the first run after the install but on the second some configurations just gest lost and so dose my ability to do anything (compile, run the monitor etc.) 

**To Reproduce**
Install vs code
Install the extension
Finish the setup
See that everything is working fine
Waite some time/ restart windows / restart code
Nothing works anymore and nothing helps (reinstalling or re doing the onboarding process)

**Expected behavior**
I expect the esp idf to work every time I open it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10 Education 10.0.19041
- VSCode Version: 1.50.0
- ESP-IDF Version: v4.2 - The same happens with previous versions 
- Python Version: 2.9.0 - The same happens with previous versions 

**Logs**
If applicable, please share the log file which can be obtained from

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5360804/esp_idf_vsc_ext.log)

brianignacio5, 2020-10-12:
Can you try changing your terminal in Visual Studio Code for cmd or powershell and try again ?

Asteliks, 2020-10-14:
This seems to solve all my problems. O boi. Why can't the plugin button use the correct terminal

brianignacio5, 2020-10-15:
I believe the terminal can be configured by the user when pressing F1 in VS Code and typing/selecting Terminal: Select Default Shell also can be set in each project `.vscode/settings.json` using `terminal.integrated.shell.*` like `"terminal.integrated.shell.windows": "C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\powershell.exe"`.

Hope this helps.

Closing the issue since the user since the issue is fixed.

closed 2020-10-15 by brianignacio5