espressif/vscode-esp-idf-extension #201: [Bug Report]: "ESP-IDF: Set Espressif device target" to "ESP32-S2" impossible

littleboot, 2020-10-21:
Not able to set the device target to an ESP32-S2 inside vscode.
this results in the following error.
`{
  error: [Error: non zero exit code 2
  
  Usage: c:\Users\Admin\Desktop\esp-idf\tools\idf.py set-target 
             [OPTIONS] [esp32|esp32s2beta]
  Try 'c:\Users\Admin\Desktop\esp-idf\tools\idf.py set-target --help' for help.
  
  Error: Invalid value for '[esp32|esp32s2beta]': invalid choice: esp32s2. (choose from esp32, esp32s2beta)
  
  	at ChildProcess.<anonymous> (c:\Users\Admin\.vscode\extensions\espressif.esp-idf-extension-0.5.1\dist\extension.js:1:4876)
  	at ChildProcess.emit (events.js:223:5)
  	at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)]
}`
 
Setting the target to "ESP-32" works fine (inside vscode).
Setting the target using the command line "idf.py" works fine, but I can't build or flash the device using vscode.

**To Reproduce**
Press F1
Type: "ESP-IDF: Set Espressif device target"
Type: "ESP32-S2"

**Environment (please complete the following information):**
- OS Windows 10
- VSCode Version: 1.50.1
- ESP-IDF Version: 4.1p
- Python Version: 3.7.3

**Additional context**
I already tried removing all ESP files from PC and dependencies python and git;
Reinstalled at least 6 times. using the installer and also the cmd line.
reinstalled using the VScode extension onboarding instruction. both manual idf location and new download.

Can't get it to work

pwmb, 2020-10-21:
IDF Version you're using doesn't support `esp32s2` but instead `esp32`or `esp32s2beta`, suggest you to use the latest released IDF version.

To check if your IDF Version support `esp32s2` version you could run this command in your `ESP-IDF Terminal`

- Press `F1`
- Type _"Open ESP-IDF Terminal"_
- Once the terminal is open, type `idf.py set-target esp32` into the terminal, if the command ran sucessfully, then your IDF Version is supported for `esp32s2`.

littleboot, 2020-10-25:
Hi PWMB,
Thank you for the information and quick response, I got the IDE setup and working propperly.
(I did not know the latest stable version of the ESP-IDF didn't support the ESP32S2 SOC)

closed 2020-10-25 by littleboot