espressif/vscode-esp-idf-extension #204: onboarding fault:pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter.

mynameisjyy, 2020-10-26:
HI, i got a fault at vscode esp-idf onboarding phase.
system:ubuntu 20 
fault detal：at vscode onboarding last phase：
[image]

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Saving /usr/bin/python to create python virtual environment.
pkg_resources cannot be imported probably because the pip package is not installed and/or using a legacy Python interpreter. Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Command failed: "/home/ji/esp/.espressif/python_env/idf4.1_py3.8_env/bin/python" "/home/ji/esp/esp-idf/tools/check_python_dependencies.py" -r "/home/ji/esp/esp-idf/requirements.txt"
can someone help me to solve this,thanks.

brianignacio5, 2020-10-26:
Hi, it seems there is an error in the created virtual environment. 

Can you try `/home/ji/esp/.espressif/python_env/idf4.1_py3.8_env/bin/python -m pip list` to see if there is **pkg_resources** in the output list ?

Also please provide more information about your setup, as specified in the issue template. Maybe there was an error while creating the virtual environment.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

mynameisjyy, 2020-10-28:
HI, 
I have reinstall the OS system and vscode , now it appear another fault, the information as below:
system: ubuntu 20.04   vscode:1.50.1   esp-IDF:V4.1  python :2.7.18
the information at vscode  output:
[vscode output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/5449996/vscode.output.txt)
the logs:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5450008/esp_idf_vsc_ext.log)
please help to check .

mynameisjyy, 2020-10-28:
HI, I search this isssue on internet, and install the python-dev, now it`s working correctly ,  thanks.

brianignacio5, 2020-10-29:
We still recommend you to use Python 3.5+ since our Debug Adapter requires it to work properly. You can also just use the C/C++ extension debug adapter as explained in this repository's docs/Debugging.md

Closing this issue since user has fixed the issue.

closed 2020-10-29 by brianignacio5