espressif/vscode-esp-idf-extension #206: [Feature Request]: Instructions for MDF environment and project setup

Milan991, 2020-10-29:
Are there any instructions on how to set up the MDF project after successful ESP-IDF extension installation (successful build of ESP-IDF examples)?

I've run the `ESP-IDF: Install ESP-MDF command`. After that, I tried to build one of the MDF examples but I got the error that the toolchain is not properly configured:

> -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
> -- Building ESP-IDF components for target esp32
> -- Checking Python dependencies...
> Python requirements from C:/Users/ASUS/.espressif/esp-mdf/esp-idf/requirements.txt are satisfied.
> -- Building empty aliyun_iot component due to configuration
> Loading defaults file C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/sdkconfig.defaults...
> C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/sdkconfig.defaults:18 CONFIG_TIMER_TASK_STACK_DEPTH was replaced with CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
> C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/sdkconfig.defaults:27 CONFIG_TASK_WDT_PANIC was replaced with CONFIG_ESP_TASK_WDT_PANIC
> C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/sdkconfig.defaults:28 CONFIG_TASK_WDT_TIMEOUT_S was replaced with CONFIG_ESP_TASK_WDT_TIMEOUT_S
> C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/sdkconfig.defaults:30 CONFIG_TIMER_TASK_STACK_SIZE was replaced with CONFIG_ESP_TIMER_TASK_STACK_SIZE
> C:\Users\ASUS\AppData\Local\Temp\confgen_tmps5_f6pfp:36 line was updated to CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=n
> C:\Users\ASUS\AppData\Local\Temp\confgen_tmps5_f6pfp:37 line was updated to CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=n
> CMake Error at C:/Users/ASUS/.espressif/esp-mdf/esp-idf/components/esp32/project_include.cmake:21 (message):
>   Internal error, toolchain has not been set correctly by project (or an
>   invalid CMakeCache.txt file has been generated somehow)
> Call Stack (most recent call first):
>   C:/Users/ASUS/.espressif/esp-mdf/esp-idf/tools/cmake/build.cmake:306 (include)
>   C:/Users/ASUS/.espressif/esp-mdf/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
>   C:/Users/ASUS/.espressif/esp-mdf/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
>   CMakeLists.txt:7 (project)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "C:/Projects/Upwork/Richard/examples/esp-idf-hello-world/.vscode/get-started/build/CMakeFiles/CMakeOutput.log".
> The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -Command cmake -G Ninja .." terminated with exit code: 1.

It seems to me that some kind of configuration is missing. Can anybody give me help with this?

brianignacio5, 2020-10-30:
Can you try deleting the `build` folder and retry build again ? ESP-MDF seems to be configured (MDF_PATH is saved in the extension configuration settings)

Milan991, 2020-10-30:
@brianignacio5 Thank you, that solved the issue.

closed 2020-10-30 by Milan991