espressif/vscode-esp-idf-extension #207: [Feature Request]: Configurable flash baudrate

Sarrk, 2020-11-01:
**Background**
It seems to me that when I use the ESP-IDF: Flash command (i.e. <kbd>Ctrl</kbd> + <kbd>E</kbd>  <kbd>F</kbd>) it gets the flash baud rate (the parameter passed to `-b`) from the configured device baudrate. I would like to flash at the fastest possible baud rate (i.e. 2MHz), but would not always want to communicate at that speed (prefer 115200 for comms).

```
> Executing task in folder sensor_platform: /home/cal/.espressif/python_env/idf4.3_py2.7_env/bin/python /home/cal/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 2000000 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 partition_table/partition-table.bin 0xe000 ota_data_initial.bin 0x1000 bootloader/bootloader.bin 0x10000 sensor_platform.bin <
```

**Describe the solution you'd like**
A separate configurable parameter 
A clear and concise description of what you want to happen.

**Alternatives Considered**
I tried using the workspace's tasks.json and modifying this, but it also seems to me that this is Task: Flash, and not the ESP-IDF: Flash command, so it did not override the built-in method.

I ended up modifying the extension's `/dist/extension.js` on my install and replacing `this.model.baudRate` with a fixed value of a higher baud rate.

**Additional context**
I realize that the configurable parameter would likely be rooted in the esp-idf config, but thought it would be best to start here.
I also note that changing the flash mode, from DIO to QIO in the esp config, did not seem to have an effect on the parameter passed to `--flash_mode` when flashing via the extension command.

brianignacio5, 2020-11-02:
As specified in [Readme's Board/Chip Settings](https://github.com/espressif/vscode-esp-idf-extension#board-chip-specific-settings) we have defined a baud rate configuration setting `idf.baudRate`.

Press F1 -> type Settings (UI) to open Vscode Settings -> Search for Baud rate, you will find `Idf: Baud rate`

Or open your project/workspace settings.json and add `idf.baudRate` with the value to use. 

In [Extension flash command](https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/extension.ts#L1716) you can see the value is read from configuration setting.

Please let us know if this is not working for you, in that case please share fill out the issue template with your setup so we can take a look.

dordnung, 2020-11-03:
That would work, but then monitor baud rate is also 2 MHz: https://github.com/espressif/vscode-esp-idf-extension/issues/37

brianignacio5, 2020-11-06:
It will change on the Websocket enabled  ESP-IDF Monitor but the regular `ESP-IDF: Monitor your device` command (the one the icon bar uses also CTRL/CMD + E M) monitor uses `idf.py monitor` which uses `CONFIG_ESPTOOLPY_MONITOR_BAUD` from the project's sdkconfig.

PR #209 (2020-11-13): add flash monitor baud rate (VSC-476)
Add Monitor and Flash baud rate configuration settings for vscode with higher default baud rate.

Fix #207
closed 2020-11-13 by brianignacio5