espressif/vscode-esp-idf-extension #211: [Bug Report]: Tasks:Run Task then choose one function,don't work and no freeback

QianchengYan, 2020-11-07:
[image]

brianignacio5, 2020-11-09:
**Please fill out the issue template to give us more information to understand your problem.**

Tasks.json are defined in your project's .vscode/tasks.json

Can you share the content of this field.

There is also some output from the task execution in the Output tab (near Terminal) Can you share the output to see what happens?

Have you configure the extension using the onboarding steps ?

QianchengYan, 2020-11-09:
thanks for your reply.
I had fix this problem which is caused by setting.json.
[image]
then I fix this problems:
`    "task.slowProviderWarning": true,
    "task.problemMatchers.neverPrompt": true,`
finally, "Tasks:run task" work well.

closed 2020-11-13 by brianignacio5