espressif/vscode-esp-idf-extension #218: Cannot build, flash, launch gui config tool from ESP-IDF key shortcut or taskbar.

agata277, 2020-11-18:
**Describe the bug**
ESP-IDF VSCode extension installation went through properly after few trials. At the moment I can easily build and flash and monitor project from terminal. The problem emerges when I choose to use graphic buttons at the bottom of the screen (result of ESP-IDF extension) or a key shortcut (ctr+E+B for example). It does show a pop-up with notification Building Project as if build was happening, but then nothing happens, nothing shows up in terminal, and I have to resort to using terminal commands like idf.py build etc.

**To Reproduce**
Install esp-idf and tools from Installation Wizard: https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe . Install extension and followe Onboarding steps from https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md . Open workspace, click on ESP-IDF Build project or ESP-IDF Flash project on taskbar or main menu.

**Expected behavior**
Equivalent of running idf.py build or idf.py flash or idf.py monitor from terminal

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Environment (please complete the following information):**

Extension version: 0.5.1
VS Code version: Code 1.51.1 (e5a624b788d92b8d34d1392e4c4d9789406efe8f, 2020-11-10T23:34:32.027Z)
OS version: Windows_NT x64 10.0.18363
Python version: 3.9.0

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

brianignacio5, 2020-11-19:
In your configuration settings, what is the value of `idf.customExtraPaths` ?

agata277, 2020-11-22:
> 
> 
> In your configuration settings, what is the value of `idf.customExtraPaths` ?

In Folder settings I have paths to bin files from Espressif tools fodler: 

C:\Espressif\tools\ccache\3.7;C:\Espressif\tools\cmake\3.13.4\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\idf-exe\1.0.1;C:\Espressif\tools\ninja\1.9.0;C:\Espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin

In `idf.customExtraVars` I have this:

{"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20191114\\openocd-esp32\\share\\openocd\\scripts"}

Both values for User and Workspace settings are empty

[image]

brianignacio5, 2020-11-26:
Can you please share the issue ? The screenshot doesn't provide us with any information about the error. 

Also please share the extension log in:
Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

From what you say, it seems the extension is not being properly configured and there is is an error somewhere.

agata277, 2020-11-26:
> 
> 
> The screenshot doesn't provide us with any information about the error please share the extension log in:
> Windows: %USERPROFILE%.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
> Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log
> 
> From what you say, it seems the extension is not being properly configured and there is is an error somewhere.

Here is the .log file

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5601837/esp_idf_vsc_ext.log)

brianignacio5, 2020-12-07:
Can you try replacing `idf.pythonBinPathWin` instead of `C:\Python39\python.exe` use the python executable from `C:\Espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe` (or your existing python env folder)

agata277, 2020-12-07:
I did that:
[image]

and it doesn't seem to resolve the issue. Furthermore I get this error after trying to flash my device from terminal:
[image]

This is perhaps unrelated to the extension, but my existing problem with the extension are still valid. I could build the project only from the terminal, not by using graphic buttons or key shortcuts.

brianignacio5, 2020-12-09:
Based on the error, you need to select the correct serial port. You can use `ESP-IDF:  Select port to use` command to choose the right port.

brianignacio5, 2020-12-22:
Closing this issue since original error seems to be fixed.

closed 2020-12-22 by brianignacio5