espressif/vscode-esp-idf-extension #219: [Bug Report]:

l4bspace, 2020-11-21:
**Describe the bug**
During the onboarding phase I download the ESP-IDF Tools. When verifying them *openocd-esp32* failes. It dialog tells that **v0.10.0-esp32-20191114** is expected, but **0.10.0** is found. Looking into .espressif/tools/openocd-esp32 a folder with the expected name exists.

**To Reproduce**
Delete previous tools and follow the onboarding dialog till **Verify ESP-IDF Tools**.

**Expected behavior**
That the downloaded ESP-IDF Tools survive the verification.

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Ubuntu 20.04 (in Virtual Box)
- VSCode Version: 1.51.1
- ESP-IDF Version: v4.1 slected during onboarding
- Python Version: 3.8.5

**Logs**
  1 {"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\    tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n    \tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2020-11-21T13:55:15.880Z"}
  2 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2020-11-21T14:01:06.760Z"}
  3 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-11-21T14:01:12.540Z"}
  4 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2020-11-21T14:01:51.710Z"}
  5 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-11-21T14:08:02.730Z"}
  6 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-11-21T14:14:02.868Z"}
  7 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-11-21T14:16:08.863Z"}
  8 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2020-11-21T14:37:57.253Z"}
  9 {"message":"Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger 0.10.0\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://open    ocd.org/doc/doxygen/bugs.html\n\n\tat /home/j/.vscode/extensions/espressif.esp-idf-extension-0.5.1/dist/extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (interna    l/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-11-21T14:41:35.109Z"}

**Additional context**
A IDF v4.3 setup existed before trying the VS Code extension. All setup directories were renamed/deleted.

brianignacio5, 2020-11-23:
Do you have openocd in `/usr/bin` ? It appears before openocd tool in .espressif

l4bspace, 2020-11-26:
Yes, you are right! Installed OCD during tests with Rust and the Arm Discovery Board. Uninstalled it and then onboarding worked.

closed 2020-11-30 by brianignacio5