espressif/vscode-esp-idf-extension #222: [Bug Report]: IDF Onboarding error

fynnit, 2020-11-28:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
the onboarding cant be finished

**Environment**
- OS Name & Version:  MacOS 11
- VSCode Version:1.51.1
- ESP-IDF Version: v4.0.2
- Python Version: 2.7.

**Logs**
```
Creating a new Python environment in /Users/fynn/.espressif/python_env/idf4.1_py2.7_env ...

Collecting virtualenv
  Could not fetch URL https://pypi.python.org/simple/virtualenv/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:661) - skipping
  Could not find a version that satisfies the requirement virtualenv (from versions: )
No matching distribution found for virtualenv
Command failed: "/Library/Frameworks/Python.framework/Versions/2.7/bin/python" -m pip install virtualenv --user
  Could not find a version that satisfies the requirement virtualenv (from versions: )
No matching distribution found for virtualenv
Command failed: "/Library/Frameworks/Python.framework/Versions/2.7/bin/python" -m pip install virtualenv --user
  Could not find a version that satisfies the requirement virtualenv (from versions: )
No matching distribution found for virtualenv
```

brianignacio5, 2020-11-30:
Apparently pip can't find virtualenv when creating the virtual environment. Try updating pip to latest version with `sudo pip install --index-url https://pypi.python.org/simple/ --upgrade pip` and try onboarding again ?

We recommend to use python 3 since our debug adapter requires it for its dependencies.

MrityunjaiKumar, 2020-11-30:
> Command failed: "C:\Users\xyz\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe" "c:\esp-idf\tools\check_python_dependencies.py" -r "c:\esp-idf\requirements.txt"
> Traceback (most recent call last):
>   File "c:\esp-idf\tools\check_python_dependencies.py", line 17, in <module>
>     import argparse
> ModuleNotFoundError: No module named 'argparse'
> 

Unable to change the python path while onboard, It is looking for virtual environment's python.exe even when not using virtual env.

On other hand, extension is working fine with absolute python path.

brianignacio5, 2020-12-01:
> > Command failed: "C:\Users\xyz\.espressif\python_env\idf4.3_py3.7_env\Scripts\python.exe" "c:\esp-idf\tools\check_python_dependencies.py" -r "c:\esp-idf\requirements.txt"
> > Traceback (most recent call last):
> > File "c:\esp-idf\tools\check_python_dependencies.py", line 17, in 
> > import argparse
> > ModuleNotFoundError: No module named 'argparse'
> 
> Unable to change the python path while onboard, It is looking for virtual environment's python.exe even when not using virtual env.
> 
> On other hand, extension is working fine with absolute python path.

You can change the python path during onboarding's `Verify ESP-IDF Tools` step or in vscode configuration settings `idf.pythonBinPath` or `idf.pythonBinPathWin`

fynnit, 2020-12-01:
that worked! thank you

closed 2020-12-01 by fynnit