espressif/vscode-esp-idf-extension #227: Can't build project from example

Oleg-Perevyshin, 2020-12-08:
Hello!

**Expected behavior**
I wanted to build a project from examples (Hello World).

- OS Name & Version: [MacOS X 11.0.1]
- VSCode Version: [1.51.1]
- ESP-IDF Version: [v4.1]
- Python Version: [3.9.0]

**Logs**
```
> Executing task: cmake -G Ninja .. <

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/poe/esp/esp-idf/requirements.txt are satisfied.
CMake Deprecation Warning at /Users/poe/esp/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Adding linker script /Users/poe/Desktop/VSC-Workspace/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /Users/poe/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/poe/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/poe/esp/esp-idf/components/app_trace /Users/poe/esp/esp-idf/components/app_update /Users/poe/esp/esp-idf/components/asio /Users/poe/esp/esp-idf/components/bootloader /Users/poe/esp/esp-idf/components/bootloader_support /Users/poe/esp/esp-idf/components/bt /Users/poe/esp/esp-idf/components/cbor /Users/poe/esp/esp-idf/components/coap /Users/poe/esp/esp-idf/components/console /Users/poe/esp/esp-idf/components/cxx /Users/poe/esp/esp-idf/components/driver /Users/poe/esp/esp-idf/components/efuse /Users/poe/esp/esp-idf/components/esp-tls /Users/poe/esp/esp-idf/components/esp32 /Users/poe/esp/esp-idf/components/esp_adc_cal /Users/poe/esp/esp-idf/components/esp_common /Users/poe/esp/esp-idf/components/esp_eth /Users/poe/esp/esp-idf/components/esp_event /Users/poe/esp/esp-idf/components/esp_gdbstub /Users/poe/esp/esp-idf/components/esp_http_client /Users/poe/esp/esp-idf/components/esp_http_server /Users/poe/esp/esp-idf/components/esp_https_ota /Users/poe/esp/esp-idf/components/esp_https_server /Users/poe/esp/esp-idf/components/esp_local_ctrl /Users/poe/esp/esp-idf/components/esp_netif /Users/poe/esp/esp-idf/components/esp_ringbuf /Users/poe/esp/esp-idf/components/esp_rom /Users/poe/esp/esp-idf/components/esp_serial_slave_link /Users/poe/esp/esp-idf/components/esp_websocket_client /Users/poe/esp/esp-idf/components/esp_wifi /Users/poe/esp/esp-idf/components/espcoredump /Users/poe/esp/esp-idf/components/esptool_py /Users/poe/esp/esp-idf/components/expat /Users/poe/esp/esp-idf/components/fatfs /Users/poe/esp/esp-idf/components/freemodbus /Users/poe/esp/esp-idf/components/freertos /Users/poe/esp/esp-idf/components/heap /Users/poe/esp/esp-idf/components/idf_test /Users/poe/esp/esp-idf/components/jsmn /Users/poe/esp/esp-idf/components/json /Users/poe/esp/esp-idf/components/libsodium /Users/poe/esp/esp-idf/components/log /Users/poe/esp/esp-idf/components/lwip /Users/poe/Desktop/VSC-Workspace/hello_world/main /Users/poe/esp/esp-idf/components/mbedtls /Users/poe/esp/esp-idf/components/mdns /Users/poe/esp/esp-idf/components/mqtt /Users/poe/esp/esp-idf/components/newlib /Users/poe/esp/esp-idf/components/nghttp /Users/poe/esp/esp-idf/components/nvs_flash /Users/poe/esp/esp-idf/components/openssl /Users/poe/esp/esp-idf/components/partition_table /Users/poe/esp/esp-idf/components/perfmon /Users/poe/esp/esp-idf/components/protobuf-c /Users/poe/esp/esp-idf/components/protocomm /Users/poe/esp/esp-idf/components/pthread /Users/poe/esp/esp-idf/components/sdmmc /Users/poe/esp/esp-idf/components/soc /Users/poe/esp/esp-idf/components/spi_flash /Users/poe/esp/esp-idf/components/spiffs /Users/poe/esp/esp-idf/components/tcp_transport /Users/poe/esp/esp-idf/components/tcpip_adapter /Users/poe/esp/esp-idf/components/ulp /Users/poe/esp/esp-idf/components/unity /Users/poe/esp/esp-idf/components/vfs /Users/poe/esp/esp-idf/components/wear_levelling /Users/poe/esp/esp-idf/components/wifi_provisioning /Users/poe/esp/esp-idf/components/wpa_supplicant /Users/poe/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/poe/Desktop/VSC-Workspace/hello_world/build

Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.

> Executing task: cmake --build . <

[1/8] cd /Users/poe/Desktop/VSC-Workspace/hello_world/build/esp-i...****************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/8] Performing build step for 'bootloader'
ninja: no work to do.
[3/6] Generating ld/esp32.project.ld
FAILED: esp-idf/esp32/ld/esp32.project.ld 
cd /Users/poe/Desktop/VSC-Workspace/hello_world/build/esp-idf/esp32 && /Users/poe/esp/esp-idf-tools/python_env/idf4.1_py3.9_env/bin/python /Users/poe/esp/esp-idf/tools/ldgen/ldgen.py --config /Users/poe/Desktop/VSC-Workspace/hello_world/sdkconfig --fragments /Users/poe/esp/esp-idf/components/xtensa/linker.lf       /Users/poe/esp/esp-idf/components/esp_ringbuf/linker.lf /Users/poe/esp/esp-idf/components/spi_flash/linker.lf /Users/poe/esp/esp-idf/components/espcoredump/linker.lf /Users/poe/esp/esp-idf/components/esp32/linker.lf    /Users/poe/esp/esp-idf/components/esp32/ld/esp32_fragments.lf    /Users/poe/esp/esp-idf/components/soc/linker.lf /Users/poe/esp/esp-idf/components/esp_event/linker.lf /Users/poe/esp/esp-idf/components/esp_wifi/linker.lf    /Users/poe/esp/esp-idf/components/lwip/linker.lf      /Users/poe/esp/esp-idf/components/log/linker.lf /Users/poe/esp/esp-idf/components/heap/linker.lf        /Users/poe/esp/esp-idf/components/freertos/linker.lf  /Users/poe/esp/esp-idf/components/newlib/newlib.lf      /Users/poe/esp/esp-idf/components/app_trace/linker.lf /Users/poe/esp/esp-idf/components/esp_gdbstub/linker.lf --input /Users/poe/esp/esp-idf/components/esp32/ld/esp32.project.ld.in --output /Users/poe/Desktop/VSC-Workspace/hello_world/build/esp-idf/esp32/ld/esp32.project.ld --kconfig /Users/poe/esp/esp-idf/Kconfig --env-file /Users/poe/Desktop/VSC-Workspace/hello_world/build/config.env --libraries-file /Users/poe/Desktop/VSC-Workspace/hello_world/build/ldgen_libraries --objdump /usr/bin/objdump
Traceback (most recent call last):
  File "/Users/poe/esp/esp-idf-tools/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/poe/esp/esp-idf-tools/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 2622, in parseImpl
    if (instring[loc] == self.firstMatchChar and
IndexError: string index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... cut ...]
    raise ParseException("Parsing sections info for library " + sections_info_dump.name + " failed. " + p.message)
  File "/Users/poe/esp/esp-idf-tools/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 272, in __getattr__
    raise AttributeError(aname)
AttributeError: message
ninja: build stopped: subcommand failed.
Процесс терминала "/bin/zsh '-c', 'cmake --build .'" был завершен с кодом выхода 1.

Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.
```

brianignacio5, 2020-12-09:
This seems to be ESP-IDF issue. Have you tried https://github.com/espressif/esp-idf/issues/5996 ?

Oleg-Perevyshin, 2020-12-09:
I moved the project to another folder and deleted the directory "build".
I don't know what the reason was.

closed 2020-12-09 by Oleg-Perevyshin