espressif/vscode-esp-idf-extension #230: [Bug Report]:

ViktorTarasovA, 2020-12-11:
**Describe the bug**
I try to build the project in VSCode with help of the extension. The project is built successfully in ESP-IDF Command Prompt but I can't do it in VSCode regarding to linker error (see Logs).

**To Reproduce**
Just press F1 and start building. Interesting note: sometimes it says that the build process was successfully finished.

**Expected behavior**
Built binary.

**Environment:**

- OS Name & Version: Win10
- VSCode Version: 1.51.1
- ESP-IDF Version: 4.1
- Python Version: 3.7.9
- board: wrover kit

**Logs**
During building:
```
linker script generation failed for D:/Viktor/software/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: failed to parse D:/Viktor/software/esp-idf/components/esp_event/linker.lf
required keys ['entries'] for fragment not found (at char 41), (line:2, col:5)
```
<details>
  <summary>Extension log:</summary>
  
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:36:26.153Z"}
{"message":"Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:36:33.690Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:36:33.691Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:57:01.007Z"}
{"message":"Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.9_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n","stack":"Error: Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.9_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:57:12.781Z"}
{"message":"Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.9_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n","stack":"Error: Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.9_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T13:57:12.782Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T14:04:24.223Z"}
{"message":"Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n","stack":"Error: Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T14:04:32.933Z"}
{"message":"Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n","stack":"Error: Command failed: \"D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe\" \"d:\\Viktor\\software\\esp-idf\\tools\\check_python_dependencies.py\" -r \"c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\esp_debug_adapter\\requirements.txt\"\n\n\tat ChildProcess.exithandler (child_process.js:304:12)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T14:04:32.933Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\viktar\\.vscode\\extensions\\espressif.esp-idf-extension-0.5.1\\dist\\extension.js:1:6763\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2020-12-10T14:06:04.604Z"}
[... cut ...]
  "idf.customExtraVars": "D:\\Viktor\\software\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20191114\\openocd-esp32\\share\\openocd\\scripts\\"}",
  "idf.pythonBinPath": "D:\\Viktor\\software\\.espressif\\python_env\\idf4.1_py3.7_env\\Scripts\\python.exe",
  "idf.port": "COM9",
  "idf.openOcdConfigs": [
      "interface\\ftdi\\esp32_devkitj_v1.cfg",
      "board\\esp32-wrover.cfg"
  ],  
</details>

**Additional context**
It works in ESP-IDF Command Prompt. Is it a bug or just wrong configuration?

brianignacio5, 2020-12-14:
Please change 
```
"idf.customExtraVars": "D:\Viktor\software\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\share\openocd\scripts\"}"
```
to
```
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"D:\Viktor\software\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\share\openocd\scripts\"}"
```

ViktorTarasovA, 2020-12-14:
Thank you, @brianignacio5. Changed this line and installed some python requirements. It works. I will be more careful in future. The issue can be closed.

closed 2020-12-14 by ViktorTarasovA

ViktorTarasovA, 2020-12-14:
@brianignacio5 To be honest, `"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"D://Viktor/software/.espressif/tools/openocd-esp32/v0.10.0-esp32-20191114/openocd-esp32/share/openocd/scripts\"}",`
works better :)