espressif/vscode-esp-idf-extension #236: VS Code Buttons do not work

MartyMcFlyInTheSky, 2020-12-15:
Hi guys

My VS Code Plugin buttons do not work properly. I have to use idf.py from the command line all the time.

When I click on "build" for example, it just displays the window on the top left, indicating that it is building. But this task just runs indefinitely without ever accomplishing something. The other buttons behave like this as well. The only one that is working i suppose is the "select device port" option, which let's me specify on what COM port i want to flash.

```
C:\dev\30_https_ota>idf.py --version
ESP-IDF v4.3-dev-1901-g178b122c1
```

[image]

I'm using Win10 64bit if that matters.

brianignacio5, 2020-12-16:
Have you configure the extension using the `ESP-IDF: Configure ESP-IDF extension` ?

See https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md#how-to-configure-this-extension-from-zero-using-esp-idf-configure-esp-idf-extension-window

MartyMcFlyInTheSky, 2020-12-17:
Hey @brianignacio5 thanks for the answer. I did install it using the configuration. So the VS intern variables should be set correctly. I also figured out that the monitoring button works just fine.

In any case, what values would I have to change and where to make the buttons trigger the right functions (with the right location)?

brianignacio5, 2020-12-18:
Please share this information to understand the issue in your system.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

Also please share your extension settings values

`idf.espIdfPathWin`
`idf.customExtraPaths`
`idf.pythonBinPathWin`
`idf.customExtraVars`

There should also be a Build or Flash task output log in the terminal tab.

brianignacio5, 2020-12-23:
Closing this issue since it is configuration error.

Explanation of extension configuration is shown in [ONBOARDING](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md)

MartyMcFlyInTheSky, 2022-02-09:
Hey @brianignacio5 can you reopen this issue? The bug still persists. Here's all the information you need:

esp_idf_vsc_ext.log
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/8031640/esp_idf_vsc_ext.log)

Here are all idf related settings in local folder settings.json:

```
{
    ...
    "idf.adapterTargetName": "esp32",
    "idf.espIdfPath": "${env:IDF_PATH}",
    "idf.toolsPathWin": "${env:IDF_TOOLS_PATH}",
    "idf.customExtraPaths": "C:/dev/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin;C:/dev/esp/.espressif/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin;C:/dev/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin;C:/dev/esp/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin;C:/dev/esp/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin;C:/dev/esp/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin;C:/dev/esp/.espressif/tools/cmake/3.16.4/bin;C:/dev/esp/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin;C:/dev/esp/.espressif/tools/ninja/1.10.2;C:/dev/esp/.espressif/tools/idf-exe/1.0.1;C:/dev/esp/.espressif/tools/ccache/3.7;C:/dev/esp/.espressif/tools/dfu-util/0.9/dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:/dev/esp/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.pythonBinPathWin": "C:/dev/esp/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM5",
}
```

Note that I have set up the IDF_PATH and IDF_TOOLS_PATH variables in environment before installing the extension (so the installer finds these locations by itself and installed it there - that worked fine btw.).

The output from the terminal is just:

```
> Executing task in folder aurora-light-controller: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

Setting IDF_PATH: C:\dev\esp\esp-idf

Adding ESP-IDF tools to PATH...
    C:\dev\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin
    C:\dev\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin
    C:\dev\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin
    C:\dev\esp\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin      
    C:\dev\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin
    C:\dev\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\dev\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\dev\esp\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\dev\aurora-light-controller\build>
```

So essentially, export.bat is run without follow up. Judging from the esp_idf_vsc_ext.log it seems however that the paths are still not found after exporting? What could be the cause?

brianignacio5, 2022-02-09:
What about the doctor command output ?

MartyMcFlyInTheSky, 2022-02-09:
Thanks! This is the output from the doctor command:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 C:\dev\esp\env 
System environment variable PATH 
 C:\Program Files (x86)\gtkmm\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\dev\esp\esp-idf;C:\PHP7;C:\Program Files\PuTTY\;C:\Go\bin;C:\Users\samue\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files\Git\cmd;C:\Users\samue\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\samue\AppData\Local\Programs\Python\Python38\;C:\Users\samue\AppData\Local\Programs\Python\Python37\;C:\Users\samue\appdata\local\programs\python\python37\;C:\Users\samue\appdata\local\programs\python\python37\scripts\;C:\Users\samue\AppData\Local\Programs\Python\Python39\;C:\Users\samue\appdata\local\programs\python\python39\scripts\;C:\Users\samue\appdata\local\programs\python\python39\;C:\Users\samue\AppData\Local\Programs\Python\Python36\;C:\Users\samue\appdata\local\programs\python\python36\scripts\;C:\Users\samue\appdata\local\programs\python\python36\;C:\Users\samue\AppData\Local\Programs\Python\Python35\;C:\Users\samue\appdata\local\programs\python\python35\scripts\;C:\Users\samue\appdata\local\programs\python\python35\;C:\Users\samue\appdata\local\programs\python\python35\scripts\;C:\Users\samue\appdata\local\programs\python\python35\;C:\Users\samue\AppData\Local\Microsoft\WindowsApps;C:\Users\samue\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\samue\AppData\Roaming\npm;C:\dev\esp\.espressif\tools\cmake\3.20.3\bin;C:\Go;C:\Sencha\Cmd;C:\Sencha;C:\Users\samue\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\msys64\mingw64\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.64.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.3.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\dev\esp\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\dev\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\dev\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\dev\esp\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\dev\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\dev\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\dev\esp\.espressif\tools\cmake\3.16.4\bin;C:\dev\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\dev\esp\.espressif\tools\ninja\1.10.2;C:\dev\esp\.espressif\tools\idf-exe\1.0.1;C:\dev\esp\.espressif\tools\ccache\3.7;C:\dev\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\dev\\esp\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\dev\esp\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: false
Access to C:\dev\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: false
Access to C:\dev\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: false
Access to C:\dev\esp\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: false
Access to C:\dev\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\dev\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\dev\esp\.espressif\tools\cmake\3.16.4\bin: false
Access to C:\dev\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: false
Access to C:\dev\esp\.espressif\tools\ninja\1.10.2: true
Access to C:\dev\esp\.espressif\tools\idf-exe\1.0.1: true
Access to C:\dev\esp\.espressif\tools\ccache\3.7: false
Access to C:\dev\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
[... cut ...]
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\dev\ritune\docs\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

```

I changed it to "idf.espIdfPathWin" and put the idf.notificationSilentMode: false - Option in there but still the same. The process does however open a new terminal window.

brianignacio5, 2022-02-09:
As you can see the values being used are not working. The `idf.pythonBinPathWin` should be `C:/dev/esp/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe` but is showing to be `python` and `idf.customExtraPaths` is using different tools than the one from your terminal output.

Seems your current vscode window is not picking the right values.
Do you open your project as a workspace or a single folder ?

MartyMcFlyInTheSky, 2022-02-09:
@brianignacio5 Thanks yes I can see that that is rather strange. Also, it doesn't seem to recognise ninja so no wonder it can't build. I'm always opening my project via workspace.

brianignacio5, 2022-02-16:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/165231474.zip) ? 

I made some changes with regards to configuration settings reading in #641

brianignacio5, 2022-02-21:
@MartyMcFlyInTheSky Did it fixed the issue ?

brianignacio5, 2022-02-25:
Closing this issue for lack of response, let me know if you have any issues.

closed 2022-02-25 by brianignacio5

jrive87, 2024-01-22:
I'm having the same issues that were reported.  I can build, flash, monitor the project from the terminal, but none of the buttons works.  I seemingly just get terminal intialization, which ends in 👍 

> 
Checking if Python packages are up to date...
Constraint file: c:/users/jeriv/.espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Users\jeriv\esp\esp-idf\tools\requirements\requirements.core.txt
Python being checked: c:\users\jeriv\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

my settings.json has the following configuration settings:
```
"editor.minimap.enabled": false,
    "windows.zoomlevel": 0,
    "workbench.iconTheme": "vscode-icons",
    "editor.fontsize": 22,
    "editor.fontligatures": null,
    "idf.gitPathWin": "C:/Users/jeriv/esp/tools/idf-git/2.43.0/cmd/git.exe",
    "idf.espIdfPathWin": "C:\\Users\\jeriv\\esp\\esp-idf",
    "idf.pythonBinPathWin": "c:\\users\\jeriv\\.espressif\\python_env\\idf5.1_py3.11_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "c:/users/jeriv/.espressif",
    "idf.customExtraPaths": "c:\\users\\jeriv\\.espressif\\tools\\xtensa-esp-elf-gdb\\12.1_20221002\\xtensa-esp-elf-gdb\\bin;c:\\users\\jeriv\\.espressif\\tools\\riscv32-esp-elf-gdb\\12.1_20221002\\riscv32-esp-elf-gdb\\bin;c:\\users\\jeriv\\.espressif\\tools\\xtensa-esp32-elf\\esp-12.2.0_20230208\\xtensa-esp32-elf\\bin;c:\\users\\jeriv\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-12.2.0_20230208\\xtensa-esp32s2-elf\\bin;c:\\users\\jeriv\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-12.2.0_20230208\\xtensa-esp32s3-elf\\bin;c:\\users\\jeriv\\.espressif\\tools\\riscv32-esp-elf\\esp-12.2.0_20230208\\riscv32-esp-elf\\bin;c:\\users\\jeriv\\.espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;c:\\users\\jeriv\\.espressif\\tools\\cmake\\3.24.0\\bin;c:\\users\\jeriv\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230921\\openocd-esp32\\bin;c:\\users\\jeriv\\.espressif\\tools\\ninja\\1.10.2;c:\\users\\jeriv\\.espressif\\tools\\idf-exe\\1.0.3;c:\\users\\jeriv\\.espressif\\tools\\ccache\\4.8\\ccache-4.8-windows-x86_64;c:\\users\\jeriv\\.espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;c:\\users\\jeriv\\.espressif\\tools\\esp-rom-elfs\\20230320",
    "idf.customExtraVars": {
        "OPENOCD_SCRIPTS": "c:\\users\\jeriv\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts",
        "IDF_CCACHE_ENABLE": "1",
        "ESP_ROM_ELF_DIR": "c:\\users\\jeriv\\.espressif\\tools\\esp-rom-elfs\\20230320/"
    },
    "git.openRepositoryInParentFolders": "always",
    "cmake.showOptionsMovedNotification": false,
}
```
I'm new to all this...where exactly are the buttons configured?
Where do I look for the files you requested from @MartyMcFlyInTheSky?
What is the doctor command?

brianignacio5, 2024-01-25:
Need to check the extension log output located in: 

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Please review the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting).

rafalwrob, 2024-05-05:
Hi, do you guys managed to solve this problem? I'm struggling with the same thing. My settings:"
```
{
  "workbench.colorTheme": "Monokai",
  "[html]": {
    "editor.defaultFormatter": "esbenp.prettier-vscode"
  },
  "editor.formatOnSave": true,
  "explorer.confirmDragAndDrop": false,
  "explorer.confirmDelete": false,
  "[typescriptreact]": {
    "editor.defaultFormatter": "esbenp.prettier-vscode"
  },
  "[vue]": {
    "editor.defaultFormatter": "esbenp.prettier-vscode"
  },
  "workbench.iconTheme": "material-icon-theme",
  "editor.fontSize": 15,
  "security.workspace.trust.untrustedFiles": "open",
  "svelte.enable-ts-plugin": true,
  "window.zoomLevel": 1,
  "github.copilot.enable": {
    "*": true,
    "plaintext": false,
    "markdown": false,
    "scminput": false
  },
  "terminal.integrated.defaultProfile.windows": "Command Prompt",
  "terminal.integrated.profiles.windows": {
    "PowerShell": {
      "source": "PowerShell",
      "icon": "terminal-powershell"
    },
    "Command Prompt": {
      "path": [
        "${env:windir}\\Sysnative\\cmd.exe",
        "${env:windir}\\System32\\cmd.exe"
      ],
      "args": [
        "/k",
        "C:/Espressif/frameworks/esp-idf-v5.2.1/export.bat esp-idf-95fc7f73a381402d241e04a22ce81be3"
      ],
      "icon": "terminal-cmd"
    },
    "Git Bash": {
      "source": "Git Bash"
    }
  },
  "cmake.showOptionsMovedNotification": false,
  "git.openRepositoryInParentFolders": "never",
  "idf.espIdfPathWin": "C:/Espressif/frameworks/esp-idf-v5.2.1/",
  "idf.pythonBinPathWin": "C:/Espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe",
  "idf.toolsPathWin": "C:\\Espressif",
  "idf.customExtraPaths": "C:\\Espressif\\tools\\xtensa-esp-elf-gdb\\12.1_20231023\\xtensa-esp-elf-gdb\\bin;C:\\Espressif\\tools\\riscv32-esp-elf-gdb\\12.1_20231023\\riscv32-esp-elf-gdb\\bin;C:\\Espressif\\tools\\xtensa-esp-elf\\esp-13.2.0_20230928\\xtensa-esp-elf\\bin;C:\\Espressif\\tools\\riscv32-esp-elf\\esp-13.2.0_20230928\\riscv32-esp-elf\\bin;C:\\Espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;C:\\Espressif\\tools\\cmake\\3.24.0\\bin;C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230921\\openocd-esp32\\bin;C:\\Espressif\\tools\\ninja\\1.11.1;C:\\Espressif\\tools\\idf-exe\\1.0.3;C:\\Espressif\\tools\\ccache\\4.8\\ccache-4.8-windows-x86_64;C:\\Espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;C:\\Espressif\\tools\\esp-rom-elfs\\20230320",
  "idf.customExtraVars": {
    "OPENOCD_SCRIPTS": "C:\\Espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts",
    "IDF_CCACHE_ENABLE": "1",
    "ESP_ROM_ELF_DIR": "C:\\Espressif\\tools\\esp-rom-elfs\\20230320/"
  },
  "idf.gitPathWin": "C:/Espressif/tools/idf-git/2.43.0/cmd/git.exe",
  "idf.ninjaArgs": []
}
```
The logs: 
```
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1397142\n\tat ChildProcess.exithandler (node:child_process:415:7)\n\tat ChildProcess.emit (node:events:517:28)\n\tat maybeClose (node:internal/child_process:1098:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:303:5)","level":"error","timestamp":"2024-05-05T19:44:50.557Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at \\dist\\extension.js:2:1397142\n    at ChildProcess.exithandler (node:child_process:415:7)\n    at ChildProcess.emit (node:events:517:28)\n    at maybeClose (node:internal/child_process:1098:16)\n    at Process.ChildProcess._handle.onexit (node:internal/child_process:303:5)","level":"error","timestamp":"2024-05-05T19:44:57.770Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at \\dist\\extension.js:2:1397142\n    at ChildProcess.exithandler (node:child_process:415:7)\n    at ChildProcess.emit (node:events:517:28)\n    at maybeClose (node:internal/child_process:1098:16)\n    at Process.ChildProcess._handle.onexit (node:internal/child_process:303:5)","level":"error","timestamp":"2024-05-05T19:44:58.383Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at \\dist\\extension.js:2:1397142\n    at ChildProcess.exithandler (node:child_process:415:7)\n    at ChildProcess.emit (node:events:517:28)\n    at maybeClose (node:internal/child_process:1098:16)\n    at Process.ChildProcess._handle.onex [... cut ...]
Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.2.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.2.1\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\dev\ESP32\hello_world>
```
The build terminal don't show up. I will be grateful for any advice.

brianignacio5, 2024-05-15:
I feel the issue is that you configured your terminal `terminal.integrated.profiles.windows` to run the export.sh so any task in the extension that uses a shell command (like the build task) will not work.