espressif/vscode-esp-idf-extension #237: [Bug Report]: Failed building wheel for cryptography

kayleejacob1, 2020-12-16:
Below is log that i am getting this error from i have done the upgrade command it says upgrade successful
i also added the environment variables 

  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\Home\.espressif\python_env\idf4.1_py3.7_env\include -IC:\Users\Home\AppData\Local\Programs\Python\Python37\include -IC:\Users\Home\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.25.28610\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /Tcbuild\temp.win-amd64-3.7\Release\_openssl.c /Fobuild\temp.win-amd64-3.7\Release\build\temp.win-amd64-3.7\Release\_openssl.obj
  _openssl.c
  build\temp.win-amd64-3.7\Release\_openssl.c(33): fatal error C1083: Cannot open include file: 'pyconfig.h': No such file or directory
  
      =============================DEBUG ASSISTANCE=============================
      If you are seeing a compilation error please try the following steps to
      successfully install cryptography:
      1) Upgrade to the latest pip and try again. This will fix errors for most
         users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
      2) Read https://cryptography.io/en/latest/installation.html for specific
         instructions for your platform.
      3) Check our frequently asked questions for more information:
         https://cryptography.io/en/latest/faq.html
      =============================DEBUG ASSISTANCE=============================
  
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.25.28610\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 19.2.3, however version 20.3.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

The following paths should be added to env PATH
.;C:\Users\Home\AppData\Local\Programs\Python\Python37;C:\Users\Home\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin;C:\Users\Home\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin;C:\Users\Home\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Home\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Home\.espressif\tools\cmake\3.13.4\bin;C:\Users\Home\.espressif\tools\openocd-esp32\v0.10.0-esp32-20191114\openocd-esp32\bin;C:\Users\Home\.espressif\tools\ninja\1.9.0;C:\Users\Home\.espressif\tools\idf-exe\1.0.1;C:\Users\Home\.espressif\tools\ccache\3.7

kayleejacob1, 2020-12-16:
C:\Users\Home>python -m pip install --upgrade pip
Requirement already satisfied: pip in c:\users\home\appdata\local\programs\python\python37\lib\site-packages (20.3.3)

ihouses, 2020-12-21:
Hi! Check if before this error you get some missing .h error. If this is the case probably you need to update your vs_build_tools: https://devblogs.microsoft.com/cppblog/introducing-the-visual-studio-build-tools/

As I say if you get an error of a missing .H you can google it since probably will be the primary cause of the problem.

brianignacio5, 2020-12-23:
Closing this issue since doesn't seem to be related to this extension.

closed 2020-12-23 by brianignacio5