espressif/vscode-esp-idf-extension #241: Cannot launch SDK GUI configuration in VS Code. PermissionError: [Errno 13]

bonmotwang, 2020-12-20:
I just got VS Code and esp-idf extension set up on my laptop.
Build was fine, Flash was fine, Monitor was fine.
Setting Serial Port was fine. Beautiful!

But when I try to use SDKCONFIG GUI. I got PermissionError: [Errno 13].

The whole output is here:
Tried run VS Code as administrator, tried to set cmake.exe to administrator. Didn't work.

I like this extension a lot. Please help this SDKCONFIG GUI issue. I know I probably only use it once in the whole project life cycle, but still want to have it.
Thanks

`Python requirements from C:\Users\Paul\esp\esp-idf\requirements.txt are satisfied.

[0/1] cmd.exe /C "cd /D 
C:\Projects\ict\blinky\blink\build\esp-idf && C:\Users\Paul\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E env "COMPONENT_KCONFIGS= C:/Users/Paul/esp/esp-idf/components/app_trace/Kconfig C:/Users/Paul/esp/esp-idf/components/bt/Kconfig C:/Users/Paul/esp/esp-idf/components/driver/Kconfig C:/Users/Paul/esp/esp-idf/components/efuse/Kconfig C:/Users/Paul/esp/esp-idf/components/esp32/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_adc_cal/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_event/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_http_client/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_http_server/Kconfig C:/User
s/Paul/esp/esp-idf/components/esp_https_ota/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_wifi/Kconfig C:/Users/Paul/esp/esp-idf/components/espcoredump/Kconfig C:/Users/Paul/esp/esp-idf/components/ethernet/Kconfig C:/Users/Paul/esp/esp-idf/components/fatfs/Kconfig C:/Users/Paul/esp/esp-idf/
components/freemodbus/Kconfig C:/Users/Paul/esp/esp-idf/components/freertos/Kconfig C:/Users/Paul/esp/esp-idf/components/heap/Kconfig C:/Users/Paul/esp/esp-idf/components
/libsodium/Kconfig C:/Users/Paul/esp/esp-idf/components/log/Kconfig C:/Users/Paul/esp/esp-idf/components/lwip/Kconfig C:/Users/Paul/esp/esp-idf/component
s/mbedtls/Kconfig C:/Users/Paul/esp/esp-idf/components/mdns/Kconfig C:/Users/Paul/esp/esp-idf/components/mqtt/Kconfig C:/Users/Paul/esp/esp-idf/components/nvs_flash/Kcon
fig C:/Users/Paul/esp/esp-idf/components/openssl/Kconfig C:/Users/Paul/esp/esp-idf/components/pthread/Kconfig C:/Users/Paul/esp/esp-idf/components/spi_flash/Kconf
ig C:/Users/Paul/esp/esp-idf/components/spiffs/Kconfig C:/Users/Paul/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/Paul/esp/esp-idf/components/unity/Kconfig C:/Users/Paul
/esp/esp-idf/components/vfs/Kconfig C:/Users/Paul/esp/esp-idf/components/wear_levelling/Kconfig" "COMPONENT_KCONFIGS_PROJBUILD= C:/Users/Paul/esp/esp-idf/components/app_update/Kconfig.pro
jbuild C:/Users/Paul/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/Paul/esp/esp-idf/components/esptool_py/Kconfig.projbuild C:/Projects/ict/blinky/blink/main/Kconfig.projbuild C:/Users/Pau
l/esp/esp-idf/components/partition_table/Kconfig.projbuild" C:\Users\Paul\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe C:/Users/Paul/esp/esp-idf/tools/kconfig_new/confserver.py --kconfig C:/Users/Paul/esp/esp-idf/Kconfig --config C:/Projects/ict/blinky/blink/sdkconfig"

---------------------------ERROR--------------------------

Traceback (most recent call last):
  File "C:/Users/Paul/esp/esp-idf/tools/kconfig_new/confserver.py", line 267, in <module>
    main()
  File "C:/Users/Paul/esp/esp-idf/tools/kconfig_new/confserver.py", line 57, in main
    run_server(args.kconfig, args.config)
  File "C:/Users/Paul/esp/esp-idf/tools/kconfig_new/confserver.py", line 68, in run_server
    deprecated_options.replace(sdkconfig_in=f_o.name, sdkconfig_out=sdkconfig)
  File "C:\Users\Paul\esp\esp-idf\tools\kconfig_new\confgen.py", line 91, in replace
    with open(sdkconfig_in, 'r') as f_in, open(sdkconfig_out, 'w') as f_out:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\Paul\\AppData\\Local\\Temp\\tmpahq_rpxs'

-----------------------END OF ERROR-----------------------
FAILED: esp-idf/CMakeFiles/confserver 
cmd.exe /C "cd /D C:\Projects\ict\blinky\blink\build\esp-idf && C:\Users\Paul\.espressif\tools\cmake\3.13.4\bin\cmake.exe -E env "COMPONENT_KCONFIGS= C:/Users/Paul/esp/esp-idf/components/app_trace/Kconfig C:/Users/Paul/esp/esp-idf/components/bt/Kconfig C:/Users/Paul/esp/esp-idf/components/driver/Kconfig C:/Users/Paul/esp/esp-idf/components/efuse/Kconfig C:/Users/Paul/esp/esp-idf/components/esp32/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_adc_cal/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_event/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_http_client/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_http_server/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_https_ota/Kconfig C:/Users/Paul/esp/esp-idf/components/esp_wifi/Kconfig C:/Users/Paul/esp/esp-idf/components/espcoredump/Kconfig C:/Users/Paul/esp/esp-idf/components/ethernet/Kconfig C:/Users/Paul/esp/esp-idf/components/fatfs/Kconfig C:/Users/Paul/esp/esp-idf/components/freemodbus/Kconfig C:/Users/Paul/esp/esp-idf/components/freertos/Kconfig C:/Users/Paul/esp/esp-idf/components/heap/Kconfig C:/Users/Paul/esp/esp-idf/components/libsodium/Kconfig C:/Users/Paul/esp/esp-idf/components/log/Kconfig C:/Users/Paul/esp/esp-idf/components/lwip/Kconfig C:/Users/Paul/esp/esp-idf/components/mbedtls/Kconfig C:/Users/Paul/esp/esp-idf/components/mdns/Kconfig C:/Users/Paul/esp/esp-idf/components/mqtt/Kconfig C:/Users/Paul/esp/esp-idf/components/nvs_flash/Kconfig C:/Users/Paul/esp/esp-idf/components/openssl/Kconfig C:/Users/Paul/esp/esp-idf/components/pthread/Kconfig C:/Users/Paul/esp/esp-idf/components/spi_flash/Kconfig C:/Users/Paul/esp/esp-idf/components/spiffs/Kconfig C:/Users/Paul/esp/esp-idf/components/tcpip_adapter/Kconfig C:/Users/Paul/esp/esp-idf/components/unity/Kconfig C:/Users/Paul/esp/esp-idf/components/vfs/Kconfig C:/Users/Paul/esp/esp-idf/components/wear_levelling/Kconfig" "COMPONENT_KCONFIGS_PROJBUILD= C:/Users/Paul/esp/esp-idf/components/app_update/Kconfig.projbuild C:/Users/Paul/esp/esp-idf/components/bootloader/Kconfig.projbuild C:/Users/Paul/esp/esp-idf/components/esptool_py/Kconfig.projbuild C:/Projects/ict/blinky/blink/main/Kconfig.projbuild C:/Users/Paul/esp/esp-idf/components/partition_table/Kconfig.projbuild" C:\Users\Paul\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe C:/Users/Paul/esp/esp-idf/tools/kconfig_new/confserver.py --kconfig C:/Users/Paul/esp/esp-idf/Kconfig --config C:/Projects/ict/blinky/blink/sdkconfig"
ninja: build stopped: subcommand failed.

Note: You are using Python 3.8.6. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Running ninja in directory c:\projects\ict\blinky\blink\build
Executing "ninja confserver"...
ninja failed with exit code 1

---------------------------ERROR--------------------------

Received signal: null with code: 2
-----------------------END OF ERROR-----------------------
`

brianignacio5, 2020-12-21:
There is a temp file created `'C:\Users\Paul\AppData\Local\Temp\tmpahq_rpxs` which have permission error. Do you have permission for temp folder ?

bonmotwang, 2020-12-22:
Yes, I tried everything I can. It didn't work.
But good news is I solved the problem by upgrading ESP-IDF to 4.2, tools chain to 8.4.
It was: ESP-IDF 4.0, tools chain 8.2.
I think the latest plug in is not compatible to the older versions.

closed 2020-12-22 by bonmotwang

brianignacio5, 2020-12-22:
There is some fixes in latest ESP-IDF related to `$IDF_PATH/tools/confserver.py` probably related to this.