espressif/vscode-esp-idf-extension #242: [Feature Request]:

ihouses, 2020-12-20:
I have installed my ESP-ADF framework as described in the official documentation. 
Would be great to be able to create as well the examples in the ESP-ADF framework when is available.

Is there any plan for supporting that?

brianignacio5, 2020-12-21:
We added an option to the `ESP-IDF: Show ESP-IDF Examples Projects` command to choose between ESP-IDF, ESP-ADF and ESP-MDF. For this to work, you should set the `idf.espAdfPath` configuration setting in Visual Studio Code  to your $ADF_PATH directory

ihouses, 2020-12-21:
It works!!

This is a really great tool! I was trying platformio as well that was looking very promising but without too much success, one the other hand this tools is exactly what we need.

closed 2020-12-21 by ihouses