espressif/vscode-esp-idf-extension #243: [Bug Report]:

ViktorTarasovA, 2020-12-21:
**Describe the bug**
Pressing F5 in VSCode to start debugging results in an error message:
`TypeError: __init__() got an unexpected keyword argument 'gdb_path'`
The same issue was described in:
https://bbs.esp32.com/viewtopic.php?f=40&p=65225
I even successfully fixed this issue by installing appropriate pygdbmi

**Environment (please complete the following information):**

- OS Name & Version: Win10 64bit
- VSCode Version: 1.52.1 (user setup), Commit: ea3859d4ba2f3e577a159bc91e3074c5d85c0523
- ESP-IDF Version: v4.1
- Python Version: 3.7.9

**Additional context**
Is it possible to somehow workaround it in a better way for future users?

brianignacio5, 2020-12-22:
Can you please your extensions settings and extension log ?
Is the extension properly configured ? 
There should be a `ESP-IDF Debug Adapter` output showing you the execution of the debug adapter and its arguments.

brianignacio5, 2020-12-24:
The ESP-IDF Debug Adapter `requirements.txt` pygdbmi version is updated. By configuring the extension using steps on [ONBOARDING](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) should work fine.

ViktorTarasovA, 2020-12-24:
> The ESP-IDF Debug Adapter `requirements.txt` pygdbmi version is updated. By configuring the extension using steps on [ONBOARDING](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) should work fine.

Thank you!
To be honest, I'm not sure that I will find time to install old pygdbmi version again and collect logs and configurations. So I would prefer to close the issue.

closed 2020-12-24 by ViktorTarasovA