espressif/vscode-esp-idf-extension #247: [Feature Request]: Is there clean button?

ihouses, 2020-12-21:
I don't find the "clean" button in the VS gui. I found out that with the terminal an typing idf.py fullclean is doing the same.

My problem cames that after a unsuccessful build I can not execute that command since I get this message:

[image]

If I close virual studio and open again then I can do a clean with idf.py fullclean.

brianignacio5, 2020-12-22:
I'm sorry for some reason I can't seem to see the attached picture. Will you please upload again ? 

Please follow the issue template, share your system environment details and extension log for us to understand your problem better.

bonmotwang, 2020-12-22:
I was looking for this "button" last night. But I just did it in the TERMINAL.

PR #250 (2020-12-30): Full clean button and command
Add `espIdf.fullClean` command to delete the build directory and status bar button.

Fix #247
closed 2020-12-30 by brianignacio5