espressif/vscode-esp-idf-extension #249: Clicking the build or flash icon button at the left bottom of VS CODE slows down the computer. CPU time 100% on all 4 cores!

bonmotwang, 2020-12-22:
I have been using command line with cmake. all good.
Installed this plug in, very nice, all good.

I like those short cut buttons at the bottom.
But I noticed that build and flash become very very slow.
For example: 
Click the build button, it occupies 100% of my CPU on all 4 cores. Cannot browse the web pages at the same time.
But if I just type in idf.py build in the terminal, no problem, I can multi task.

Click flash button, it take 40 seconds to download, but if type in idf.py flash, it only take 12 seconds.

I guess there is some kind of "wrapper" slowing it down?

I hope it can be fixed. But for now, keep typing in the terminal. LOL

Thanks

Paul

brianignacio5, 2020-12-23:
Can you describe a bit more about your environment setup ?

Well basically we call vscode's task api which launch a shell terminal with the CMake commands. Maybe you can specify how you run terminal commands and see the difference. The flash speed might be related to configured flash baud rate in the extension.

brianignacio5, 2021-01-06:
Can anyone reproduce this and show steps ? Such scenario I can't reproduce.

closed 2021-01-06 by brianignacio5