espressif/vscode-esp-idf-extension #251: [Bug Report]: C:\Users\make.espressif\python_env\idf4.1_py3.9_env\Scripts\pythone.exe is not accesible.

kaibin1111, 2020-12-24:
C:\Users\make.espressif\python_env\idf4.1_py3.9_env\Scripts\pythone.exe is not accesible.

mitchjs, 2020-12-24:
python**e**.exe

there is an extra e there...

kaibin1111, 2020-12-25:
> python**e**.exe
> 
> there is an extra e there... at then end of "python"
> should be "python.exe"

C:\Users\make.espressif\python_env\idf4.1_py3.9_env\Scripts\python.exe is not accesible.

mitchjs, 2020-12-25:
is that file there? did you look? could be permissions issue... to me thats a strange place for the file to be
example mine at:

"C:\Users\mitchjs\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe"

kaibin1111, 2020-12-25:
> is that file there? did you look? could be permissions issue... to me thats a strange place for the file to be
> example mine at:
> 
> "C:\Users\mitchjs.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe"

The files are there

pwmb, 2020-12-25:
You can try with python 3.8, also please use python from official site and not any other distribution.

Also could be permission issue, as mentioned earlier.

mitchjs, 2020-12-25:
can you execute the python.exe by going the folder in explorer

i just noticed... that i bet there forum is changing the path... "\mitch.espressif\"
really is
`\mitchjs\.espressif\`

pwmb, 2020-12-26:
> can you execute the python.exe by going the folder in explorer
> 
> i just noticed... that i bet there forum is changing the path... "\mitch.espressif"
> really is
> `\mitchjs\.espressif\`

@mitchjs You mean there is a missing `\` between `user directory path` and `.espressif` ?

mitchjs, 2020-12-26:
its there, just when you paste the path into the github comments it removed it
so... there is no bug!

im not sure if @kaibin1111 still having permission issues

brianignacio5, 2021-01-04:
The original issue seems to be a user access error. Make sure that `idf.pythonBinPathWin` has executable access.

Title is not related to original issue, doesn't seem to be an issue and can't reproduce it.

closed 2021-01-04 by brianignacio5