espressif/vscode-esp-idf-extension #252: [Feature Request]: Is it intelliSense/ctrl+click supported in the IDF/ADF structure model?

ihouses, 2020-12-25:
I am not sure if the intelliSense plugin in in charge of this but: I see that when I add a module in my project I can not use ctrl+click to jump to the definition.

As well many of my includes can not be found by (intelliSense?). 

I mean, this are details that usually we give by granted since in many programming IDEs is working by default. Do we need to do/install something else in order to enable that functionality?

Thanks!

brianignacio5, 2020-12-25:
Have you look into [C_CPP_CONFIGURATION.md](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ?

ihouses, 2020-12-25:
I have used the information in  C_CPP_CONFIGURATION.md  and now all headers are detected and by doubts are solved, thanks so much.

closed 2020-12-26 by pwmb