espressif/vscode-esp-idf-extension #253: [Bug Report]: Setting placed in ${workspaceFolder}/.vscode/settings.json are ignored

JimGaluska, 2020-12-29:
**Bug Description**
The esp-idf extension ignores settings placed in the workspace's settings.json file.  

**To Reproduce**
1. Open the user's setting file (e.g. $HOME/.config/Code/User/settings.json) and delete any entry for: ```idf.showOnboardingOnInit```
2. Create a blank folder that will become the folder: ${workspaceFolder}
3. Open vscode. 
4. From the Files menu, Open the folder created in step 2
5. Open the Command Palette (Ctrl+Shift+P) and select ```ESP-IDF: Add vscode configuration Folder``` 
6. Open the newly created file ```.vscode/settings.json``` file
7. Add the following to the the settings.json file: ```"idf.showOnboardingOnInit": false``` as shown below:
```JSON
{
  "C_Cpp.clang_format_style": "Visual Studio",
  "editor.formatOnSave": false,
  "[cpp]": {
    "editor.quickSuggestions": true
  },
  "[c]": {
    "editor.quickSuggestions": true
  },
  "C_Cpp.intelliSenseEngineFallback": "Enabled",
  "idf.showOnboardingOnInit": false
}
```
8. Save the settings.json file
9. From the File menu, select "Save Workspace As...". Save using the default workspace name into the folder created in step 2.
10. Close the workspace
11. Re-open the workspace

**Expected behavior**
When saving the workspace and when re-opening the workspace, the Espressif Opboarding script executes.  This is not expected since the ```idf.showOnboardingOnInit``` variable has been set to false.  

If the exact same change is made to the users settings.json file (located at $HOME/.config/Code/User/settings.json when working in Linux), then the Onboarding script will be correctly suppressed when re-opening the workspace.

I have tried to add several different settings to the workspace's settings.json file.  All have been ignored.

**Rationale behind wanting this fixed**
I am working on several ESP-32 projects that are on different versions of the ESP-IDF. It would be very convenient to save IDF version settings at the project (i.e. workspace) level.  This would make it much easier to change projects.

**Workaround**
I have discovered that I can remove all IDF settings from the user lever settings file and place them in the vscode workspace file as shown below:
``` json
{
	"folders": [
		{
			"path": "."
		}
	],
	"settings": {
		"idf.showOnboardingOnInit": false
	}
}
```
This workaround meets all of my needs, so I am not blocked.  However, it took me hours of lost productivity to figure out that workspace level settings are broken.  This feature/bug should be fixed, or at least documented, so that the next noob doesn't have to spend his time re-discovering this information.

**Environment**
- OS Name & Version: Ubuntu Ubuntu 18.04.5 LTS
- VSCode Version: 1.52.1
- ESP-IDF Version: 4.2
- Python Version: 3.6.9

brainstorm, 2021-01-02:
I concur and one-up with settings like `"idf.flashBaudrate": "921600"`, it seems to be ignored too:

[image]

brianignacio5, 2021-01-04:
The problem is the way you are viewing folders in Visual Studio Code.
Please take a look at [VSCode multi root workspaces](https://code.visualstudio.com/docs/editor/multi-root-workspaces#_adding-folders).

In your steps you are basically adding a folder to your current workspace, which basically just create an entry in the .code-workspace file with set of folders. You need to select the current folder you are working with `ESP-IDF: Pick a workspace folder` command so the extension knows which project you wish to build.

Also look at [ESP-IDF extension working with multiple projects](https://github.com/espressif/vscode-esp-idf-extension#working-with-multiple-projects) where we try to explain this scenario.

brainstorm, 2021-01-06:
@brianignacio5 I see all icons *and* I just ran that `ESP-IDF: Pick a workspace folder`... same effect :-S

Also as @JimGaluska points out, the "ESP-IDF Setup" tab pops up every time I restart the IDE:

[image]
[image]

brianignacio5, 2021-01-12:
@JimGaluska Have you fixed the issue ?

closed 2021-01-18 by brianignacio5