espressif/vscode-esp-idf-extension #254: [Bug Report]: IDF4.1 Onboarding reports missing Python requirements which are not in requierements.txt

definable, 2020-12-29:
Onboarding could not finish due to missing Python requirements:

The following Python requirements are not satisfied:
psutil>=5.5.1
pygdbmi<=0.9.0.2
requests>=2.21.0
typing>=3.6.6
xmlrunner>=1.7.7
To install the missing packages, please run "h:\projects\esp\esp-idf\esp-idf\install.bat"

However above packages are not in requirements.txt. As a result running install.bat did not help. Adding them to requirements.txt solved the issue.

brianignacio5, 2020-12-30:
As explain in the [ONBOARDING](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/ONBOARDING.md) documentation:

Make sure to install the extension and extension debug adapter Python requirements by running the following commands in your terminal:

PYTHON_INTERPRETER -m pip install -r EXTENSION_PATH/requirements.txt
PYTHON_INTERPRETER -m pip install -r EXTENSION_PATH/esp_debug_adapter/requirements.txt

brainstorm, 2020-12-31:
@brianignacio5 `requirements.txt` are indeed gone only those for the `esp_debug_adapter` are present:

```shell
$ pwd
/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.5.1
$ find . -iname requirements.txt
./esp_debug_adapter/requirements.txt
```

```shell
$ ls -alh
total 384
drwxr-xr-x  28 romanvg  staff   896B Dec 30 13:07 .
drwxr-xr-x  74 romanvg  staff   2.3K Dec 30 13:06 ..
-rw-r--r--   1 romanvg  staff   112B Dec 30 13:04 .gitmodules
-rw-r--r--   1 romanvg  staff   269B Dec 30 13:04 .prettierignore
-rw-r--r--   1 romanvg  staff    42B Dec 30 13:04 .prettierrc.yaml
-rw-r--r--   1 romanvg  staff   2.7K Dec 30 13:04 .vsixmanifest
-rw-r--r--   1 romanvg  staff   4.3K Dec 30 13:04 CHANGELOG.md
-rw-r--r--   1 romanvg  staff    11K Dec 30 13:04 LICENSE.txt
-rw-r--r--   1 romanvg  staff    25K Dec 30 13:04 README.md
drwxr-xr-x   5 romanvg  staff   160B Dec 30 13:04 dist
drwxr-xr-x  17 romanvg  staff   544B Dec 30 13:04 esp_debug_adapter
-rw-r--r--   1 romanvg  staff    21K Dec 31 10:39 esp_idf_vsc_ext.log
-rw-r--r--   1 romanvg  staff    11K Dec 30 13:04 fallback-tools.json
-rw-r--r--   1 romanvg  staff   1.0K Dec 30 13:04 get_serial_list.py
drwxr-xr-x   5 romanvg  staff   160B Dec 30 13:04 i18n
-rwxr-xr-x   1 romanvg  staff    46B Dec 30 13:04 idf_versions.txt
-rw-r--r--   1 romanvg  staff   579B Dec 30 13:04 kconfig-language-configuration.json
drwxr-xr-x   8 romanvg  staff   256B Dec 30 13:04 media
-rw-r--r--   1 romanvg  staff    22K Dec 30 13:04 package.json
-rw-r--r--   1 romanvg  staff   5.2K Dec 30 13:04 package.nls.en.json
-rw-r--r--   1 romanvg  staff   5.5K Dec 30 13:04 package.nls.es.json
-rw-r--r--   1 romanvg  staff   5.3K Dec 30 13:04 package.nls.json
-rw-r--r--   1 romanvg  staff   4.8K Dec 30 13:04 package.nls.zh-CN.json
-rw-r--r--   1 romanvg  staff   4.5K Dec 30 13:04 schema.i18n.json
drwxr-xr-x   3 romanvg  staff    96B Dec 30 13:04 snippets
drwxr-xr-x   3 romanvg  staff    96B Dec 30 13:04 syntaxes
drwxr-xr-x   6 romanvg  staff   192B Dec 30 13:04 templates
-rw-r--r--   1 romanvg  staff    13K Dec 30 13:04 tools.json
```

brainstorm, 2020-12-31:
They apparently were removed in https://github.com/espressif/vscode-esp-idf-extension/commit/7178fc84df35595578233ca5fe17f28bd783316a, `* removed redundant requirements.txt`, perhaps by accident? The docs were not updated if there's only authoritative `requirements.txt` now.

brianignacio5, 2021-01-04:
Those packages in @definable  issue are included in `EXTENSION_PATH/esp_debug_adapter/requirements.txt`. 
Extension requirements.txt are only  `gcovr` and `websocket_client`

Missing requirements.txt is fixed on current GitHub master. We will release a new version soon to update marketplace version.

Thanks for letting us know @brainstorm

brianignacio5, 2021-01-06:
Closing this issue since requirements.txt are specified in setup docs in a manual setup process. Setup wizard should install these requirements.txt automatically.

closed 2021-01-06 by brianignacio5