espressif/vscode-esp-idf-extension #257: [Feature Request]: idf.py set-target VSCode command in palette

brainstorm, 2020-12-31:
**Is your feature request related to a problem? Please describe.**

As pointed out in https://github.com/espressif/vscode-esp-idf-extension/issues/256#issuecomment-752820642, a better shortcut/command is needed to [select the current target being worked on](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#selecting-idf-target). See full description in that comment.

**Describe the solution you'd like**

I would like a `Cmd-P->ESP-IDF: Select target espressif MCU` command that unfolds a target selection dropdown with the most common ESP32 ICs: ESP32, ESP32S2, ESP32-C3, etc... then it modifies the `sdkconfig` or whatever accordingly for me.

**Describe alternatives you've considered**

Manually triggering the command `idf.py set-target esp32s2` in the VSCode shell seemed to have altered the project (after several CMake commands and compilation), but the errors in the issue comment above during monitor mode still persist, namely:

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4000f79a
Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
ets_main.c 386 
```

I'd like a relatively easy fix for that :)

boarchuz, 2020-12-31:
Look for "Set Espressif device target" command

brainstorm, 2020-12-31:
Yup, thanks! :D

closed 2020-12-31 by brainstorm