espressif/vscode-esp-idf-extension #258: [Bug Report]: Faulty default includePath

brainstorm, 2021-01-02:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
[image]

**Expected behavior**
0 Intellisense/includePath warnings after onboarding. Adding paths manually as hinted in the documentation:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md

Does not help either, neither does the builtin VSCode quick fixes:

```json
{
    "configurations": [
        {
            "name": "Mac",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${workspaceFolder}/**",
                "/Users/romanvg/dev/esp-idf/components/newlib/platform_include"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
```

**Environment (please complete the following information):**

- OS Name & Version: MacOS X 10.15.2 (Catalina)
- VSCode Version: Insiders, latest
- ESP-IDF Version: `master`
- Python Version: 3.9.1

brainstorm, 2021-01-06:
Sorry, I'm blocked by issue #276 at the moment, will try that when I can actually build through the IDE :/

brianignacio5, 2021-01-14:
compile_commands.json is generated by CMake by resolving all source dependencies there. Should maintain almost the same if your project dependencies don't change.

Closing this since it seems to me that compile_commands.json does the work. The naive configuration is not completely reliable, but we don't maintain the C/C++ extension so we can't provide a solution so far.

closed 2021-01-14 by brianignacio5