espressif/vscode-esp-idf-extension #261: Fresh Install Build Fails

jscott304, 2021-01-04:
Fresh Windows 10 Home install.
Installed VS Code, VS_BuildTools (Was Needed for ESP_IDF Tools Install), Python 3.9.1, GIT version 2.30.0.windows.1
ESP_IDF Tools 2.3. 
Added all the Paths to System Environment Variables.
I'm able to build (Hello_World) via CMD using idf.py build and flash my device.
After adding the espressif IDF Extension and getting the orientation to pass I copied Hello_World to a new folder using the examples browser and tried building it but it fails. below is the output for the build errors.
`> Executing task: cmake -G Ninja .. <

-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
The following Python requirements are not satisfied:
click>=7.0   
pyserial>=3.0
future>=0.15.2
cryptography>=2.1.4
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
kconfiglib==13.7.1
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\ESP32\esp-idf\install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: .
    Python interpreter used: C:\Python\Python39\python.exe
CMake Error at C:/ESP32/esp-idf/tools/cmake/build.cmake:272 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/ESP32/esp-idf/tools/cmake/build.cmake:398 (__build_check_python)
  C:/ESP32/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/ESP32/Projects/Blink/blink/build/CMakeFiles/CMakeOutput.log".      
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja .." terminated with exit code: 1.`

brianignacio5, 2021-01-04:
Did you configure the extension following the steps in README ?

What is the value of `idf.pythonBinPath` ? Check ESP-IDF Configuration settings (UI or JSON) for `Python bin path`. Is it the same as the one in your virtual env `C:\.espressif\python_env\idf4.1_py3.8_env\Scripts\python.exe`?

Please take a look about [SETUP doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md) for additional information.

brainstorm, 2021-01-05:
I'm having the same problems and yes, I've followed the steps in README, my `settings.json` looks like this:

```json
{
  "idf.port": "/dev/cu.SLAB_USBtoUART",
  "idf.espIdfPath": "/Users/romanvg/dev/esp-idf",
  "idf.pythonBinPath": "/Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/python",
  "idf.flashBaudrate": "921600",
  "C_Cpp.clang_format_style": "Visual Studio",
  "editor.formatOnSave": false,
  "[cpp]": {
    "editor.quickSuggestions": true
  },
  "[c]": {
    "editor.quickSuggestions": true
  },
  "C_Cpp.intelliSenseEngineFallback": "Enabled",
  "idf.adapterTargetName": "esp32s2",
  "idf.openOcdConfigs": [
    "interface/remote_bitbang.cfg",
    "target/esp32s2.cfg"
  ],
  "files.associations": {
    "freertos.h": "c",
    "twai.h": "c",
    "*.ipp": "cpp",
    "*.inc": "cpp",
    "array": "cpp",
    "*.tcc": "cpp",
    "string": "cpp",
    "vector": "cpp",
    "string_view": "cpp",
    "functional": "cpp",
    "iomanip": "cpp",
    "istream": "cpp",
    "limits": "cpp",
    "ostream": "cpp",
    "ratio": "cpp",
    "sstream": "cpp",
    "streambuf": "cpp",
    "thread": "c"
  },
  "idf.flashType": "UART"
}
```

And the error looks like this:

```shell
[variant] Loaded new set of variants
[kit] Successfully loaded 8 kits from /Users/romanvg/.local/share/CMakeTools/cmake-tools-kits.json
[main] Configuring folder: twai_network_slave 
[proc] Executing command: /usr/local/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++ -H/Users/romanvg/dev/esp32s2_can/twai_network_slave -B/Users/romanvg/dev/esp32s2_can/twai_network_slave/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Found Git: /usr/local/bin/git (found version "2.25.1") 
[cmake] -- The C compiler identification is GNU 8.4.0
[cmake] -- The CXX compiler identification is GNU 8.4.0
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: /Users/romanvg/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /Users/romanvg/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /Users/romanvg/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
[... 21 lines cut ...]
[cmake] See also "/Users/romanvg/dev/esp32s2_can/twai_network_slave/build/CMakeFiles/CMakeOutput.log".
[cmake] CMake Error at /Users/romanvg/dev/esp-idf/tools/cmake/build.cmake:276 (message):
[cmake]   Some Python dependencies must be installed.  Check above message for
[cmake]   details.
[cmake] Call Stack (most recent call first):
[cmake]   /Users/romanvg/dev/esp-idf/tools/cmake/build.cmake:402 (__build_check_python)
[cmake]   /Users/romanvg/dev/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
[cmake]   CMakeLists.txt:6 (project)
[cmake] 
[cmake] 
```

And the python venv is present:

```shell
$ ls -alh /Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/
total 16
drwxr-xr-x   7 romanvg  staff   224B Jan  5 23:18 .
drwxr-xr-x   3 romanvg  staff    96B Jan  5 23:17 ..
-rw-r--r--   1 romanvg  staff    40B Jan  5 23:17 .gitignore
drwxr-xr-x  44 romanvg  staff   1.4K Jan  6 00:38 bin
drwxr-xr-x   3 romanvg  staff    96B Jan  5 23:18 include
drwxr-xr-x   3 romanvg  staff    96B Jan  5 23:17 lib
-rw-r--r--   1 romanvg  staff   443B Jan  5 23:17 pyvenv.cfg
```

brainstorm, 2021-01-05:
```shell
$ grep IDF ~/.bashrc
export IDF_TARGET="esp32s2"
export IDF_PATH=~/dev/esp-idf
export IDF_PYTHON_ENV_PATH=$HOME/.espressif/python_env/idf4.3_py3.9_env
. ${IDF_PATH}/add_path.sh
. ${IDF_PATH}/export.sh
```

jscott304, 2021-01-06:
Not sure about brainstorms issue but after getting the latest update for VS Code ESPRESSIF it is now working. There must have been an issue that was fixed with the update.
Thanks,

brainstorm, 2021-01-06:
Which update? 0.6.0?

brianignacio5, 2021-01-06:
@brainstorm the log gives you the issue, you have missing python packages in the virtual environment of `/Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/python`. If you do:

`/Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/python -m pip install /Users/romanvg/dev/esp-idf/requirements.txt`

`/Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/python -m pip install EXTENSION_PATH/requirements.txt`

`/Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/python -m pip install EXTENSION_PATH/esp_debug_adapter/requirements.txt`

where EXTENSION_PATH might look like `/Users/romanvg/.vscode/extensions/espressif.esp-idf-extension-VERSION`

IF you install ESP-IDF Tools with the extension wizard, this is done for you automatically. If you installed other way then extension and debug adapter requirements will be missing, as explained in [SETUP](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md)

Closing this, since the original issue is fixed for @jscott304

closed 2021-01-06 by brianignacio5

brainstorm, 2021-01-06:
Thanks @brianignacio5 but all was needed was a `source /Users/romanvg/.espressif/python_env/idf4.3_py3.9_env/bin/activate` on my `~/.bashrc` instead of what you are proposing... unfortunately other issues remain (issue #276) :/