espressif/vscode-esp-idf-extension #262: [Feature Request]: Add Arduino libraries

Asteliks, 2021-01-04:
## Problem related to the needed feature
I find it usually hard to find all the needed libraries in the IDF compatible format. More often than not they poorly documented or even unavailable.

## The suggested solution I would like
I wish we could link the path to all the arduino libraries and that they can just work with the ESP-idf

## Alternative solutions
* Improve the ESP-idf library base so that there is more of quality libraries with good documentation
* improve the documentation on how to use arduino libraries with ESP-idf

projectgus, 2021-01-07:
Hi @Asteliks,

Thanks for letting us know. You're specifically talking about using Arduino libraries in your ESP-IDF projects, right?

This may be something to direct to the [arduino-esp32](https://github.com/espressif/arduino-esp32) project as currently this is where Arduino/ESP-IDF integration is provided.

Can you give examples of the kinds of libraries from Arduino that your'e aiming to use in ESP-IDF, that aren't working? What kind of support do you need that's only available by including an Arduino library?

Asteliks, 2021-01-07:
Hi @projectgus 
Thank you for answering to my request. Yes, I am talking about using the libraries (made by the community) made for Arduino with the latest ESP-IDF version.
 In general, I have started my ESP32 adventure with the Arduino-esp32 project but now I am moving towards more advanced features that for example allow me to make AppleHome kit accessories.
Basically, I got used to some of the libraries I have there and wish I could use the with the latest ESP-IDF (v4.2) but for that I would need a feature to use C++ libraries with my C code. For example, the libraries I use are: 
* [ArduinoJson](https://arduinojson.org/)
* [DHT sensor library](https://github.com/adafruit/DHT-sensor-library)
* [MFRC522](https://github.com/miguelbalboa/rfid)
* [rc-switch](https://github.com/sui77/rc-switch)
* [RFM69_LowPowerLab](https://github.com/LowPowerLab/RFM69)
* [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)
* [RadioHead](https://www.airspayce.com/mikem/arduino/RadioHead/index.html)
* [CC1101](https://github.com/SpaceTeddy/CC1101)

probably one should be able to find the esp-idf v4.2 compatible counterparts for some of them but I would appreciate it if this hassle could be saved and I could simply use these liberties in my ESP-IDFv4,2 projects.

brianignacio5, 2021-01-11:
We have a command to **Add Arduino ESP32 as ESP-IDF Component** that will add Arduino libraries as a component to your currently opened project directory ?

Have you looked into [Using Arduino as ESP-IDF Component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) ? There is an explanation to use Arduino sketches but there is probably some changes needed in each of these libraries.

Asteliks, 2021-01-11:
> We have a command to **Add Arduino ESP32 as ESP-IDF Component** that will add Arduino libraries as a component to your currently opened project directory ?

Error expected "." but got "?" at the end of the sentence. Unsure how to answer this question.

> Have you looked into [Using Arduino as ESP-IDF Component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) ? There is an explanation to use Arduino sketches but there is probably some changes needed in each of these libraries.

I have looked into Using Arduino as ESP-IDF Component but to be honest got lost at first with the description. The one you've linked seems better with the given context. I'll take a look at it.

If it dose what I need - allow me to import Arduino libraries in to my projects. Could you make it so that I may link that at the plugin level so I do not have to go through the process every time. I understand that this adds certain paths into the project my suggestion to counter that is an extension question at project creation whether the user wants to use Arduino libraries.

brianignacio5, 2021-01-12:
Used "." instead of "?"

This extension is mostly focused on working with ESP-IDF. Although we provide to add Arduino libraries as ESP-IDF component as done in [https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md] but that is it. We do not maintain any of these Arduino libraries so we can't guarantee that they work.

It is my recommendation that you navigate through several solutions implemented for ESP-IDF as listed [here](https://github.com/espressif/esp-iot-solution)

brianignacio5, 2021-01-15:
Closing this since it is a non-issue. Please follow up with the Arduino-esp32 repository for extended Arduino support or use recommended libraries before.

closed 2021-01-15 by brianignacio5