espressif/vscode-esp-idf-extension #264: [Bug Report]: Failed to configure the ESP-IDF extension

xiaolongba, 2021-01-04:
**Describe the bug**
Failed to open **ESP-IDF: Configure ESP-IDF extension**  after changing the windows user name. But I reinstall the extension, the issue still exists 

**To Reproduce**
1. open the vscode
2. Press <code>F1</code> and select the **ESP-IDF: Configure ESP-IDF extension**
3. error occurred

**Screenshots**

[image]

**Environment (please complete the following information):**

- OS Name & Version: win10
- VSCode Version: 1.52.1
- ESP-IDF Version: preview version
- Python Version: 3.9

**Logs**
cannot find any logs at this moment

brianignacio5, 2021-01-05:
On Visual Studio Code click the Help menu and select Toggle Developer Tools. This will show Visual Studio Code's Chrome Developer Tools with a console tab. Please show us the error there.

Try delete the espressif folder in your .vscode extensions folder and reinstall again also.

xiaolongba, 2021-01-05:
> Try delete the espressif folder in your .vscode extensions folder and reinstall again also.

Hi bro, i tried to do that with your tips, but the issue still exists.

> On Visual Studio Code click the Help menu and select Toggle Developer Tools. This will show Visual Studio Code's Chrome Developer Tools with a console tab. Please show us the error there.

[image]

xiaolongba, 2021-01-06:
Hi @pwmb @brianignacio5 bro,

many thanks for your great support, I think I resolved this issue already. the manner is to delete the <code>.vscode</code> folder and then reinstall the all extensions . afterwards, all extensions are working well.

thanks again

closed 2021-01-06 by xiaolongba