espressif/vscode-esp-idf-extension #265: [Bug Report]: Start App Trace fails for esp32s2 on Windows - 1) cannot create trace file 2) invalid openocd command

stokmanis, 2021-01-04:
**Describe the bug**
I can run openocd from vs code on windows 10 with esp32s2 module and read App Trace using telnet 4444 port.
When pressing Start App Trace from vs code 2 errors are reported
1) it cannot create trace file - folder exists, permissions are ok
2) it issues invalid openocd command name "esp32", which is wrong in the esp32s2 context

Projects target is set esp32s2, and even workspace setting "idf.adapterTargetName" is set to "esp32s2".
Bellow is the output.

`Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : Listening on port 3333 for gdb connections
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection
Info : accepting 'tcl' connection on tcp/6666
Info : Total trace memory: 16384 bytes
Info : Open file /c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace
❌ Error: Failed to ❌ open file /c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace
Error: apptrace: Failed to init trace data destination 'file:///c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace'!
Error: Invalid 'cmd_❌ data->poll_period' arg!
Error: Not enough args! Need 1 trace data destinations!
Error: Failed to init cmd ctx (-4)!
Total trace memory: 16384 bytes
Open file /c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace
Failed to open file /c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace
apptrace: Failed to init trace data destination 'file:///c%3A/g/e3/projects/esp32_catm_rc1180/trace/trace_1609778610494.trace'!
Invalid 'cmd_data->poll_period' arg!
Not enough args! Need 1 trace data destinations!
Failed to init cmd ctx (-4)!
Info : accepting 'tcl' connection on tcp/6666
command.c:795: Error: invalid command name "esp32"
at file "command.c", line 795
❌ command.c:795: Error: invalid command name "esp32"
at file "command.c", line 795`

**To Reproduce**
Create esp-idf project in vs code, and some trace statements in code e.g. 
esp_apptrace_write(ESP_APPTRACE_DEST_TRAX, "Hello", 5, ESP_APPTRACE_TMO_INFINITE);
esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, ESP_APPTRACE_TMO_INFINITE);
set target to esp32s2
connect esp32s2 module to esp-prog programming and jtag connectors
build, flash and monitor
Start OpenOCD server from vs code
Press Start App Trace

**Expected behavior**
Expect to view the Hello message in the output

- OS Name & Version: Windows 10
- VSCode Version: 1.52.1
- ESP-IDF Version: 4.2
- Python Version: 3.8
- ESP-IDF vs code extention: 0.60

**Logs**
If applicable, please share the log file which can be obtained from
- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\User\\.vscode\\extensions\\espressif.esp-idf-extension-0.6.0\\dist\\extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-04T16:57:54.055Z"}

brianignacio5, 2021-01-05:
What is the value of `idf.openOcdConfigs` ?

stokmanis, 2021-01-05:
It was "idf.openOcdConfigs": ["interface/ftdi/esp32_devkitj_v1.cfg","target/esp32s2.cfg"]
Now I changed to "idf.openOcdConfigs": ["interface/ftdi/esp32s2_kaluga_v1.cfg","target/esp32s2.cfg"]
Still the same issues with the same output.

pwmb, 2021-01-05:
@stokmanis could you share your log file? for us to analyse the problem properly.

Maybe there is some text like `Some error prevailed while checking the tracking status` ?

pwmb, 2021-01-05:
@gerekon we have to update this command from app tracing, for s2, will this be `esp32s2` instead of `esp32` ? https://github.com/espressif/vscode-esp-idf-extension/blob/3cbc3b9a9920a10dd3d9c0144e86478064c5b3d7/src/espIdf/tracing/appTraceManager.ts#L269

stokmanis, 2021-01-05:
I suspect that vscode.Uri.file converts windows file path from c:/my.trace to /c%3A/m.trace and it forms invalid openocd cmd
> esp apptrace start file:///c%3A/my.trace
Total trace memory: 16384 bytes
Open file /c%3A/my.trace
Failed to open file /c%3A/my.trace
apptrace: Failed to init trace data destination 'file:///c%3A/my.trace'!

But command >esp apptrace start file://c:/my.trace is ok
About - tclClient.sendCommandWithCapture("esp32 apptrace status");
I my case you can use "esp apptrace status", which does not reference esp32s2, but still works

NikitaSyrovatnikov, 2021-01-06:
The problem is in file `src/extension.ts`

Method `getTxtCmd` on line 2450

The correct code for Window and Linux will

`function getTxtCmd(variable: string, modifiedEnv: { [key: string]: string }) {
  if (process.platform === "win32") {
    const shellExecutable = path.basename(vscode.env.shell);
    switch (shellExecutable) {
      case "cmd.exe":
        return `set "${variable}=${modifiedEnv[variable]}`;
      case "powershell.exe":
      case "pwsh.exe":
        return `$Env:${variable}="${modifiedEnv[variable]}"`
    }
  }
  return `export ="${modifiedEnv[variable]}"`
}`

pwmb, 2021-01-06:
> I suspect that vscode.Uri.file converts windows file path from c:/my.trace to /c%3A/m.trace and it forms invalid openocd cmd

hmm, Interesting, Thanks for the inputs

PR #296 (2021-01-18): Fix recent change to OpenOCD config files
- Write an array to settings.json instead of a string
- Fix typo in that array
- Use updated apptrace command

Closes #265
closed 2021-01-18 by brianignacio5