espressif/vscode-esp-idf-extension #266: [Bug Report]: Monitor fails in vscode after updating to 0.6.0

bpietsch, 2021-01-04:
**Describe the bug**
After flashing (successfully), the Monitor terminal appears and prints out a couple of lines showing what appears to be the PATH environment variable. After that, no additional output. Reverting to the 0.5.1 plugin allows it to work again.

**To Reproduce**
Choose the build, flash and monitor option (or just monitor if device is already flashed)
View monitor output terminal in vscode - notice nothing is shown (beyond some environment variable output)

**Expected behavior**
Monitor to show serial output as it did in 0.5.1

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: MacOS X 11.1
- VSCode Version: 1.52.1
- ESP-IDF Version: v4.1
- Python Version: 3.8

**Logs**
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T17:40:48.171Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T17:45:58.002Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-04T17:51:40.458Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:258876)\n\tat ChildProcess.emit (events.js:223:5)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)","level":"error","timestamp":"2021-01-04T17:51:47.609Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T17:52:35.726Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-04T17:52:54.844Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:258876)\n\tat ChildProcess.emit (events.js:223:5)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)","level":"error","timestamp":"2021-01-04T17:53:01.187Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","level":"error","timestamp":"2021-01-04T17:53:25.213Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T17:55:22.215Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-01-04T17:56:25.385Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T17:56:25.507Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-01-04T17:57:25.667Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20191114 (2019-11-14-14:19)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.e [... cut ...]  reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/brian/.vscode/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n\tat ChildProcess.exithandler (child_process.js:295:7)\n\tat ChildProcess.emit (events.js:223:5)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:430:11)\n\tat Socket.emit (events.js:223:5)\n\tat Pipe.<anonymous> (net.js:664:12)","level":"error","timestamp":"2021-01-04T18:22:16.056Z"}

kosnox, 2021-01-04:
I have the same problem but this is my log

`C:\Users\Michal\Desktop\esp-idf-webserver>C:\Users\Michal\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\Michal\esp-idf\tools\idf.py -p COM3 monitor
Traceback (most recent call last):
  File "C:\Users\Michal\esp-idf\tools\idf.py", line 784, in <module>
    main()
  File "C:\Users\Michal\esp-idf\tools\idf.py", line 700, in main
    checks_output = check_environment()
  File "C:\Users\Michal\esp-idf\tools\idf.py", line 68, in check_environment
    print_idf_version()
  File "C:\Users\Michal\esp-idf\tools\idf.py", line 117, in print_idf_version
    version = idf_version()
  File "C:\Users\Michal\esp-idf\tools\idf_py_actions\tools.py", line 56, in idf_version
    "--git-dir=%s" % os.path.join(os.environ["IDF_PATH"], '.git'),
  File "E:\PYTHON3\lib\os.py", line 673, in __getitem__
    raise KeyError(key) from None
KeyError: 'IDF_PATH'`

Asteliks, 2021-01-04:
I also have the same problem - the monitor dose not open and my log is:

`PS C:\Users\axels\Desktop\hello_world> c:\ESP\python_env\idf4.2_py3.9_env\Scripts\python.exe c:\ESP\esp-idf\tools\idf.py -p COM3 monitor
Traceback (most recent call last):
  File "c:\ESP\esp-idf\tools\idf.py", line 784, in <module>
    main()
  File "c:\ESP\esp-idf\tools\idf.py", line 700, in main
    checks_output = check_environment()
  File "c:\ESP\esp-idf\tools\idf.py", line 68, in check_environment
    print_idf_version()
  File "c:\ESP\esp-idf\tools\idf.py", line 117, in print_idf_version
    version = idf_version()
  File "c:\ESP\esp-idf\tools\idf_py_actions\tools.py", line 54, in idf_version
    version = subprocess.check_output([
  File "C:\Users\axels\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 420, in check_output   
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\axels\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\axels\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 947, in __init__       
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\axels\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Nie można odnaleźć określonego pliku`

marcpiulachs, 2021-01-04:
I also have problems. 

No error is show but nothing happens:

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Source\WiFiStationFirmware> $Env:Path = "
>> $Env:IDF_PATH = "
PS C:\Source\WiFiStationFirmware> $Env:IDF_TARGET = "
>> $Env:PYTHON = "
PS C:\Source\WiFiStationFirmware> $Env:IDF_PYTHON_ENV_PATH = "
>> cls
>> C:\Users\Marc\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe C:\Users\Marc\esp\esp-idf\tools\idf.py -p COM9 monitor
>>

GOST-UA, 2021-01-04:
Have the same issue.

As a workaround I am able to use user task: `CTRL + SHIP + P` -> `Tasks: Run Task` -> `Monitor: Start the monitor`
Also created a similar task without flash dependency (in case if someone also needs this)

bpietsch, 2021-01-04:
OK, after messing around I discovered that re-running the onboarding flow for the extension seems to have resolved the issue. (ESP-IDF: Configure ESP-IDF Extension) Maybe something in the configuration isn't being automatically updated / converted when the new extension version is run the fist time?

Asteliks, 2021-01-04:
> OK, after messing around I discovered that re-running the onboarding flow for the extension seems to have resolved the issue. (ESP-IDF: Configure ESP-IDF Extension) Maybe something in the [...]

Unfortunately this did not help in my case

amalabey, 2021-01-05:
I have the same issue. Serial monitor shows below output:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

C:\Users\username\.vscode\extensions\espressif.esp-idf-extension-0.6.0> $Env:Path = "
>> $Env:IDF_PATH = "
C:\Users\username\.vscode\extensions\espressif.esp-idf-extension-0.6.0> $Env:IDF_TARGET = "
>> $Env:PYTHON = "
C:\Users\username\.vscode\extensions\espressif.esp-idf-extension-0.6.0> $Env:IDF_PYTHON_ENV_PATH = "
>> cls
>> C:\Users\username\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\amal\esp\esp-idf\tools\idf.py -p COM1 monitor
```

brainstorm, 2021-01-05:
Same in OSX, it just writes out a couple of exports and stops:

```shell
export PATH="/Users/romanvg/dev/esp-idf/components/esptool_py/esptool:/Users/romanvg/dev/esp-idf/components/espcoredump:/Users/romanvg/dev/esp-idf/components/partition_table:.:/Users/romanvg/.miniconda3/envs/espressif/bin:/Users/romanvg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/romanvg/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/romanvg/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin:/Users/romanvg/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/romanvg/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/romanvg/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:.:/Users/romanvg/dev/esp-idf/tools:/usr/local/opt/curl/bin:/usr/local/opt/scala@2.12/bin:/usr/local/opt/terraform@0.11/bin:/Users/romanvg/.miniconda3/bin:/Users/romanvg/.miniconda3/condabin:/Users/romanvg/.cargo/bin:/usr/local/sbin:/usr/local/heroku/bin:/usr
The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.

$ export PATH="/Users/romanvg/dev/esp-idf/components/esptool_py/esptool:/Users/romanvg/dev/esp-idf/components/espcoredump:/Users/romanvg/dev/esp-idf/components/partition_table:.:/Users/romanvg/.miniconda3/envs/espressif/bin:/Users/romanvg/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/romanvg/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/romanvg/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin:/Users/romanvg/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/romanvg/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/romanvg/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:.:/Users/romanvg/dev/esp-idf/tools:/usr/local/opt/curl/bin:/usr/local/opt/scala@2.12/bin:/usr/local/opt/terraform@0.11/bin:/Users/romanvg/.miniconda3/bin:/Users/romanvg/.miniconda3/condabin:/Users/romanvg/.cargo/bin:/usr/local/sbin:/usr/local/heroku/bin:/usr
```

And still does not respect the `tasks.json` definition since I have a `conda activate espressif` command defined there (see issue #256 and there's none of that in the output after clicking the button :/

brianignacio5, 2021-01-05:
Can you share which powershell version (pscore?) terminal are you guys using ? It seems terminal is breaking the PATH somewhere so is not an actual executable line.

GOST-UA, 2021-01-05:
```
Major  Minor  Build  Revision  
-----  -----  -----  --------  
5      1      18362  1171  
```

Also experiencing the same behavior in the command prompt.

MisterScience5, 2021-01-05:
Version
-------
5.1.19041.610

Same behaviour for me:

`PS E:\ESP\projects\ilume_powersave> $Env:Path = "
>> $Env:IDF_PATH = "
PS E:\ESP\projects\ilume_powersave> $Env:IDF_TARGET = "
>> $Env:PYTHON = "
PS E:\ESP\projects\ilume_powersave> $Env:IDF_PYTHON_ENV_PATH = "
>> cls
>> c:\Users\prebe\esp-tools\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\prebe\esp-idf\tools\idf.py -p COM10 monitor`

kosnox, 2021-01-05:
> I have the same problem but this is my log
>
> `C:\Users\Michal\Desktop\esp-idf-webserver>C:\Users\Michal\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\Michal\esp-idf\tools\idf.py [...]

I’ve tried reinstalling ESP-idf but that didn’t help. Still having the same problem

And my version powershell: 5.1.19041.610

NikitaSyrovatnikov, 2021-01-06:
The problem is in file `src/extension.ts`

Method `getTxtCmd` on line 2450

The correct code for Window and Linux will

```
function getTxtCmd(variable: string, modifiedEnv: { [key: string]: string }) {
  if (process.platform === "win32") {
    const shellExecutable = path.basename(vscode.env.shell);
    switch (shellExecutable) {
      case "cmd.exe":
        return `set "${variable}=${modifiedEnv[variable]}`;
      case "powershell.exe":
      case "pwsh.exe":
        return `$Env:${variable}="${modifiedEnv[variable]}"`;
    }
  }
  return `export ${variable}="${modifiedEnv[variable]}"`;
}
```

fahadalisarwar1, 2021-01-06:
> Have the same issue.
> 
> As a workaround I am able to use user task: `CTRL + SHIP + P` -> `Tasks: Run Task` -> `Monitor: Start the monitor`
> Also created a similar task without flash dependency (in case if someone also needs this)

Not working for me!

chandini29, 2021-01-06:
> getTxtCmd

Where is src/extension.ts file located?

GOST-UA, 2021-01-06:
@chandini29 In this repo. If you want to manually fix this you need to clone this repo, make the fix as @NikitaSyrovatnikov described, and build using this guide: https://github.com/espressif/vscode-esp-idf-extension#build-from-source-code

or just wait till version v0.6.1 will be available 😉

fahadalisarwar1, 2021-01-06:
> @chandini29 In this repo. If you want to manually fix this you need to clone this repo, make the fix as @NikitaSyrovatnikov described, and build using this guide: https://github.com/espressif/vscode-esp-idf-extension#build-from-source-code
> 
> or just wait till version v0.6.1 will be available 😉

And when will version v0.6.1 be released ?

Asteliks, 2021-01-06:
> And when will version v0.6.1 be released ?

Probably when all v0.6.1 tickets are closed: https://github.com/espressif/vscode-esp-idf-extension/milestone/17

brainstorm, 2021-01-08:
@GOST-UA I'm trying to fix another issue and I'm finding that the extension is not so easy to debug, see https://github.com/espressif/vscode-esp-idf-extension/issues/276#issuecomment-756715995

Did you manage to switch to `development` mode (via `yarn debugPreTask`?).

GOST-UA, 2021-01-08:
@brainstorm  I didn't debug this extension on my own - just left a bit more detailed description of how to run it from sources.

resetnow, 2021-02-14:
Somehow I'm still having this problem on 0.6.1 - but only with zsh and omz

brianignacio5, 2021-02-18:
I've tested on zsh

resetnow, 2021-02-18:
Here's how it looks:

[image]

[image]

brianignacio5, 2021-02-18:
Which shell are you using for Visual Studio Code ? Apparently is truncating received command text somehow.

resetnow, 2021-02-18:
It's system default - zsh. And it looks to me like it's very similar to the first screenshot in this thread

brianignacio5, 2021-02-19:
I tested it using VSCodium and zsh also but could not reproduce on MacOS. 

Maybe you can provide additional information for your setup to further debug the issue

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.

resetnow, 2021-02-19:
Wow, thank you for going the extra mile to debug it. I'll try to collect all of this this evening.

resetnow, 2021-02-19:
Here are some additional details:

<details>
  <summary>esp_idf_vsc_ext.log</summary>

```
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:7755\n\tat ChildProcess.exithandler (child_process.js:303:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)","level":"error","timestamp":"2021-02-14T13:48:35.851Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:7755\n\tat ChildProcess.exithandler (child_process.js:303:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1021:16)\n\tat Socket.<anonymous> (internal/child_process.js:443:11)\n\tat Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:674:12)","level":"error","timestamp":"2021-02-14T14:48:01.647Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-02-14T15:14:42.422Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-02-14T15:17:43.863Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-02-14T15:18:28.099Z"}
{"user":true,"level":"warn","message":"ESP-IDF Monitor was closed.","timestamp":"2021-02-14T15:18:43.272Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:264560)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-02-14T19:29:25.165Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat Function.<anonymous> (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:47123)\n\tat Generator.next (<anonymous>)\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:46231\n\tat new Promise (<anonymous>)\n\tat o (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:45976)\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:46966\n\tat v.fire (/Applications/VSCodium.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)\n\tat b.$onDidEndTaskProcess (/Applications/VSCodium.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:48187)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:97:5)","level":"error","timestamp":"2021-02-14T19:29:25.189Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:264560)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)","level":"error","timestamp":"2021-02-14T19:30:06.262Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat Function.<anonymous> (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:47123)\n\tat Generator.next (<anonymous>)\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:46231\n\tat new Promise (<anonymous>)\n\tat o (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:45976)\n\tat /Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:46966\n\tat v.fire (/Applications/VSCodium.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)\n\tat b.$onDidEndTaskProcess (/Applications/VSCodium.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:48187)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:97:5)","level":"error","timestamp":"2021-02-14T19:30:06.285Z"}
{"message":"ESP-IDF Terminal process ended with exit code 1.","stack":"Error: ESP-IDF Terminal process ended with exit code 1.\n\tat ChildProcess.<anonymous> (/Users/vladislavivanov/.vscode-oss/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:264560)\n\ [... cut ...]
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/Users/vladislavivanov/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\"}",
```
</details>

Looking at the behaviour, it seems to me like the issue is the sendText method sending commands way too early. And there doesn't seem to be any synchronisation mechanism:

https://github.com/Microsoft/vscode/issues/11383

I think the only reliable way to fix this would be to put commands in a temporary file and then pass it as shellArgs. Or perhaps the whole thing can go into shellArgs.

resetnow, 2021-02-26:
@brianignacio5 I added a 500ms delay in function `createMonitor` and this solved the issue for me. But in general, proper fix would pass the command you want to execute as `shellArgs`. I also implemented it that way, but I don't have a Windows machine to test the powershell stuff and I'm not familiar with the code - some parts have logic that needs to be reworked. Would you maybe consider reopening this issue?

By the way, there is one more downside of typing commands into the terminal: they pollute shell history.

brianignacio5, 2021-03-01:
Yes @resetnow this is a whole set of issues in vscode API, basically we are trying to update some environment variables (IDF_PATH, PATH, IDF_TARGET, IDF_PYTHON_ENV_PATH) but the vscode doesn't have a clean way to achieve this yet (options.env just post append variables). The thing with shellArgs is that can only be used when terminal is created and we have several export VAR commands. I'll test some ways and let you try in PR.

resetnow, 2021-03-01:
@brianignacio5 thank you for reopening.

My impression is that they intend those terminals to be used in a "stateless" way - create a terminal with a given state, and if you need to change something, dispose of it and create a new one.

brianignacio5, 2021-03-01:
As far as I know, the intended approach is to use [environmentVariableCollection](https://code.visualstudio.com/updates/v1_46#_environment-variable-collection) but replace or preapprend doesn't work on macOS as shown in [here](https://github.com/microsoft/vscode/issues/99878).

PR #274 (2021-01-06): Fix monitor init cmd
Fix commands in `getTxtCmd`

Fix ESP-IDF version regex

Fix #266
PR #337 (2021-03-03): Relative links in tasks output, truncated export cmd in terminal
Fix #333 

Fix #266 

Fix #329

Use vscode ShellExecution to fix relative paths link in build/flash task outputs. PTAL @boarchuz 

Remove export cmd and use strict env for Monitor terminal and ESP-IDF Terminal . PTAL @resetnow

Fix closing monitor issues when flash is trigger. PTAL @seulater
closed 2021-03-03 by brianignacio5