espressif/vscode-esp-idf-extension #269: Flash jtag failed

simark1979, 2021-01-05:
I just updated extention to 0.6
I`m getting:
Minimum OpenOCD version v0.10.0-esp32-20201125 is required

But extention installs v0.10.0-esp32-20191114....
While installing I set github download source.

pwmb, 2021-01-05:
This is because you have to download that version of openocd manually from the repo. to run JTAG flashing, and then point the path to that version, or otherwise you can wait till IDF will update their [tools.json](https://github.com/espressif/esp-idf/blob/master/tools/tools.json) for the particular version of IDF you're consuming

closed 2021-01-05 by pwmb

simark1979, 2021-01-05:
I`m sorry. It was downloaded by extention.
No any manuall downloading....

Before.
1. I unistalled old version of extention inside of VSCode. (extention flder from '.vscode\extensions' not removed)
2. .espressif and esp-idf also removed
3. Installed new extention (as download source selected github)

As result:
I have v0.10.0-esp32-20191114 version again

pwmb, 2021-01-05:
Extension will download open-ocd or tools which are part of [tools.json](https://github.com/espressif/esp-idf/blob/master/tools/tools.json), you have to __manually__ download specified version of openocd from github and then replace that with the current installed open-ocd.

This is required because that version of open-ocd is still not part of the IDF releases, but in the future will be.

simark1979, 2021-01-05:
Thanks a lot