espressif/vscode-esp-idf-extension #270: [Bug Report]: after updating to 0.6.0，build failed.

JerryZheng89, 2021-01-06:
- windows 10
- Visual Studio Code, Version: 1.52.1

when update to 0.6.0, every time open esp32 project forder, the `OUTPUT` tab will show esp_idf tools version check failed and a lot Garbled character. 

But the ESP-IDF terminal works well.

And when I go back to 0.5.1 , everything works well!

brianignacio5, 2021-01-06:
Can you provide additional information ?

There is a issue template with required info

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

Also please share the content of your extension configuration settings

`idf.espIdfPath` , `idf.customExtraPaths`, `idf.customExtraVars`

meltdown03, 2021-01-07:
I get this error when building on 0.6.0 unless I add back in `/usr/bin` to `idf.customExtraPaths`:
```
> Executing task: /home/neal/.espressif/python_env/idf4.3_py3.9_env/bin/python /home/neal/esp/esp-idf/tools/idf.py build <

/home/neal/.zshenv:4: command not found: tty
Traceback (most recent call last):
  File "/home/neal/esp/esp-idf/tools/idf.py", line 811, in <module>
    main()
  File "/home/neal/esp/esp-idf/tools/idf.py", line 726, in main
    checks_output = check_environment()
  File "/home/neal/esp/esp-idf/tools/idf.py", line 77, in check_environment
    debug_print_idf_version()
  File "/home/neal/esp/esp-idf/tools/idf.py", line 132, in debug_print_idf_version
    version = idf_version()
  File "/home/neal/esp/esp-idf/tools/idf_py_actions/tools.py", line 54, in idf_version
    version = subprocess.check_output([
  File "/usr/lib/python3.9/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.9/subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1819, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'git'
The terminal process "/usr/bin/zsh '-c', '/home/neal/.espressif/python_env/idf4.3_py3.9_env/bin/python /home/neal/esp/esp-idf/tools/idf.py build'" failed to launch (exit code: 1).
```

brianignacio5, 2021-01-07:
I think this is an error of your terminal itself. Apparently in your system environment PATH, you don't have `/usr/bin` for some reason. You can manually add it to `idf.customExtraPaths`, but this is a path most user will want to have in their global environment to have common tools such as tty, git and other OS common. Should be a default setting, maybe `/usr/bin` is ignored in your `.zhenv` ?

meltdown03, 2021-01-07:
I was kinda thinking about that. I know some dotfiles only load under login
shells. Maybe that's why.

meltdown03, 2021-01-07:
I actually have my PATH set in ~/.pam_environment because I had issues with
VS Code not load system wide environment variables if I launched it via
desktop shortcut.

brianignacio5, 2021-01-07:
IF you run `ESP-IDF: Terminal` it will load env variables as we do fo the monitor but in a terminal you can type `echo $PATH` to see what is the resulting value. We basically do `PATH={idf.customExtraPaths}:PATH`. 

We had some issue adding `/usr/bin` to idf.customExtraPaths (mostly on onboarding) so that's why we removed it.

meltdown03, 2021-01-07:
The PATH is correct in `ESP-IDF: Terminal`.
I found a global workaroung by adding this to my global `settings.json`:
`"terminal.integrated.shellArgs.linux": ["-l"]`
It runs the shell task as a login shell, here is the output:
```
> Executing task: /home/neal/.espressif/python_env/idf4.3_py3.9_env/bin/python /home/neal/esp/esp-idf/tools/idf.py build <

/home/neal/.zshenv:4: command not found: tty
Executing action: all (aliases: build)
Running ninja in directory /home/neal/esp32_examples/nmea0183_parser/build
Executing "ninja all"...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.

Project build complete. To flash, run this command:
/home/neal/.espressif/python_env/idf4.3_py3.9_env/bin/python ../../esp/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/nmea_parser.bin
or run 'idf.py -p (PORT) flash'

Terminal will be reused by tasks, press any key to close it.
```

TL;DR
Before 0.6.0, the IDF extension setup would put `/usr/bin` in the `customExtraPaths` field on its own. Now it doesn't. The `tasks.json` file sets the PATH as `${config:idf.customExtraPaths}` ONLY.
It's like the Tasks run differently, they're set as `shell` types in the JSON file, and only use the PATH set within the file.
Also, If I start a new terminal and run `echo $PATH`, I see `/usr/bin`.
I have `Inherit Env` option set to True in VS Code, without it nothing works.

With the update to 0.6.0, any changes I make globally to `idf.customExtraPaths` change back automatically everytime I open VS Code. I have to change the PATH in the projects `settings.json` file for every project.

brianignacio5, 2021-01-07:
Hmm well tasks.json are defined on each project and are user defined tasks. I guess vscode updated the tasks to completely override env variables. A solution is just change `PATH: "${config:idf.customExtraPaths}"` to `PATH: "${env:PATH}:${config:idf.customExtraPaths}"`. or `PATH: "/usr/bin:${config:idf.customExtraPaths}"` but we can't add it templates since windows uses ; and Mac/linux :

meltdown03, 2021-01-07:
> Hmm well tasks.json are defined on each project and are user defined tasks. I guess vscode updated the tasks to completely override env variables. A solution is just change `PATH: "${config:idf.customExtraPaths}"` to `PATH: "${env:PATH}:${config:idf.customExtraPaths}"`. or `PATH: "/usr/bin:${config:idf.customExtraPaths}"` but we can't add it templates since windows uses ; and Mac/linux :
> 
> Is there a reason why you are not using the extension commands ?

Do you mean the buttons at the bottom of VS Code? I like seeing the output of the build process, which they didn't used to show, I see now that they do. So it's basically just habit at this point.
I did already try changing the PATH setting like you suggested and it works. So I might just change the templates and recompile it if I stay stubborn about using Tasks and my workaround causes other issues, lol.

meltdown03, 2021-01-07:
What about this for the template:
``` bash
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "Build - Build project",
            "type": "shell",
            "command": "${config:idf.pythonBinPath} ${config:idf.espIdfPath}/tools/idf.py build",
            "windows": {
                "command": "${config:idf.pythonBinPathWin} ${config:idf.espIdfPathWin}\\tools\\idf.py build",
                "options": {
                    "env": {
                        "PATH": "${env:PATH};${config:idf.customExtraPaths}"
                    }
                }
            },
            "options": {
                "env": {
                    "PATH": "${env:PATH}:${config:idf.customExtraPaths}"
                }
            },
...
```

Windows can have it's own `options` and `env`.

brianignacio5, 2021-01-12:
@JerryZheng89 Have you fixed the issue ?

embeblue, 2021-01-14:
Just FYI, we are having the same problem Here. I've rolledback to v0.5.1 until a new version is released.

closed 2021-01-18 by brianignacio5

JerryZheng89, 2021-07-01:
> @JerryZheng89 Have you fixed the issue ?

No. The problem still there!