espressif/vscode-esp-idf-extension #271: [Bug Report]: Invalid ESP-IDF when configuring the ESP-IDF extension

xiaolongba, 2021-01-06:
**Describe the bug**
**Invalid ESP-IDF** error information occurred when configuring the extension. However, version 0.51 has not such error information when  doing the same instructions

**To Reproduce**
Configure the ESP-IDF extension and then fill in the relevant path and then an error occurred

**Expected behavior**
Can configure the ESP-IDF extension successfully
**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: Win10
- VSCode Version: 1.52.1
- ESP-IDF Version: 0.60
- Python Version: 3.9

**Logs**

there are not any logs at this moment

brianignacio5, 2021-01-06:
What is the value of `git describe --tags` in your ESP-IDF directory ?

xiaolongba, 2021-01-06:
Hi Bro,

it showed me the following information when I inputted the command you mentioned above:

```
$ git describe --tags
fatal: No names found, cannot describe anything.
```

However, I got the ESP-IDF from espressif repository. also, there is no error that occurred when using the v0.5.1

brianignacio5, 2021-01-06:
It seems there is a small issue with some ESP-IDF version, will fix soon. Thanks

xiaolongba, 2021-01-06:
all right, what can I only do at this moment is to build, flash or monitor manually

closed 2021-01-06 by xiaolongba