espressif/vscode-esp-idf-extension #276: [Bug Report]: Build crashing on custom OpenOCD build

brainstorm, 2021-01-06:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**

I'm having similar symptoms as https://github.com/espressif/vscode-esp-idf-extension/issues/270, but my case is slightly different, I'll be thorough on this bug report...

After enabling `remote_bitbang` on OpenOCD (see PR https://github.com/espressif/openocd-esp32/pull/134), I cannot seem to be able to do anything with the extension anymore since the following error appears on any action (i.e Build):

```json
{
  "message": "Open On-Chip Debugger  v0.10.0-esp32-20201202-2-g7e342cad (2021-01-02-23:47)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n",
  "stack": "Error: Open On-Chip Debugger  v0.10.0-esp32-20201202-2-g7e342cad (2021-01-02-23:47)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n    at /Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/dist/extension.js:1:7720\n    at ChildProcess.exithandler (child_process.js:303:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:483:12)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Socket.<anonymous> (internal/child_process.js:443:11)\n    at Socket.emit (events.js:315:20)\n    at Socket.EventEmitter.emit (domain.js:483:12)\n    at Pipe.<anonymous> (net.js:674:12)",
  "level": "error",
  "timestamp": "2021-01-06T11:34:05.666Z"
}
```

The error log above occurs as well if having OpenOCD up and running... my best guess is that `idf_tools.py` version checking logic is interfering with the extension Build task/command?

**To Reproduce**

Install a slightly tweaked OpenOCD version (different than the one shipped with the extension).

**Expected behavior**

Everything working fine as it should? :)

**Screenshots**
On each Build task, the following error block appears in the extension's error log:

[image]

**Environment (please complete the following information):**

- OS Name & Version: MacOS X 10.15.2 (Catalina)
- VSCode Version: Insiders, latest
- ESP-IDF Version: `master`
- Python Version: 3.9.1

**Logs**

As shown above in the description.

**Additional context**

Please see PR https://github.com/espressif/openocd-esp32/pull/134 for more context. Also providing env vars I'm using:

```
  "idf.espIdfPath": "/Users/romanvg/dev/esp-idf",
  "idf.openOcdConfigs": [
    "interface/remote_bitbang.cfg",
    "target/esp32s2.cfg"
  ],
  "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/Users/romanvg/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/share/openocd/scripts\"}",
```

brainstorm, 2021-01-06:
On the other hand, typing `idf.py build` builds via regular VSCode console without a problem :/

brainstorm, 2021-01-08:
@brianignacio5 I have reason to believe that you have closed this issue a bit too fast, could you reopen?

I've compiled and installed the extension [as indicated in the README.md](https://github.com/espressif/vscode-esp-idf-extension#build-from-source-code) and the error still persists with your changes against `master` (soon to be release 0.6.1):

```
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20201202-2-g7e342cad (2021-01-02-23:47)
Licensed under GNU GPL v2
For bug reports, read
   http://openocd.org/doc/doxygen/bugs.html",
"stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20201202-2-g7e342cad (2021-01-02-23:47)
Licensed under GNU GPL v2
For bug reports, read
   http://openocd.org/doc/doxygen/bugs.html

   at /Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.1/dist/extension.js:1:7720
   at ChildProcess.exithandler (child_process.js:303:7)
   at ChildProcess.emit (events.js:315:20)
   at maybeClose (internal/child_process.js:1021:16)
   at Socket.<anonymous> (internal/child_process.js:443:11)
   at Socket.emit (events.js:315:20)
   at Pipe.<anonymous> (net.js:674:12)",
"level":"error",
"timestamp":"2021-01-08T11:29:13.104Z"
}
```

brainstorm, 2021-01-08:
Also the "Build from source code" in the README.md should be a bit specific about the following point:

> Press F5 to Run with Debugger, this will launch a new VSCode Extension Development Host to debug the extension.

Since I'm getting this prompt to configure the `debugPreTask` right after pressing F5?:

[image]

brianignacio5, 2021-01-08:
oh really ? I removed the tools check from build flash and monitor.  Will  reopen.

If you executed `yarn` and then press F5 it should start the extension with debug. This task is defined in package.json. There should be some error in the terminal

brainstorm, 2021-01-08:
Fixed by:

```diff
diff --git a/.vscode/launch.json b/.vscode/launch.json
index ee48d96..b850207 100644
--- a/.vscode/launch.json
+++ b/.vscode/launch.json
@@ -12,7 +12,7 @@
       "runtimeExecutable": "${execPath}",
       "args": ["--extensionDevelopmentPath=${workspaceFolder}"],
       "outFiles": ["${workspaceFolder}/dist/**/*.js"],
-      "preLaunchTask": "npm: debugPreTask",
+      //"preLaunchTask": "npm: debugPreTask",
       "env": {
         "VSCODE_EXTENSION_MODE": "development"
       }
```

brianignacio5, 2021-01-08:
There is a OpenOCD output tab (next to terminal) which is the openocd server log. 
The log you shown in the issue description is just printing openOCD version as an error, not related to the issue.

brianignacio5, 2021-01-08:
Also how is openOCD related to the build process in any way ? Your log output is not related to this.

In the terminal, when you try to build, there should be a Build Task output showing the error.

brainstorm, 2021-01-08:
Funnily enough, when debugging the extenstion in the "Extension Development Host" VSCode it all seems to build and work fine... but when I install the `.vsix`, the Build fails immediately and that error above shows up.

> In the terminal, when you try to build, there should be a Build Task output showing the error.

No there's no task output on the ".vsix"-installed VSCode editor, when clicking Build, the "Building" popup disappears in 1-2 seconds and the `.log` under the extension dir register one more json log line as shown above.

GOST-UA, 2021-01-08:
@brainstorm 

> EDIT: yarn debugPreTask to switch into development mode...

Looks like the extension was moved to `yarn` some time ago but some pieces of code still working with `npm` though. I would recommend you instead of commenting out `"preLaunchTask": "npm: debugPreTask",` line in launch.json try to replace it with 
`"preLaunchTask": "yarn: debugPreTask",` 

Regarding the issue itself: did you check OpenOCD output on the output tab (`Ctrl+Shift+U`) when this issue occurs?
[image]

brainstorm, 2021-01-09:
> Looks like the extension was moved to `yarn` some time ago but some pieces of code still working with `npm` though. I would recommend you instead of commenting out `"preLaunchTask": "npm: debugPreTask",` line in launch.json try to replace it with
> `"preLaunchTask": "yarn: debugPreTask",`

None of those variants actually work for me:

1. `npm: debugPreTask`
2. `yarn: debugPreTask`
3. `npm debugPreTask`
4. `yarn debugPreTask`

All of the above end up like this:

[image]

That's why commenting it out and pressing `F5` works well since it spaws the "Extension Development Host" (just an extra VSCode window to test/debug the extension)... I guess that running that `yarn debugPreTask` at the terminal once suffices if one doesn't change any of the *.ts code?

> Regarding the issue itself: did you check OpenOCD output on the output tab (`Ctrl+Shift+U`) when this issue occurs?
> [image]

I wiped the extension, rebuilt the VSIX from `master` and reinstalled and now it works! Time to close this and focus on the esp_dap and DEBUGGING.txt next :)

PR #281 (2021-01-07): rm check tools before cmds update tasks
Fix #280
Fix #278 
Fix #277
Fix #276 

Add env:PATH and reorder monitor args in tasks.json
Fix quotes on cmd.exe
Remove check esp-idf tools are valid from build flash commands.
closed 2021-01-09 by brainstorm