espressif/vscode-esp-idf-extension #282: [Bug Report]: Installation on macOS 11.0.1, VS Code 1.52.1, ESP-IDF v 4.1, python v 3.x fails.

mahoefer, 2021-01-07:
**Describe the bug**
Installation on macOS 11.0.1, VS Code 1.52.1, ESP-IDF v 4.1, python v 3.x fails. 

after installation failure I can install the packages manually as shown below

/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip
/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --no-cache-dir "<pkg + range>"

after setting the path variables for IDF_PATH and IDF_PATH/tools I am able to call:

idf.py build, idf.py menuconfig, idf.py -p XXX flash, etc without any issues. 

However, I cannot get the extension to recognize that ESP-IDF has been installed. If I select "Show Examples Projects" from the Command Palette then VS Code asks for a framework with no options. If I manually copy the example project using:

cp -r $IDF_PATH/examples/get-started/hello_world .

and then open the project directory in VS Code I am prompted to install the extension.

How can I point the extension to use the already installed esp-idf toolchains?

Install log:

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Downloading ESP-IDF v4.1 (release version)...
esp-idf-v4.1.zip has (457864) KB
...
Downloaded v4.1 (release version).

Extracted /Users/mark/esp2/esp-idf-v4.1.zip in /Users/mark/esp2.

/bin/sh: esp32s2ulp-elf-as: command not found
Command failed: esp32s2ulp-elf-as --version
/bin/sh: esp32s2ulp-elf-as: command not found
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
/bin/sh: esp32s2ulp-elf-as: command not found

/bin/sh: esp32ulp-elf-as: command not found
Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
/bin/sh: esp32ulp-elf-as: command not found

/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: xtensa-esp32s2-elf-gcc: command not found

/bin/sh: xtensa-esp32-elf-gcc: command not found
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: xtensa-esp32-elf-gcc: command not found

/bin/sh: openocd: command not found
Command failed: openocd --version
/bin/sh: openocd: command not found
Error checking openocd-esp32 version
Error: Command failed: openocd --version
/bin/sh: openocd: command not found

Downloading Toolchain for Xtensa (ESP32) based on GCC
...
xtensa-esp32-elf-gcc8_2_0-esp-2020r2-macos.tar.gz progress: 100.00% (90040.39 / 90040.39) KB
Downloading Toolchain for Xtensa (ESP32-S2) based on GCC
xtensa-esp32s2-elf-gcc8_2_0-esp-2020r2-macos.tar.gz has (90394) KB
...
xtensa-esp32s2-elf-gcc8_2_0-esp-2020r2-macos.tar.gz progress: 100.00% (90393.25 / 90393.25) KB
Downloading Toolchain for ESP32 ULP coprocessor
binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz has (8665) KB
...
binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz progress: 100.00% (8664.92 / 8664.92) KB
Downloading Toolchain for ESP32-S2 ULP coprocessor
binutils-esp32s2ulp-macos-2.28.51-esp-20191205.tar.gz has (8669) KB
...
binutils-esp32s2ulp-macos-2.28.51-esp-20191205.tar.gz progress: 100.00% (8668.16 / 8668.16) KB
Downloading OpenOCD for ESP32
openocd-esp32-macos-0.10.0-esp32-20191114.tar.gz has (1725) KB
...
openocd-esp32-macos-0.10.0-esp32-20191114.tar.gz progress: 100.00% (1724.19 / 1724.19) KB

Installing package Toolchain for Xtensa (ESP32) based on GCC
Installing tar.gz package Toolchain for Xtensa (ESP32) based on GCC
Installing package Toolchain for Xtensa (ESP32-S2) based on GCC
Installing tar.gz package Toolchain for Xtensa (ESP32-S2) based on GCC
Installing package Toolchain for ESP32 ULP coprocessor
Installing tar.gz package Toolchain for ESP32 ULP coprocessor
Installing package Toolchain for ESP32-S2 ULP coprocessor
Installing tar.gz package Toolchain for ESP32-S2 ULP coprocessor
Installing package OpenOCD for ESP32
Installing tar.gz package OpenOCD for ESP32

Creating a new Python environment in /Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env ...

Collecting wheel
  Using cached wheel-0.36.2-py2.py3-none-any.whl (35 kB)
Installing collected packages: wheel
Successfully installed wheel-0.36.2
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip' command.

Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/lib/python3.9/site-packages (from -r /Users/mark/esp2/esp-idf/requirements.txt (line 4)) (49.2.1)
Collecting click>=5.0
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting future>=0.15.2
  Using cached future-0.18.2.tar.gz (829 kB)
Collecting cryptography>=2.1.4
  Using cached cryptography-3.3.1.tar.gz (539 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
    Preparing wheel metadata: started
    Preparing wheel metadata: finished with status 'done'
Collecting pyparsing<2.4.0,>=2.0.3
  Using cached pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
Collecting pyelftools>=0.22
  Using cached pyelftools-0.27-py2.py3-none-any.whl (151 kB)
Collecting cffi>=1.12
  Using cached cffi-1.14.4.tar.gz (471 kB)
Collecting six>=1.4.1
  Using cached six-1.15.0-py2.py3-none-any.whl (10 kB)
Collecting pycparser
  Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)
Building wheels for collected packages: future, cryptography, cffi
Failed to build future cryptography cffi
  WARNING: Building wheel for future failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/2f'
  WARNING: Building wheel for cryptography failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/d4'
  WARNING: Building wheel for cffi failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/30'
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip' command.
Command failed: "/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python" -m pip install --no-warn-script-location -r "/Users/mark/esp2/esp-idf/requirements.txt"
  WARNING: Building wheel for future failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/2f'
  WARNING: Building wheel for cryptography failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/d4'
  WARNING: Building wheel for cffi failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/30'
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip' command.
Command failed: "/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python" -m pip install --no-warn-script-location -r "/Users/mark/esp2/esp-idf/requirements.txt"
  WARNING: Building wheel for future failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/2f'
  WARNING: Building wheel for cryptography failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/d4'
  WARNING: Building wheel for cffi failed: [Errno 13] Permission denied: '/Users/mark/Library/Caches/pip/wheels/30'
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/Users/mark/esp2/.espressif/python_env/idf4.1_py3.9_env/bin/python -m pip install --upgrade pip' command.

mahoefer, 2021-01-08:
I was finally able to finish the install by pressing F1 and then searching for Configure ESP-IDF Extension and selecting using existing setup as outlined here: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md.

Oddly when accessing  Configure ESP-IDF Extension through "View/Command Palette" the using existing setup was never shown.

brianignacio5, 2021-01-08:
I think this happens because of what @dobairoland explains [here](https://github.com/espressif/esp-idf/issues/6111)

mahoefer, 2021-01-08:
thanks for the link, but I had already read it prior to opening the ticket. I also had the same issue with python 3.8 as stated in the original message. (I should've changed the title to python 3.x instead of python 3.9).

The issue was after fixing all the dependencies and setting up the python environment the vs code extension wouldn't recognize the installation and kept asking me to (re-)install. Only when accessing Configure ESP-IDF Extension through F1, did the extension recognize the existing installation

brianignacio5, 2021-01-11:
Closing this issue since it is fixed.

closed 2021-01-11 by brianignacio5