espressif/vscode-esp-idf-extension #285: [Feature Request]: DEBUGGING.txt needs a bit more clarity?

brainstorm, 2021-01-08:
**Is your feature request related to a problem? Please describe.**

Please see the commit comments on: https://github.com/espressif/vscode-esp-idf-extension/commit/b4b91b959b884d2dfae2776380c42f066021bdd6#r45766576 ... for a good description of the current situation and what I've been trying.

**Describe the solution you'd like**

I would love to have a "reasonable defaults" config for the `esp_debug_adapter`... I understand that there might not be an "one size fits all" solution for this, but the docs should be less succint for new users?

brainstorm, 2021-01-09:
I went for the following launch settings:

```json
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
    
        {
            "type": "espidf",
            "name": "Launch",
            "request": "launch",
            //"debugPort": 4444,
            "logLevel": 5,
            "mode": "auto",
        }
    ]
}
```

Then I see this on the Glasgow (remote_bitbang JTAG side):

```
$ glasgow run jtag-openocd --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 -V 3.3 tcp:localhost:5555
I: g.device.hardware: device already has bitstream ID 9cfbf1aff24341a17d570a320b7e700f
I: g.cli: running handler for applet 'jtag-openocd'
I: g.applet.interface.jtag_openocd: port(s) A voltage set to 3.3 V
I: g.applet.interface.jtag_openocd: socket: listening at tcp:localhost:5555
I: g.applet.interface.jtag_openocd: socket: new connection from [::1]:57047
```

And this is the output I'm seeing on the DAP side:

```
2021-01-09 19:07:40,411 - Debug Adapter (main) - DEBUG - START_TIME_2021_01_09_19-07-40
2021-01-09 19:07:40,411 - Debug Adapter (main) - DEBUG - Init errors:None
2021-01-09 19:07:40,412 - Debug Adapter (main) - INFO - Working directory: /
2021-01-09 19:07:40,412 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
    'device_name': 'esp32s2',
    'elfpath': (   '/Users/romanvg/dev/esp32s2_can/twai_network_slave/build/twai_network_slave.elf',),
    'log_file': '/Users/romanvg/dev/esp32s2_can/twai_network_slave/debug.log',
    'log_mult_files': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'toolchain_prefix': None}
DEBUG_ADAPTER_STARTED
2021-01-09 19:07:40,413 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2021-01-09 19:07:40,413 - Debug Adapter (main) - INFO - Starting. Cmd: /Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter_main.py -d 5 -e /Users/romanvg/dev/esp32s2_can/twai_network_slave/build/twai_network_slave.elf -l /Users/romanvg/dev/esp32s2_can/twai_network_slave/debug.log -p 43474 -dn esp32s2 -a 0x10000

2021-01-09 19:07:40,415 - Debug Adapter (main) - INFO - Listening on port 43474
DEBUG_ADAPTER_READY2CONNECT
2021-01-09 19:07:40,415 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2021-01-09 19:07:40,417 - Debug Adapter (main) - DEBUG - Got connection
2021-01-09 19:07:40,418 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 386\\r\\n'<<

2021-01-09 19:07:40,419 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

[... 80 lines cut ...]
2021-01-09 19:07:50,476 - Debug Adapter (main) - ERROR - Failed to connect to "remote 127.0.0.1:3333"!
Traceback (most recent call last):
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/command_processor.py", line 52, in __call__
    on_request(protocol_message)
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/command_processor.py", line 88, in on_initialize_request
    self.da.adapter_init()
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 282, in adapter_init
    raise e
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 278, in adapter_init
    self.start_gdb()
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 506, in start_gdb
    raise e
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 503, in start_gdb
    self._gdb.connect()
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 485, in connect
    self.target_select(remote_mode, self._remote_target, tmo=tmo)
  File "/Users/romanvg/.vscode-insiders/extensions/espressif.esp-idf-extension-0.6.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 219, in target_select
    raise DebuggerError('Failed to connect to "%s %s"!' % (tgt_type, tgt_params))
debug_adapter.debug_backend.debug_backend.defs.DebuggerError: Failed to connect to "remote 127.0.0.1:3333"!
2021-01-09 19:07:50,478 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Failed to connect to \"remote 127.0.0.1:3333\"!\n", "source": {}}, "seq": 6}

2021-01-09 19:07:50,478 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Failed to connect to \"remote 127.0.0.1:3333\"!\n", "source": {}}, "seq": 7}
```

So it's essentially dropping the OpenOCD GDB connection on port `:3333`... investigating why :/

brainstorm, 2021-01-09:
The OpenOCD daemon is up and running:

```shell
$ lsof -i tcp:3333
COMMAND   PID    USER   FD   TYPE            DEVICE SIZE/OFF NODE NAME
openocd 47428 romanvg    7u  IPv4 0xec3526865396c27      0t0  TCP localhost:dec-notes (LISTEN)
```

And connects and replies, apparently:

```shell
$ nc 127.0.0.1 3333
+
$
```

It shows that `+` sign and quickly quits though :-S

brainstorm, 2021-01-10:
Ok, after examining the output from OpenOCD to a single `target remote 127.0.0.1:3333` I'm hitting the following:

```
Error: Target not examined yet
Error executing event gdb-attach on target esp32s2:

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

Looking into how to "examine the target" in OpenOCD...

brainstorm, 2021-01-10:
I got the following but just once (and cannot seem to reproduce), perhaps by mere chance? How can I make sure that the ESP32S2 is in a "good state" upon `gdb-attach` (when I issue the `target remote 127.0.0.1:3333` in the other shell, @igrr?:

```
$ lldb openocd -- -f interface/remote_bitbang.cfg -f target/esp32s2.cfg
(...)
(lldb) run
Process 16819 launched: '/Users/romanvg/.espressif/tools/openocd-esp32/v0.10.0-esp32-20201202/openocd-esp32/bin/openocd' (x86_64)
Open On-Chip Debugger  v0.10.0-esp32-20201202-2-g7e342cad (2021-01-02-23:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : FreeRTOS creation
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:5555
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s2: Debug controller was reset.
Info : esp32s2: Core was reset.
Error: esp32s2: DSR (FFFFFFFF) indicates target still busy!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: esp32s2: DSR (FFFFFFFF) indicates target still busy!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: esp32s2: DSR (FFFFFFFF) indicates target still busy!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Info : esp32s2: Target halted, PC=0xFFFFFFFF, debug_reason=00000003
Error: Failed to read break instruction!
Error: esp32s2: DSR (FFFFFFFF) indicates target still busy!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: esp32s2: DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32s2: Failed writing 4 bytes at address 0x3f41f064
Error: Failed to write ESP32_S2_TIMG0WDT_PROTECT (-4)!
Polling target esp32s2 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : esp32s2: Debug controller was reset.
Info : esp32s2: Core was reset.
Polling target esp32s2 failed, trying to reexamine
Info : Listening on port 3333 for gdb connections
Info : esp32s2: Core was reset.
Info : esp32s2: Core was reset.
(...)
Info : esp32s2: Core was reset.
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32s2: Debug controller was reset.
Info : esp32s2: Core was reset.
Info : esp32s2: Target halted, PC=0x00000000, debug_reason=00000000
Warn : Unknown ESP32-S2 chip revision (0x0)!
Error: Invalid stub!
Warn : Failed to get flash mappings (-4)!
Error: Invalid stub!
Error: Invalid stub!
Info : Auto-detected flash bank 'esp32s2.flash' size 4194303 KB
Info : Using flash bank 'esp32s2.flash' size 4194303 KB
Error: Invalid stub!
Warn : Failed to get flash mappings (-4)!
Error: Invalid stub!
Info : Using flash bank 'esp32s2.irom' size 0 KB
Error: Invalid stub!
Warn : Failed to get flash mappings (-4)!
Error: Invalid stub!
Info : Using flash bank 'esp32s2.drom' size 0 KB
Error: Failed to find HALTED core!
Error: Target not examined yet
Error: Target not examined yet
Error: esp32s2: Failed to halt target to remove flash BPs (-4)!
Info : dropped 'gdb' connection
```

brainstorm, 2021-01-10:
Passing `-d` to OpenOCD (as in `$ lldb openocd -- -d -f interface/remote_bitbang.cfg -f target/esp32s2.cfg`):

```
Info : 1339 1157869 server.c:100 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 1340 1157869 breakpoints.c:358 breakpoint_clear_target_internal(): Delete all breakpoints for target: esp32s2
Debug: 1341 1157869 breakpoints.c:552 watchpoint_clear_target(): Delete all watchpoints for target: esp32s2
Debug: 1342 1157869 FreeRTOS.c:1018 FreeRTOS_clean(): FreeRTOS_clean
Debug: 1343 1157869 FreeRTOS.c:491 FreeRTOS_update_threads(): FreeRTOS_update_threads
Warn : 1344 1157870 FreeRTOS.c:497 FreeRTOS_update_threads(): No symbols for FreeRTOS!
Debug: 1345 1157870 target.c:1636 target_call_event_callbacks(): target event 19 (gdb-attach) for core esp32s2
Debug: 1346 1157870 target.c:4637 target_handle_event(): target(0): esp32s2 (esp32s2) event: 19 (gdb-attach) action:
	#$_TARGETNAME xtensa smpbreak BreakIn BreakOut
	# necessary to auto-probe flash bank when GDB is connected
	reset
	halt

Debug: 1347 1157870 command.c:143 script_debug(): command - reset reset
Debug: 1349 1157870 target.c:1654 target_call_reset_callbacks(): target reset 1 (run)
Debug: 1350 1157870 command.c:143 script_debug(): command - target target names
Debug: 1351 1157870 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event reset-start
Debug: 1352 1157870 command.c:143 script_debug(): command - transport transport select
Debug: 1353 1157870 command.c:143 script_debug(): command - jtag jtag arp_init-reset
Debug: 1354 1157870 core.c:1576 jtag_init_reset(): Initializing with hard TRST+SRST reset
Debug: 1355 1157870 core.c:742 jtag_add_reset(): JTAG reset with TLR instead of TRST
Debug: 1356 1157870 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 1357 1157870 core.c:1462 jtag_init_inner(): Init JTAG chain
Debug: 1358 1157870 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 1359 1157870 core.c:1128 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 1360 1157870 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Info : 1361 1157970 core.c:1027 jtag_examine_chain_display(): JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Debug: 1362 1157970 core.c:1258 jtag_validate_ircapture(): IR capture validation scan
Debug: 1363 1157974 core.c:1316 jtag_validate_ircapture(): esp32s2.cpu: IR capture 0x01
Debug: 1364 1157974 command.c:143 script_debug(): command - transport transport select
Debug: 1365 1157974 command.c:143 script_debug(): command - esp32s2 esp32s2 cget -chain-position
Debug: 1366 1157974 command.c:143 script_debug(): command - jtag jtag tapisenabled esp32s2.cpu
Debug: 1367 1157974 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event examine-start
Debug: 1368 1157974 command.c:143 script_debug(): command - esp32s2 esp32s2 arp_examine allow-defer
Debug: 1369 1157974 xtensa.c:685 xtensa_examine(): xtensa_examine coreid=0
Debug: 1370 1157984 xtensa.c:698 xtensa_examine(): OCD_ID = 03339fd2
Debug: 1371 1157984 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event examine-end
Debug: 1372 1157984 command.c:143 script_debug(): command - esp32s2 esp32s2 invoke-event reset-assert-pre
Debug: 1373 1157984 command.c:143 script_debug(): command - transport transport select
Debug: 1374 1157984 command.c:143 script_debug(): command - esp32s2 esp32s2 cget -chain-position
Debug: 1375 1157984 command.c:143 script_debug(): command - jtag jtag tapisenabled esp32s2.cpu
Debug: 1376 1157984 command.c:143 script_debug(): command - esp32s2 esp32s2 arp_reset assert 0
Debug: 1377 1157984 target.c:2010 target_free_all_working_areas_restore(): freeing all working areas
Debug: 1378 1157984 target.c:2010 target_free_all_working_areas_restore(): freeing all working areas
Debug: 1379 1157984 esp32s2.c:293 esp32s2_assert_reset(): esp32s2: begin
Debug: 1380 1157984 esp32s2.c:373 esp32s2_soc_reset(): start
Debug: 1381 1157984 esp32s2.c:377 esp32s2_soc_reset(): esp32s2_soc_reset: Target not halted before SoC reset, trying to halt it first
Debug: 1382 1157984 xtensa.c:1132 xtensa_halt(): xtensa_halt, target: esp32s2
Debug: 1383 1157993 xtensa.c:1144 xtensa_halt(): esp32s2: Core status 0x0
Debug: 1384 1158012 esp_xtensa.c:210 esp_xtensa_poll(): esp32s2: Clear debug stubs info
Debug: 1385 1158019 target.c:3115 target_wait_state(): waiting for target halted...
Debug: 1386 1158037 xtensa.c:889 xtensa_fetch_all_regs(): esp32s2: start
Debug: 1387 1159275 xtensa.c:787 xtensa_core_status_check(): esp32s2: DSR (00000000)
Debug: 1388 1160285 xtensa.c:787 xtensa_core_status_check(): esp32s2: DSR (00000000)
Debug: 1389 1160323 xtensa.c:787 xtensa_core_status_check(): esp32s2: DSR (00000000)
Debug: 1390 1160323 xtensa.c:1847 xtensa_poll(): esp32s2: Target halted, pc=0x00000000, debug_reason=00000000, oldstate=00000003
Debug: 1391 1160323 xtensa.c:1852 xtensa_poll(): esp32s2: Halt reason=0x00000000, exc_cause=0, dsr=0x00000000
Info : 1392 1160323 xtensa.c:1855 xtensa_poll(): esp32s2: Target halted, PC=0x00000000, debug_reason=00000000
Debug: 1393 1160337 target.c:1636 target_call_event_callbacks(): target event 0 (gdb-halt) for core esp32s2
Debug: 1394 1160337 esp32s2.c:621 esp32s2_handle_target_event(): 0
Debug: 1395 1160337 esp_xtensa.c:109 esp_xtensa_handle_target_event(): 0
Debug: 1396 1160337 xtensa.c:2433 xtensa_handle_target_event(): 0
[... cut ...]
I (4546) BrainCAN: Transmitted TWAI packet
I (6646) BrainCAN: Driver started
I (6646) BrainCAN: Driver stopped
I (6646) BrainCAN: Transmitted TWAI packet
I (8746) BrainCAN: Driver started
I (8746) BrainCAN: Driver stopped
I (8746) BrainCAN: Transmitted TWAI packet
I (10846) BrainCAN: Driver started
I (10846) BrainCAN: Driver stopped
I (10846) BrainCAN: Transmitted TWAI packet
I (12946) BrainCAN: Driver started
I (12946) BrainCAN: Driver stopped
I (12946) BrainCAN: Transmitted TWAI packet
(...)
```

brianignacio5, 2021-01-11:
What do you think are reasonable defaults for `launch.json` ? Can you expand on your feature request ?

Based on the information you shared, seems that default launch.json seems sufficient (with the exception of debug level 5, in the [debug adapter repository this value is 2 by default](https://github.com/espressif/esp-debug-adapter/blob/master/debug_adapter/cli.py#L87)) so I believe your desired behavior is related to openOCD project itself.

Please let us keep related issues to its respective repository, so we can fix these issues.

closed 2021-01-11 by brianignacio5

brainstorm, 2021-01-11:
That's right @brianignacio5, I kept this issue as sort-of a journal as I was fixing this, I'll move to openocd-esp32 repo instead.

W.r.t DEBUGGING.txt, I would have expected an overall description/diagram of which ports communicate with which components since there are a few moving parts (ports: 4444 (telnet/tcl??), 5555 (openocd), 3333 (gdb), 43474 (DAP), etc...

brainstorm, 2021-01-11:
@brianignacio5 At least this reference?: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/jtag-debugging/index.html#how-it-works