espressif/vscode-esp-idf-extension #287: [Bug Report]: Build status notification doesn't disappear when build fails

Elijahg, 2021-01-10:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A build failure results in the build progress notification at the bottom right staying visible and stuck at whatever point the build fails at. This also means subsequent builds require you to click cancel, otherwise the error "Wait for ESP-IDF build or flash to finish" shows. Clicking cancel also closes the active terminal so you can't see the build log and its reason for failure. 

**To Reproduce**
Cause a build error by trying to compile invalid C.

**Expected behavior**
A build failed notification to be displayed and the build progress notification to disappear, also the current terminal/build log to be preserved.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment (please complete the following information):**

- OS Name & Version: macOS 11.0.1
- VSCode Version: 1.52.1
- ESP-IDF Version: 0.6.0
- Python Version: 3.9.0

**Logs**
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/5792183/esp_idf_vsc_ext.log)

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath`: /Users/elijah/Scratch/esp-idf
- `idf.pythonBinPath`: /Users/elijah/.espressif/python_env/idf4.2_py3.9_env/bin/python3
- `idf.openOcdConfigs`: board/esp32-wrover-kit-1.8v
- `idf.customExtraPaths`: /Users/elijah/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/elijah/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/elijah/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/elijah/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/elijah/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin"
- `idf.customExtraVars`: {\"OPENOCD_SCRIPTS\":\"/Users/elijah/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\"}
- `idf.adapterTargetName` esp32

**Additional context**
The previous version of the extension worked fine!

Elijahg, 2021-01-10:
Further investigation reveals this may be related to incorrect toolchain versions, since I get the following early in the build process: 

```-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /Users/elijah/Scratch/esp-idf/requirements.txt are satisfied.
CMake Warning at /Users/elijah/Scratch/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain
  /Users/elijah/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  version 8.2.0 is not the supported version 8.4.0.  Check Getting Started
  documentation or proceed at own risk.
Call Stack (most recent call first):
  /Users/elijah/Scratch/esp-idf/components/esp32/project_include.cmake:30 (gcc_version_check)
  /Users/elijah/Scratch/esp-idf/tools/cmake/build.cmake:306 (include)
  /Users/elijah/Scratch/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
  /Users/elijah/Scratch/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at /Users/elijah/Scratch/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  /Users/elijah/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
  crosstool-ng version esp-2020r2 doesn't match supported version esp-2020r3.
  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  /Users/elijah/Scratch/esp-idf/components/esp32/project_include.cmake:31 (crosstool_version_check)
  /Users/elijah/Scratch/esp-idf/tools/cmake/build.cmake:306 (include)
  /Users/elijah/Scratch/esp-idf/tools/cmake/build.cmake:451 (__build_process_project_includes)
  /Users/elijah/Scratch/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Deprecation Warning at /Users/elijah/Scratch/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
```

That was post-upgrade of the VSCode extension, and after running onboarding again manually. Seems the extension could do with asking the user if they want to upgrade the toolchain rather than continuing to use the old one. I've manually removed the toolchain and run onboarding again. But this time the build fails with an error `/Users/elijah/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-ar: No such file or directory`. This is despite `idf.customExtraPaths` being 
``` 
"/Users/elijah/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/Users/elijah/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/Users/elijah/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/elijah/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/elijah/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin"
```

So for some reason, the build system is getting the old path from somewhere...

brianignacio5, 2021-01-11:
Have you checked if your project has a `.vscode/settings.json` file with `idf.customExtraPaths` ? This will override any Global settings such as User settings or Workspaces settings?

sle118, 2021-01-12:
Reporting the same behavior here. VS Code build doesn't seem to end on build failures. 

I have to manually close the build task, and only then VS Code is able to show errors.

brianignacio5, 2021-01-15:
Thanks for reporting this issue. Will fix #291 

Please give it a try if you have some time.

Elijahg, 2021-01-19:
> Have you checked if your project has a `.vscode/settings.json` file with `idf.customExtraPaths` ? This will override any Global settings such as User settings or Workspaces settings?

Sorry @brianignacio5 I didn't have notifications on! I did have an `idf.customExtraPaths` and ended up deleting the .vscode/settings.json to fix it, and re-running Onboarding. Seems there must have been a duplicate, incorrect path somehow. Maybe one time I ran onboarding I chose to save the config in the wrong place. Thanks for your help! :)

closed 2021-01-19 by brianignacio5

cgimenez, 2022-01-07:
I'm having the exact same problem on a clean & fresh install : the build failed notification never disappears

And btw, is there any setting in esp-idf configuration where I can disable all these notifications ? I find them annoying because telling me "flash done" is useless, I can see if flashing failed or not right in the terminal tab... So...

brianignacio5, 2022-01-07:
To disable them, just set the `idf.notificationSilentMode: true` in your settings.json as defined in the [settings documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md).

cgimenez, 2022-01-07:
This setting have no effect : I updated settings.json, restarted vs code and the notifications are still here.

I'm must had, that with notifications enabled, sometimes none of them disappear. Eg. 10 minutes ago, I changed a setting in the SDK configuration editor and rebuilt my project. I got 3 notifications "SDK configuration saved", "Build success" and "Flash done", I had to close them all myself...

cgimenez, 2022-01-08:
In fact it's pretty strange :

if in .vscode/settings.json, `idf.notificationSilentMode: false`  has not effect
if in my global settings.json file (the one which is modified by the esp-idf setup process), I still get notifications but there's no more informations displayed in the 'terminal' panel - it's the opposite of what I need (terminal display AND no notifications)

cgimenez, 2022-01-10:
So, this behaviour seems to be "by-design" and in that case I really wonder why. 

For those who really want to get ride of notifications while keeping terminal output :
- open ~/.vscode/extentions/espressif.esp-idf-extension-X.Y.Z/dist
- make a copy of the extension.js file (just in case)
- open the original file with a text editor
- locate the string "Invalid error level"
- just above you should see a line like "Transport{constructor(e){super(e)}log(e,t,n,a){const"
- insert "return;" before the const keyword
- restart vscode

Flash done, build done, build failed and some others will be lost in Mordor - progress will still be here though

Of course, these changes will be overwritten on next extension update...

brianignacio5, 2022-01-11:
@cgimenez I've tested on Windows and MacOS and setting `idf.notificationSilentMode: true` in your settings.json will not show those *Flash done* messages. Maybe there is a local setting in your project settings.json or workspace settings overriding the global user setting value.

With regards to terminal display, is true that the terminal is not being focused. I'll make a PR to fix this soon.

brianignacio5, 2022-01-13:
The thing is Visual Studio Code has a [settings precedence](https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence) so you could have settings saved in the global settings.json, a .code-workspace file or the `<your-project-dir>/.vscode/settings.json`. From the latter to the former. There is also default values also.

cgimenez, 2022-01-13:
I'm aware of the settings precedence, but settings.json is ignored in my case, eg with "idf.flashBaudRate": "921600"
- in global settings.json only -> ok
- in .code-workspace file -> ok with precedence over global
- in .vscode/settings.json -> ignored

Not a big deal in my case because I've one workspace equals one project (a "main" subfolder)
And it's not related with idf.flashBaudRate setting, the same occurs with all idf settings

brianignacio5, 2022-01-13:
I see. I'm not sure why but I can't reproduce this in macOS. For example, I tested `idf.notificationSilentMode` in a `.vscode/settings.json` project and seems to change the value. I'll keep trying to reproduce

PR #291 (2021-01-19): fix terminal output close bugs
Fix task output parsing for extension tasks on Windows.
Fix emit close event code !== 0

Fix #287 
Fix #290